espressif/esp-idf #2337: TG1WDT_SYS_RESET triggered during Wi-Fi communication

pellebk, 2018-08-27:
## Environment

- Development Kit: none
- Core (if using chip or module): ESP32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): d91c4251
- Development Env:	 other
- Operating System: Windows
- Power Supply: 3.3V

## Problem Description
I'm trying to continuously send data (TCP) but get resets caused by TG1WDT_SYS_RESET during normal Wi-Fi operation. ESP32 operating in STA mode only is connected to AP. It seems to happen more often when connected to another ESP32 acting as AP than when connected to a Cisco AP, but it still happens in both cases.
It also seems to be more frequent when running the cores on 160 than 240 MHz.

I don't get any debug prints when it happens.

### Expected Behavior
No reset is expected in this setup

### Actual Behavior
The chip is reset by a TG1WDT_SYS_RESET

### Steps to reproduce
Connect ESP32 in STA mode to another ESP32 acting as AP.
Continuously send random TCP data from the STA to the AP. I'm sending ~92kbps but have seen it at higher speeds as well.
Wait for the WD to trigger.

TimXia, 2018-08-27:
@pellebk Is there any log? Can any WiFi example of IDF reproduce the exception? If not, can you provide a demo which can reproduce the exception?

pellebk, 2018-08-28:
No, I don't get any logs for this, I can just see what happens with the reset reason code.

I'm trying to reproduce this using the WiFi iperf example, but I run into problem when starting the application with the following log;
> E (48) flash_parts: partition 0 invalid magic number 0x0
> E (49) boot: Failed to verify partition table
> E (50) boot: load partition table error!

Any idea?

TimXia, 2018-08-30:
@Ivan @angus Please help.

pellebk, 2018-08-30:
I've been able to run the iperf sample, but not been able to reproduce the error with this (yet).

TimXia, 2018-08-30:
What is the difference between your code and the iperf example? Can you provide a demo which can reproduce it?

TimXia, 2018-08-31:
@pellebk If use single core, can you reproduce this issue? @igrr Please help. It seems that dport register access causes interrupt watch dog.

pellebk, 2018-08-31:
No, I'm not able to reproduce this with single core, nor when I pin the application to the same core used by WiFi.

TimXia, 2018-08-31:
@pellebk We will provide you a patch next monday. Please try again then.

pellebk, 2018-09-03:
Thank you, I'm trying to reproduce the issue now. What should my expectations be? Is it a try for a fix or something to get more debug information?

TimXia, 2018-09-03:
@pellebk It is a fix.

closed 2018-09-04 by FayeY

sjoteppagol, 2018-09-06:
@TimXia I tried to use this patch on **ESP-IDF v3.0-dirty 2nd stage bootloader**, but I got error as undefined reference to `phy_enter_critical' and undefined reference to `phy_exit_critical'. Is there any dependency !?
Could you please confirm whether this patch is part of new ESP-IDF v3.1 !? If no then when can I expect the next release with this patch ?

Thank you

TimXia, 2018-09-07:
@sjoteppagol This patch is not for IDF v3.0. It's for IDF master branch(v3.2). Later for IDF v3.1 bugfix release version(maybe IDF v3.1.1) and IDF v3.0.4, it will be also fixed.

hvishu, 2018-09-07:
Thanks @TimXia
We are using 3.0 base code, good to know that this issue being fixed for that base as well. When can we expect v3.0.4 release ( I see it currently in RC1)?

pellebk, 2018-09-10:
The patch only seems to help for the STA. When running our setup, we see what seems to be a similar crash but much later, now on the AP (that is receiving data).

Both CPUs are using the uxPortCompareSet function, CPU0 coming from:
esp-idf/components/freertos/tasks.c:2511: portTICK_TYPE_ENTER_CRITICAL( &xTickCountMutex ); and CPU1 coming from:
esp-idf/components/freertos/portmux_impl.inc.h:86: PORTMUX_COMPARE_SET_FN_NAME

TimXia, 2018-09-18:
@pellebk Could you please show me the whole log?

pellebk, 2018-09-18:
I've have not been able to reproduce this issue with the latest lib that was built for the 3.1 branch

TimXia, 2018-09-18:
@pellebk So, we can close this issue?

pellebk, 2018-09-18:
Yes, at least for the moment, I will continue to try to reproduce this with the current setup.