espressif/esp-idf #2338: Bluedroid ATT Server: Read By Type Request stalls server

mringwal, 2018-08-27:
When the Bluedroid GATT Server receives an ATT_READ_BY_TYPE_REQUEST query, it responds correctly the first time. However, further ATT Requests are not answered. Similarly, if the responste to the ATT_READ_BY_TYPE_REQUEST does not include the end handle, and the GATT Client sends another  ATT_READ_BY_TYPE_REQUEST, it stalls in the same way.

I've attached a HCI Log from BTstack performing a Read Value of Characteristic by UUID16 twice against the unmodified esp-idf gatt_server_service_table example application. The first time it is answered, while the second request times out.

[att-read-by-type.pklg.zip](https://github.com/espressif/esp-idf/files/2324989/att-read-by-type.pklg.zip)

Weijian-Espressif, 2018-09-30:
@mahavirj I test our gatt_server_service_table example, it works well.
[image]

mringwal, 2018-09-30:
Interesting. Looking at my previous log, the GATT answered three times and only stalled at the fourth request.

I'll repeat the test again master tomorrow.

mringwal, 2018-10-02:
Hi! I've retested sending the Read By Type Request multiple times and don't see it hanging. Thanks for fixing/testing it.

closed 2018-10-02 by mringwal