espressif/esp-idf #2339: esp_ota_write function blocking,but other task is running

jaskle, 2018-08-28:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Core (if using chip or module): ESP-WROOM32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): d91c4251
- Development Env:	visualGDB 5.4p4
- Operating System:	Windows
- Power Supply:         USB

## Problem Description
esp_ota_write function blocking,but ble is running!!!!
Every 10 times or so.
Basically blocked every time, when ble data exchange is frequent

### Expected Behavior

### Actual Behavior

### Steps to repropduce

I tried to remove all code, only uart ota(by xmodem), work ok.
I tried to use the wifi demo, the fault is still
Other modules still work when a fault occurs

I am trying to track the blocking position:
esp_partition_write->spi_flash_write->spi_flash_guard_start

I think that the execution lock could not be obtained at a specific time, causing the function to be blocked.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    xTaskCreate(jlog_task, "jlog_task", 4096, NULL, 0, NULL);
i try use 8000 usStackDepth and high uxPriority,The error remains

The ota function is called in the task.

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

jaskle, 2018-08-30:
I found that it was caused by esp_ipc_call,
esp_err_t ret = esp_ipc_call(other_cpuid, &spi_flash_op_block_func, (void*) other_cpuid);
I tried to increase the ipc stack, but the fault is still

jaskle, 2018-09-03:
I tried to add vDelay after the write function, the fault basically disappeared, but it still exists.
Is it because I frequently try to read the buffer and the task time is insufficient?

Alvin1Zhang, 2018-09-20:
@jaskle Could you help share some updates, or could yo reliably reproduce this issue? Many thanks.

Alvin1Zhang, 2018-10-17:
@jaskle Could you help share if there are any updates for this issue? Thanks.

mahavirj, 2018-12-20:
Closing, please feel to reopen if there is any update.

closed 2018-12-20 by mahavirj