espressif/esp-idf #2342: FreeRTOS component, possible xTimerDelete / prvProcessReceivedCommands flaw?

paulreimer, 2018-08-28:
Hopefully I have missed something or am using FreeRTOS incorrectly, but I have the following issue:

Preconditions:
- Using FreeRTOS Timer API
- Multiple "one-shot" timers or recurring timers which are cancelled from other tasks.
- (Optionally) Sometimes deleting these timers from within the timer svc task itself.
- (Possibly) Unprocessed timer events for the same `Timer_t` in the timer svc's queue at time of deletion
- (Possibly) Multiple events for the same `Timer_t` in the timer svc's queue e.g. in the case of a fast timer that occasionally does some enough work to delay the next event.

As I understand it, the timer svc is constantly servicing the queue in its own task here:
https://github.com/espressif/esp-idf/blob/cc8ad721f98ffbc7953ece70616c07422b58e06b/components/freertos/timers.c#L701

And if a timer is deleted by a different task, then taking this xTimer pointer would be invalid:
https://github.com/espressif/esp-idf/blob/cc8ad721f98ffbc7953ece70616c07422b58e06b/components/freertos/timers.c#L731

So this does seem to be the case for me, if I omit `xTimerDelete` completely (leak), then everything is fine. However if I set a fast-ish timer (e.g. 50ms) then I get the following stacktrace:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400904a2  PS      : 0x00060330  A0      : 0x8008e980  A1      : 0x3ffc4350
0x400904a2: vListInsert at esp-idf/components/freertos/list.c:188 (discriminator 1)

A2      : 0x3ffb63ec  A3      : 0x3ffd8a9c  A4      : 0x1c378132  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3ffc0480  A8      : 0x00000004  A9      : 0x3ffe1eb4
A10     : 0x00000006  A11     : 0x00007203  A12     : 0x00060320  A13     : 0x3ffc43ac
A14     : 0x3ffd8a98  A15     : 0x3ffd8a98  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x400904a2:0x3ffc4350 0x4008e97d:0x3ffc4370 0x4008ed71:0x3ffc4390 0x4008ee41:0x3ffc43d0
0x400904a2: vListInsert at esp-idf/components/freertos/list.c:188 (discriminator 1)

0x4008e97d: prvInsertTimerInActiveList at esp-idf/components/freertos/timers.c:961

0x4008ed71: prvProcessReceivedCommands at esp-idf/components/freertos/timers.c:961

0x4008ee41: prvTimerTask at esp-idf/components/freertos/timers.c:961
```

igrr, 2018-09-09:
Timer deletion is handled from timer task itself, so there shouldn't be a race condition between stopping and deleting a timer. Does this issue happen in dual core mode? If yes, could you please try in single core mode? Also, if you could share some code which can trigger the issue, it would help a lot. Thanks.

igrr, 2018-09-09:
> multiple pending events for the same timer in the timer task's queue, when the delete happens

Could you please sketch the code which would cause such scenario? Unless the application pushes more events related to the same timer into the queue, delete event should be the last one.

sudeep-mohanty, 2022-11-28:
@paulreimer Apologies for getting to this so late. As I analyse this issue, it seems that what you have reported is a corner condition which may occur under high system load and a specific timer command sequence wherein the tmrCOMMAND_DELETE is not that last command. Usually, this should never occur in a real world scenario. However, to protect against such a scenario, some checks could be added to the timer module to track whether a particular timer has already been deleted before processing any other command.
It would not make sense for us to inflict such changes to the kernel in the FreeRTOS module in ESP-IDF but rather have this change discussed with/added to the upstream FreeRTOS repository if possible. I would request you to pursue the same and hence closing the ESP-IDF issue for now.

closed 2022-11-28 by sudeep-mohanty