espressif/esp-idf #2343: xtensa-esp32-elf-gdb crash

v1993, 2018-08-28:
## Environment

- Development Kit:      noname
- Core (if using chip or module): ESP-WROOM-32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 30545f4c
- Development Env:		Eclipse
- Operating System:		Ubuntu
- Power Supply:         USB (5v via JTAG)

## Problem Description

Debugger crashes when I try to debug my problem.

### Expected Behavior
Debug my program.

### Actual Behavior
Crash with error message.

### Steps to repropduce

It is pretty complicated in my case…

## Debug Logs

```
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".

[New Remote target]

[New Thread 1073451264]
[New Thread 1073449364]
[New Thread 1073453920]
[New Thread 1073437004]
[New Thread 1073412744]
[New Thread 1073435616]
[New Thread 1073438136]
[Switching to Thread 1073447248]

Temporary breakpoint 5, app_main () at /home/v/compile/esp32-sound-vorbis-test/main/main.cpp:73
73	extern "C" void app_main(void) {

[New Thread 1073555376]
[Switching to Thread 1073555376]

Breakpoint 1, SoundVorbis::SoundProviderVorbis::task_code (this=0x3ffbdf80) at /home/v/compile/esp32-sound-vorbis-test/components/esp32-sound-vorbis/soundProviderVorbis.cpp:133
133			std::unique_lock<std::mutex> heapLock(heapUsed);
[Switching to Thread 1073412744]

Breakpoint 2, Sound::SoundMixer::soundCallback (this=0x3ffba148) at /home/v/compile/esp32-sound-vorbis-test/components/esp32-sound/soundMixer.cpp:170
170								if (sound->repeat) {
[Switching to Thread 1073555376]

Breakpoint 1, SoundVorbis::SoundProviderVorbis::task_code (this=0x3ffbdf80) at /home/v/compile/esp32-sound-vorbis-test/components/esp32-sound-vorbis/soundProviderVorbis.cpp:133
133			std::unique_lock<std::mutex> heapLock(heapUsed);
[Switching to Thread 1073412744]

Breakpoint 2, Sound::SoundMixer::soundCallback (this=0x3ffba148) at /home/v/compile/esp32-sound-vorbis-test/components/esp32-sound/soundMixer.cpp:170
170								if (sound->repeat) {
[Switching to Thread 1073555376]

Breakpoint 1, SoundVorbis::SoundProviderVorbis::task_code (this=0x3ffbdf80) at /home/v/compile/esp32-sound-vorbis-test/components/esp32-sound-vorbis/soundProviderVorbis.cpp:133
133			std::unique_lock<std::mutex> heapLock(heapUsed);
[Switching to Thread 1073412744]

Breakpoint 2, Sound::SoundMixer::soundCallback (this=0x3ffba148) at /home/v/compile/esp32-sound-vorbis-test/components/esp32-sound/soundMixer.cpp:170
170								if (sound->repeat) {
[Switching to Thread 1073555376]

Breakpoint 1, SoundVorbis::SoundProviderVorbis::task_code (this=0x3ffbdf80) at /home/v/compile/esp32-sound-vorbis-test/components/esp32-sound-vorbis/soundProviderVorbis.cpp:133
133			std::unique_lock<std::mutex> heapLock(heapUsed);
[Switching to Thread 1073412744]

Breakpoint 2, Sound::SoundMixer::soundCallback (this=0x3ffba148) at /home/v/compile/esp32-sound-vorbis-test/components/esp32-sound/soundMixer.cpp:170
170								if (sound->repeat) {
/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]

This is a bug, please report it.  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.

/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
```

## Other items if possible

None.

v1993, 2018-09-01:
Glad to see progress on it! How soon (about) it will be fixed?

igrr, 2018-09-01:
There will be two steps: first, we will add an option (enabled by default) to ESP-IDF to wrap user provided task function into another function. This will mean that situation when inline function is called from outer stack frame (which is what makes gdb crash) will not occur. It will cost 16 bytes of stack and may be disabled in production builds. Timeline is likely next week.

The second step will be to update the tool chain, adding a patch for GDB to handle this case correctly. Since we are also updating GCC to 8.2, this will be a pretty big change, and we are holding it off until IDF 4.0 to reduce the chance of breaking user applications. Timeline for this is TBD, current plan is to have IDF 4.0 released late this year or early 2019.

v1993, 2018-09-01:
I know about gcc's `__attribute__((noinline))`. Which functions should be marked with it to do debugging just fine before fix?

igrr, 2018-09-01:
Well, hard to say without looking at your task code. You can add -fno-inline-functions to CFLAGS or CXXFLAGS of the file where your task is defined. E.g. in your component.mk:

myfile.o: CFLAGS += -fno-inline-functions

v1993, 2018-09-01:
Ok, thank you. Should I noinline only functions which are passed to xCreateTask or some other too? BTW, my project is opensource and can be found here: https://github.com/v1993/esp32-sound/tree/dev

v1993, 2018-09-07:
@igrr I'm trying to continue debugging now. Now there is new error message:

    /builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:343: internal-error: skip_inline_frames: Assertion `find_inline_frame_state (ptid) == NULL' failed.
How can I overpass this one? I think that it is same issue.

v1993, 2018-09-07:
@igrr Even after adding `-fno-inline-functions` to all files (via `Makefile.projbuild`) I still can't view stack of one my task. Same error as before:

```
/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]

This is a bug, please report it.  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.

/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
```

igrr, 2018-09-09:
@v1993 is that task also part of the code repository you have linked above? Can you please point which task is that?

v1993, 2018-09-09:
@iggr task code can be found here: https://github.com/v1993/esp32-sound-vorbis/blob/master/soundProviderVorbis.cpp

And `xTaskCreate` can be found here: https://github.com/v1993/esp32-sound/blob/dev/soundProviderTask.cpp

igrr, 2018-09-09:
Thanks @v1993. Can you please point to the top level project used to compile this component, and sdkconfig file used? I see https://github.com/v1993/esp32-sound-pcm-test repository on your Github page which looks like it uses same components, do you see the issue with this project as well?

loceur, 2018-09-09:
Having the same issues.  Thanks for working on this!

v1993, 2018-09-10:
@iggr Top level project isn't shared yet (I can send .zip if needed). That one is for different component, pcm backend, I didn't tested that one. It works and doesn't require debugging.

I think that linker does link-time optimization and inlines function from one module to another (it is possible).

igrr, 2018-09-10:
Thanks, please share the project zip either by attaching it here, or by sending it to ivan at espressif.com. We will have a look what is going on.

iggr, 2018-09-10:
Hi everyone, 
What do I need to do so people stopped sending me messages intended for @igrr? 
P.S. changing account is not an option :) 
Thank you for your attention.

RA5040, 2018-09-10:
What I really want to do is to disable all inline expansions for debugging. The Og switch obviously doesn't do that, so what's needed is an O0 switch. Is that what you are working on? Is there a way of doing it manually?

If not, is there a way of passing '-fno-inline-small-functions' to gcc?

'attribute ((noinline))' doesn't work ... in that dbg still crashes.

Do you have an estimate of when this will be fixed?

(Also, is there anywhere to see a summary of current serious problems (like this one), with a workaround if applicable and a fix schedule? At the moment what I'm doing is stumbling on bugs and spending a lot of time (with help from you guys, for sure!) figuring out if it's my fault or not and what to do about it. It's a terrible (but avoidable) waste of time and energy.)

gerekon, 2018-09-14:
> Do you have an estimate of when this will be fixed?

@RA5040, @igrr has provided the schedule [here](https://github.com/espressif/esp-idf/issues/2343#issuecomment-417853341)

gerekon, 2018-09-14:
@v1993 we are waiting for your project to reproduce the problem.

loceur, 2018-09-23:
hey all, I've narrowed it down in my project to FreeRTOS's Tmr Svc.  I can reproduce this every time when I attempt to get thread info.  I'm running a very bare main.c, and using platformio to setup my build env.

```
[New Thread 1073506928]
[New Thread 1073505540]
[New Thread 1073509584]
[New Thread 1073498260]
[New Thread 1073494992]
[New Thread 1073496788]
[New Thread 1073499392]
[Switching to Thread 1073503936]

Temporary breakpoint 1, main () at src/main.c:408
408	        printf("x");
(gdb) c
+c
Continuing.
Info : Target halted. PRO_CPU: PC=0x400D06DF (active)    APP_CPU: PC=0x400D0C28
Target halted. PRO_CPU: PC=0x400D06DF (active)    APP_CPU: PC=0x400D0C28

Breakpoint 2, main () at src/main.c:408
408	        printf("x");
(gdb) c
+c
Continuing.
Info : Target halted. PRO_CPU: PC=0x400D06DF (active)    APP_CPU: PC=0x400D0C28
Target halted. PRO_CPU: PC=0x400D06DF (active)    APP_CPU: PC=0x400D0C28

Breakpoint 2, main () at src/main.c:408
408	        printf("x");
(gdb) c
+c
Continuing.
Info : Target halted. PRO_CPU: PC=0x400D06DF (active)    APP_CPU: PC=0x400D0C28
Target halted. PRO_CPU: PC=0x400D06DF (active)    APP_CPU: PC=0x400D0C28

Breakpoint 2, main () at src/main.c:408
408	        printf("x");
(gdb) info thread
+info thread
  Id   Target Id         Frame
  8    Thread 1073499392 (dport) 0x40084b36 in vTaskDelete (
    xTaskToDelete=<optimized out>)
    at /Users/wyatt/.platformio/packages/framework-espidf/components/freertos/tasks.c:1316
  7    Thread 1073496788 (ipc0) 0x4008408c in xQueueGenericReceive (
    xQueue=0x3ffc3e7c, pvBuffer=0x0, xTicksToWait=2164427244, xJustPeeking=0)
    at /Users/wyatt/.platformio/packages/framework-espidf/components/freertos/queue.c:1591
  6    Thread 1073494992 (esp_timer) 0x4008408c in xQueueGenericReceive (
    xQueue=0x3ffc2b78, pvBuffer=0x0, xTicksToWait=2164427244, xJustPeeking=0)
    at /Users/wyatt/.platformio/packages/framework-espidf/components/freertos/queue.c:1591
  5    Thread 1073498260 (ipc1) 0x4008408c in xQueueGenericReceive (
    xQueue=0x3ffc443c, pvBuffer=0x0, xTicksToWait=2164427244, xJustPeeking=0)
    at /Users/wyatt/.platformio/packages/framework-espidf/components/freertos/queue.c:1591
  4    Thread 1073509584 (Tmr Svc)
/Volumes/build/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: void inline_frame_this_id(struct frame_info *, void **, struct frame_id *): Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]

This is a bug, please report it.  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.

/Volumes/build/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: void inline_frame_this_id(struct frame_info *, void **, struct frame_id *): Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
```

```
int main()
{
while(1)
    {
        printf("x");
       // vTaskDelay(100/portTICK_PERIOD_MS);
    }
}

void app_main()
{
    main();
    
    
    
}
```

igrr, 2018-09-23:
Could you please pull latest master and try again? If it doesn't work, please attach your sdkconfig.

v1993, 2018-09-23:
@igrr Sorry for waiting so long. There is that archive (clone https://github.com/v1993/esp32-sound and https://github.com/v1993/esp32-sound to `components/`).
[esp32-sound-vorbis-test.zip](https://github.com/espressif/esp-idf/files/2408639/esp32-sound-vorbis-test.zip)

loceur, 2018-09-23:
```
Open On-Chip Debugger 0.10.0-dev (2018-09-20-12:54)
```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2408873/sdkconfig.txt)

igrr, 2018-09-24:
Sorry, I forgot that respective changes have been merged but haven't made it to GitHub yet, as some of the tests are currently failing. I'll ping you to try the master branch again, once the sync happens.

v1993, 2018-10-26:
@igrr Are they merged to master now?

v1993, 2018-10-26:
Well, I guess no… Just updated, still occur.

gerekon, 2018-10-26:
@loceur You example works well for me. I used:

1. IDF's github master
2. Your sdkconfig
3. gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 
4. Open On-Chip Debugger 0.10.0-dev (2018-09-20-12:54)

v1993, 2018-10-26:
@gerekon Still get this bug on real project, master openocd/esp-idf

gerekon, 2018-10-26:
@v1993 Ok. I am tryin to build and run your above project

gerekon, 2018-10-26:
@v1993 Builf process gives a bunch of errors. I flashed the binary from your project. I have no SD card right now at my hands so app fails to mount it and reboots. Could you rebuild you app to use in-memory buffer for vorbis files (I saw such option in menuconfig)?

v1993, 2018-10-26:
Did you clonned repos to components/?

gerekon, 2018-10-26:
@v1993 yes, I cloned [esp32-sound-vorbis](https://github.com/v1993/esp32-sound-vorbis) and [esp32-sound](https://github.com/v1993/esp32-sound).

```
CXX build/esp32-sound-vorbis/soundProviderVorbis.o
In file included from <command-line>:0:0:
/home/alexey/projects/esp/esp32-sound-vorbis-test/components/esp32-sound-vorbis/config/tremor/os.h:25:26: fatal error: ogg/os_types.h: No such file or directory
```

gerekon, 2018-10-26:
`components/esp32-sound-vorbis/ogg` and `components/esp32-sound-vorbis/tremor` are empty. Is it ok?

v1993, 2018-10-26:
You need recursive cloning (`git clone -- recursive`) to download
submodules (ogg and tremor).

gerekon, 2018-10-26:
Ok. Build failed with other errors. See attached log.
[build.log](https://github.com/espressif/esp-idf/files/2520119/build.log)

v1993, 2018-10-26:
You need dev branch from esp32-sound repo.

gerekon, 2018-10-26:
@v1993, I set 3 breakpoints as in your debug log

1. app_main
2. SoundVorbis::SoundProviderVorbis::task_code
3. Sound::SoundMixer::soundCallback

Debugger hits only the first two. Code do not get to Sound::SoundMixer::soundCallback.
When I check backtrace on the second breakpoint via `bt` command I get the following message 
```
[New Thread 1073450092]
[New Thread 1073448192]
[New Thread 1073452748]
[New Thread 1073435788]
[New Thread 1073412772]
[New Thread 1073434400]
[New Thread 1073436920]
[Switching to Thread 1073446076]

Breakpoint 1, app_main () at /home/alexey/projects/esp/esp32-sound-vorbis-test/main/main.cpp:73
73	extern "C" void app_main(void) {
(gdb) c
Continuing.
Target halted. PRO_CPU: PC=0x40102478             APP_CPU: PC=0x40104800 (active)
[New Thread 1073539004]
[Switching to Thread 1073539004]

Breakpoint 2, SoundVorbis::SoundProviderVorbis::task_code (this=0x3ffb9ef8)
    at /home/alexey/projects/esp/esp32-sound-vorbis-test/components/esp32-sound-vorbis/soundProviderVorbis.cpp:132
132		void SoundProviderVorbis::task_code() {
(gdb) c
Continuing.
Target halted. PRO_CPU: PC=0x40082F6E             APP_CPU: PC=0x40104800 (active)
[New Thread 1073436148]
[Switching to Thread 1073436148]

Breakpoint 2, SoundVorbis::SoundProviderVorbis::task_code (this=0x3ffb9ef8)
    at /home/alexey/projects/esp/esp32-sound-vorbis-test/components/esp32-sound-vorbis/soundProviderVorbis.cpp:132
132		void SoundProviderVorbis::task_code() {
(gdb) c
Continuing.
Target halted. PRO_CPU: PC=0x40084C83             APP_CPU: PC=0x40104800 (active)

Breakpoint 2, SoundVorbis::SoundProviderVorbis::task_code (this=0x3ffb9ef8)
    at /home/alexey/projects/esp/esp32-sound-vorbis-test/components/esp32-sound-vorbis/soundProviderVorbis.cpp:132
132		void SoundProviderVorbis::task_code() {
(gdb) bt
#0  SoundVorbis::SoundProviderVorbis::task_code (this=0x3ffb9ef8)
    at /home/alexey/projects/esp/esp32-sound-vorbis-test/components/esp32-sound-vorbis/soundProviderVorbis.cpp:132
#1  0x401166e7 in Sound::SoundProviderTask::taskProviderCode (this=0x3ffb9ef8)
    at /home/alexey/projects/esp/esp32-sound-vorbis-test/components/esp32-sound/soundProviderTask.cpp:53
#2  0x401166f4 in sound_provider_start (arg=0x3ffb9ef8)
    at /home/alexey/projects/esp/esp32-sound-vorbis-test/components/esp32-sound/soundProviderTask.cpp:4
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) q
```
I will look at this error message, but it may turn to be normal since `sound_provider_start` is the task entry function.
So currently I can not reproduce all your conditions. On Monday I will try to run it with SD card.

gerekon, 2018-11-05:
@v1993 Unfortunately. SD card pins interfere with JTAG ones so I failed to test your case on that board using SD card. There is new OpenOCD release is planned (maybe this week). Could you try it?

nospam2000, 2019-01-04:
The bug can also be verified in a standalone configuration without Visual Studio Code, just with gdb and openocd. 

My tool versions:
```
# ~/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gdb --version
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
# ~/esp/openocd-esp32/bin/openocd --version
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:08)
```

gdb crashes in `info threads` when any thread is in an inline function. In my case this is always the `Tmr Svc` thread.

Here the normal part, everythings looks fine so far:
```
iMac:lib michael$ ~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb ~/Documents/PlatformIO/Projects/ESP32DebugTest/.pioenvs/lolin_d32_pro:debug/firmware.elf
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_apple-darwin16.3.0 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /Users/michael/Documents/PlatformIO/Projects/ESP32DebugTest/.pioenvs/lolin_d32_pro:debug/firmware.elf...done.
(gdb) set remotetimeout 60
(gdb) target remote | /Users/michael/esp/openocd-esp32/bin/openocd -c "gdb_port pipe" --search /Users/michael/esp/openocd-esp32/share/openocd/scripts -f interface/jlink.cfg -f board/esp32-wrover.cfg
Remote debugging using | /Users/michael/esp/openocd-esp32/bin/openocd -c "gdb_port pipe" --search /Users/michael/esp/openocd-esp32/share/openocd/scripts -f interface/jlink.cfg -f board/esp32-wrover.cfg
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:08)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
esp32 interrupt mask on
Info : No device selected, using first device.
Info : J-Link ARM V8 compiled Nov 28 2014 13:44:46
Info : Hardware version: 8.00
Info : VTarget = 3.274 V
Info : Reduced speed from 20000 kHz to 12000 kHz (maximum).
Info : Reduced speed from 20000 kHz to 12000 kHz (maximum).
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x400D07BB (active)    APP_CPU: PC=0x400E0A8A 
Info : Detected debug stubs @ 3ffc2804 on core0 of target 'esp32'
Info : accepting 'gdb' connection from pipe
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x400E0A8A 
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 23 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 66 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x400E0A8A 
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x400E0A8A 
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 23 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 66 KB
Info : Using flash size 68 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x400E0A8A 
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 23 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 66 KB
Info : Using flash size 24 KB
Error: cpu0: xtensa_read_memory (line 654): DSR (8000CC19) indicates DIR instruction generated an overrun!
Warn : cpu0: Failed reading 8 bytes at address 0x3F400E8B
xTaskCreate (pvTaskCode=0x400d07b0 <blink_task_wrapper(void*)>, pcName=0x3f400e8b "", usStackDepth=768, pvParameters=0x0, uxPriority=5, pvCreatedTask=0x0)
    at /Users/michael/.platformio/packages/framework-espidf/components/freertos/include/freertos/task.h:438
438			return xTaskCreatePinnedToCore( pvTaskCode, pcName, usStackDepth, pvParameters, uxPriority, pvCreatedTask, tskNO_AFFINITY );
[... 42 lines cut ...]
```

Now calling `info thread` causes the crash when trying to display the data from the `Tmr Svc` thread:
```
(gdb) info thread
  Id   Target Id         Frame 
  7    Thread 1073498656 (ipc0) 0x40084308 in xQueueGenericReceive (xQueue=0x3ffc45c8, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /Users/michael/.platformio/packages/framework-espidf/components/freertos/queue.c:1591
  6    Thread 1073496860 (esp_timer) 0x40084308 in xQueueGenericReceive (xQueue=0x3ffc32c4, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /Users/michael/.platformio/packages/framework-espidf/components/freertos/queue.c:1591
  5    Thread 1073500128 (ipc1) 0x40084308 in xQueueGenericReceive (xQueue=0x3ffc4b88, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /Users/michael/.platformio/packages/framework-espidf/components/freertos/queue.c:1591
  4    Thread 1073512476 (Tmr Svc) 
/Volumes/build/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: void inline_frame_this_id(struct frame_info *, void **, struct frame_id *): Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.

A problem internal to GDB has been detected,
[... cut ...]
Quit this debugging session? (y or n) y

This is a bug, please report it.  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.

/Volumes/build/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: void inline_frame_this_id(struct frame_info *, void **, struct frame_id *): Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) y
```
  Michael

nospam2000, 2019-01-05:
A workaround for my problem is to use the following build flags, e.g. set this config in `platformio.ini`:
`build_flags = -g -Og -fno-inline`

Michael

tom-borcin, 2021-03-25:
Hello @v1993, we are closing this issue now. It was resolved by commit: https://github.com/espressif/esp-idf/commit/3ba63a520c91b961a2632ad7385c8bac9039e197

If you think this issue should remain open, please let me know. Thank you!

closed 2021-03-25 by tom-borcin