espressif/esp-idf #2348: IDF Monitor errors

Elektrik1, 2018-08-29:
After latest update idf_monitor errors with this:

`Traceback (most recent call last):
  File "C:/msys32/home/esp/esp-idf/tools/idf_monitor.py", line 770, in <module>
    main()
  File "C:/msys32/home/esp/esp-idf/tools/idf_monitor.py", line 685, in main
    monitor.main_loop()
  File "C:/msys32/home/esp/esp-idf/tools/idf_monitor.py", line 357, in main_loop
    self.handle_serial_input(data)
  File "C:/msys32/home/esp/esp-idf/tools/idf_monitor.py", line 416, in handle_serial_input
    self.console.write_bytes(line + b'\n')
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/tools/miniterm.py", line 63, in write_bytes
    self.byte_output.write(byte_string)
  File "C:/msys32/home/esp/esp-idf/tools/idf_monitor.py", line 737, in write
    data = data.encode('latin-1')
UnicodeDecodeError: 'ascii' codec can't decode byte 0xd0 in position 61: ordinal not in range(128)
make: *** [/home/esp/esp-idf/components/esptool_py/Makefile.projbuild:106: monitor] Error 1
`

Error occurs on AV event:
`ESP_LOGI(BT_AV_TAG, "AVRC remote features %x", rc->rmt_feats.feat_mask);
where rmt_feats.feat_mask is uint32_t value.

negativekelvin, 2018-08-29:
https://stackoverflow.com/questions/26938252/converting-utf-8-to-latin-1-in-python

Elektrik1, 2018-08-29:
Requirements are met.
Sorry, I have debugged it more, actually line that causes it:
`ESP_LOGI(BT_AV_TAG, "AVRC metadata rsp: attribute id 0x%x, %s", rc->meta_rsp.attr_id, rc->meta_rsp.attr_text);

It happens when meta_rsp.attr_text has non ascii symbols (Russian in this particular case).

dobairoland, 2018-08-30:
Thank you @Elektrik1. I'm working on the issue and will get back to you ASAP...

dobairoland, 2018-08-31:
Hi @Elektrik1. The attached patch should solve the issue. Thank you again for reporting it.

[0001-idf_monitor-Fix-Unicode-decoding-on-Windows.txt](https://github.com/espressif/esp-idf/files/2339705/0001-idf_monitor-Fix-Unicode-decoding-on-Windows.txt)

The patch will be pushed to master soon.

Elektrik1, 2018-08-31:
I do confirm it fixes this issue.

Thank you!
Have a nice day.

closed 2018-08-31 by Elektrik1