espressif/esp-idf #2349: Callback when the time is set on SNTP

cperezvinsite, 2018-08-29:
## Environment
ESP32-Wrover-Kit
- Development Kit:     ESP32-Wrover-Kit
- IDF version 698d3baf
- Operating System: MacOS

## Problem Description

There is some actions that depends on time change, for instance the initialize cron jobs. Then I would recommend an event system based on callbacks in sntp component in order to be capable of find out the time is set.

### Expected Behavior
sntp has a funcition to matriculate a callback, then, when the time is set in `SNTP_SET_SYSTEM_TIME_US` or `SNTP_SET_SYSTEM_TIME` the callback is called in order to be able to know when the time is set

### Actual Behavior
There is no callback

### My code
I create this function to be able to matriculate an event 
`sntp.c`
```cpp
void 
sntp_set_time_callback(void (* callback)(time_t *now))
{
  sntp_time_callback = callback;
}
```
Then I call the callback on lines `225` or `235` but just if the callback is set

I already have this code, and if you want I could make a pull request in order to add this feature into ESP-IDF

liuzfesp, 2018-09-04:
Thanks @cperezvinsite for your suggestion.  I guess this is not an feature in original LWIP release, right? In IDF v3.2, we have updated the LWIP to v2.0.3 and the goal is to minimize Espressif specific change unless it's a bug or very important feature. So currently  I'm not sure we will add it in the future.

vonnieda, 2018-09-04:
There's an active PR for this and Espressif has said they will merge it: https://github.com/espressif/esp-idf/pull/1668

X-Ryl669, 2018-11-19:
What about a callback in settimeofday (since SNTP will call it anyway, but any other function that's changing time). IMHO, it's better to hook here because:
1. If you have a CRON like behaviour, you also want to be informed when the user change the time herself (provided you have this option)
2. It does not depend on LWIP or any version of it, so there's much less maintenance to the code
3. You can set your callback to null when the time is first set so it won't be called continuously by the NTP client
4. You don't need to call gettimeofday since `timeval` will already be available (so it's faster)

KonstantinKondrashov, 2019-06-17:
Hi @cperezvinsite. We added this feature in [7e5be1b](https://github.com/espressif/esp-idf/commit/7e5be1b58cab05cb9c45f06d519f8845175938c8). There has been added the `void __attribute__((weak)) sntp_sync_time(struct timeval *tv)()` function which you can define in your code to update time yourself.
Or use the `sntp_set_time_sync_notification_cb()` function to set a callback function to notify about the time synchronization process.
Also you can use `sntp_get_sync_status()` function to get time synchronization status.

Alvin1Zhang, 2019-06-17:
@cperezvinsite Thanks for reporting the issue, and feel free to reopen the issue if you still have the question. Thanks.

closed 2019-06-17 by Alvin1Zhang