espressif/esp-idf #2352: esp_derive_local_mac missing?

tablatronix, 2018-08-30:
## Environment
arduino esp32 staging

## Problem Description
Compile error 

### Steps to repropduce
compile using func
```C++
// <esp_system.h>
esp_err_t esp_derive_local_mac(uint8_t* local_mac, const uint8_t* universal_mac);
```

### Code to reproduce this issue

```cpp
#include <esp_system.h>
uint64_t _chipmacid = 0LL;
esp_efuse_mac_get_default((uint8_t*) (&_chipmacid));
uint64_t _localmacid = 0LL;
esp_derive_local_mac((uint8_t*)(&_localmacid),(uint8_t*)(&_chipmacid));
```

## Debug Logs

```
.pioenvs\esp32doit-devkit-v1\src\GetChipID.ino.cpp.o:(.literal._Z5setupv+0x40): undefined reference to `esp_derive_local_mac'
.pioenvs\esp32doit-devkit-v1\src\GetChipID.ino.cpp.o: In function `setup()':
GetChipID.ino.cpp:(.text._Z5setupv+0xf2): undefined reference to `esp_derive_local_mac'
```

## Other items if possible
https://github.com/espressif/arduino-esp32/issues/1817

zhangyanjiaoesp, 2018-09-04:
@tablatronix I have call this function as you describled above in the IDF code to test, it can compile successfully. Can you provide a complete code ? Maybe you have miss something.

igrr, 2018-09-05:
Noticed that system_api.c defines `esp_derive_mac` but `esp_system.h` header declares `esp_derive_local_mac`, perhaps that's the issue?

zhangyanjiaoesp, 2018-09-05:
It should be the root cause. I will fix it.

closed 2018-09-05 by igrr