espressif/esp-idf #2353: C++ with ESP IDF

germacovitch, 2018-08-31:
How do I use C++ with ESP IDF? I am interesting in developing some functions in C++, but I did not find much information on that. In special, I would like to manage linked lists with vector class.
Any exemple or advise are welcome 
BR
Gilberto

vijaygill, 2018-08-31:
Here's one project on github. Might be great to have another contributor!

https://github.com/PerMalmberg/Smooth

germacovitch, 2018-08-31:
Tanks!!! I will check it!

hak8or, 2018-08-31:
You should be able to just use C++. Granted, you can't use C++ when compiling with specifically GCC in a .c file because by default GCC assumes .c files have only c. You can instead create files with the .cpp file extension, which will pump it through either G++ or GCC with flags indicating it's c++, like [this](https://stackoverflow.com/questions/5853664/whats-the-difference-between-gcc-and-g-gcc-c).

What I tend to do is use the cmake branch and have the main file be c++ while exporting main to the C world via declaring main with:
```c
extern "C" {
   int main(void);
}

closed 2018-09-03 by germacovitch

exocode, 2019-05-18:
Hi @hak8or, I tried that. But when I include `esp_wps.h` the compiling fails.

What can I do to use C++? 

**main.cpp**
```
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_wps.h"

static const char *TAG = "example_wps";
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);

extern "C" void app_main()
{
 printf("Hello");
}
```

**platform.ini**
```
[env:heltec_wifi_kit_32]
platform = espressif32
board = heltec_wifi_kit_32
framework = espidf
monitor_speed = 115200
```

**ERROR**

```

Compiling .pioenvs/heltec_wifi_kit_32/src/main.o
In file included from src/main.cpp:5:0:
/Users/janjezek/.platformio/packages/framework-espidf/components/esp32/include/esp_wps.h:86:1: error: C99 designator 'manufacturer' outside aggregate initializer
}
^
src/main.cpp:10:37: note: in expansion of macro 'WPS_CONFIG_INIT_DEFAULT'
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
^
/Users/janjezek/.platformio/packages/framework-espidf/components/esp32/include/esp_wps.h:86:1: error: C99 designator 'model_number' outside aggregate initializer
}
^
src/main.cpp:10:37: note: in expansion of macro 'WPS_CONFIG_INIT_DEFAULT'
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
^
/Users/janjezek/.platformio/packages/framework-espidf/components/esp32/include/esp_wps.h:86:1: error: C99 designator 'model_name' outside aggregate initializer
}
^
src/main.cpp:10:37: note: in expansion of macro 'WPS_CONFIG_INIT_DEFAULT'
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
^
/Users/janjezek/.platformio/packages/framework-espidf/components/esp32/include/esp_wps.h:86:1: error: C99 designator 'device_name' outside aggregate initializer
}
^
src/main.cpp:10:37: note: in expansion of macro 'WPS_CONFIG_INIT_DEFAULT'
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
^
src/main.cpp:9:20: warning: 'TAG' defined but not used [-Wunused-variable]
static const char *TAG = "example_wps";
^
src/main.cpp:10:25: warning: 'wpsConfig' defined but not used [-Wunused-variable]
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
^
*** [.pioenvs/heltec_wifi_kit_32/src/main.o] Error 1
====================================== [ERROR] Took 13.58 seconds ======================================
The terminal process terminated with exit code: 1
```

rustyeddy, 2019-05-19:
Hello, 

I was just able to turn the blink example into a C++ program, linked, loaded and working!

The code is: http://github.com/mobilerobot-io/esp/blink

I basically changed the blink.c to main.cpp, I extracted the gpio code into a class LED, which is now defined in LED.h and LED.cpp.   I added an additional LED, now I have two independently blinking LED object defined instantiated from the same LED class.

I was pretty easy and works great once, as previous mentioned, once I properly defined:
  extern "C" void main_app();

Allowing the esp-idf runtime to link with the un-mangled C++ version of app_main().

One gotcha I ran into is remembering that 'void app_main()' and 'void app_main(int argc, char **argv)' signatures represent different functions. :).  I defined app_main(int argc, char **argv) with arguments in the header, however the working implementation in main.cc was defined as 'void app_main()' without args.  

Consistency is key, it works with or without arguments, just make sure that the extern signature matches the signature you are using in your program.

exocode, 2019-05-20:
I found a solution for my problem, which is already in a PR waiting for approval. 

https://github.com/espressif/esp-idf/pull/2869