espressif/esp-idf #2355: Looks like IRAM_ATTR doesn't works for c++ lambda

FoxKeys, 2018-09-01:
This code doesn't works as expected:
timer_isr_register( TIMER_GROUP_0, TIMER_0, [](void* obj)IRAM_ATTR{ static_cast<Protection*>
(obj)->ProtectionISR(); }, this, ESP_INTR_FLAG_IRAM, NULL )

Where ProtectionISR() is method of Protection class.

Probably, IRAM addres check failed in esp_intr_alloc_intrstatus()
This is not 100%, but 95% reason. Not 100, because functions return ESP_ERR_INVALID_ARG, that can be returned because of other reasons. But, because if replace lambda to "normal" function - all works fine, i suppose, problem in not working IRAM_ATTR for lambda.

FoxKeys, 2018-09-01:
P.S. More info.
Looks like any sort of section attribute are silently ignored by compiler for lambda (even, attributes  must work ([proof1](http://www.enseignement.polytechnique.fr/informatique/INF478/docs/Cpp/en/cpp/language/lambda.html), [proof2](https://stackoverflow.com/questions/24668870/how-to-put-lambda-in-a-different-section-with-gcc)) for lambdas too)

I have tried things like:

> auto lambda = [](void* pProtection) [[gnu::section(".iram1")]] -> void { static_cast<Protection*>(pProtection)->TaskFunction(); };
> 
> auto lambda = [](void* pProtection) __attribute__((section(".iram0.text"))) -> void { static_cast<Protection*>(pProtection)->TaskFunction(); };

In any combinations of attribute sintax and section names.

xtensa-esp32-elf-objdump always show result like this:

> 400d0f1c <_ZZN10ProtectionC4EP3App10gpio_num_tS2_yENUlPvE0_4_FUNES3_>:
> 400d0f1c:	004136        	entry	a1, 32
> 400d0f1f:	02ad      	mov.n	a10, a2
> 400d0f21:	afefe5        	call8	40080e20 <_ZN10Protection12TaskFunctionEv>

400d0f1c is from 6 .flash.text   0001620d  400d0018  400d0018  000266f0  2**2 section

Unfortunately, i'm not GCC expert, and can't go deeper my self. Hope, anybody can....

P.P.S. - why i need so? Because i want to hide timer ISR inside of my class. And, ISR must have access to private class fields. And, ISR must be in IRAM, of course. Now, i'm using workaround, made fields public and ISR is "normal" function. But, this is ugly in my case.

igrr, 2018-09-03:
Hi @FoxKeys, indeed GCC does not support section attributes on lambdas. However, even if this could be made to work, it still doesn't guarantee that all the functions used to dispatch this lambda call would also be placed into IRAM.

For your original requirement, to handle an interrupt from a member function, the usual pattern would look like this:

```c++
// header file
class Timer{
public:
    Timer();
protected:
    intr_handle_t m_handle;
    static void interrupt_handler_static(void* arg);
    void interrupt_handler();
};

// source file
void Timer::Timer()
{
    timer_isr_register(TIMER_GROUP_0, TIMER_0, &Timer::interrupt_handler_static, this, 0, &m_handle);
}

void IRAM_ATTR Timer::interrupt_handler_static(void* arg)
{
    reinterpret_cast<Timer*>(arg)->interrupt_handler();
}

void IRAM_ATTR Timer::interrupt_handler()
{
    // actual handler
}
```

Note that the functions involved in interrupt handling are not defined inline (in class definition), and IRAM_ATTR are used to place them into the correct section.

FoxKeys, 2018-09-03:
Yes, static method works too. I have found this workaround. Moreover, no need to have proxy function. For some reason, compiler not block access to private fields from static (i'm not c++ expert, so can't explain is this correct or not. But, this is works. And, minus one excess call for ISR (that is important).
Code looks like this:

```
// header file
class Timer{
public:
    Timer();
private:
    int privateField = 0;
    intr_handle_t m_handle;
    static void interrupt_handler_static(void* pTimer);
};

// source file
void Timer::Timer()
{
    timer_isr_register(TIMER_GROUP_0, TIMER_0, &Timer::interrupt_handler_static, this, 0, &m_handle);
}

void IRAM_ATTR Timer::interrupt_handler_static(void* pTimer)
{
    Timer* self = reinterpret_cast<Timer*>(pTimer);
    self->privateField = 1;//NO problem to write to private field...
}
```

FoxKeys, 2018-09-03:
P.S. Maybe, you can check this and write down as recommendation for c++ writers somewhere in examples (for ISR, for RTOS callbacks, etc)?
It was not easy to find this solution for me, at least.

projectgus, 2018-09-04:
> For some reason, compiler not block access to private fields from static method

The critical difference here is that the static function is still a class member. In terms of function signature & calling parameters it behaves exactly as if it wasn't a class member, but the compiler will give it access to the class internals.

The pattern you wrote looks fine to me, but my C++ is a bit rusty also. Although I think that `static_cast` is a better choice for casting from the `void *`, as you want "simple" cast behaviour here.

FoxKeys, 2018-09-04:
Looks like you are right. I'm read more about different c++ type casts and static_cast is more restrictive (but enough in this case).
In my code (unlike of @igrr sample on this page) i'm used static_cast (not because i'm more smart than @igrr, just because copy-pasted from elswere :))
Updated my post.

rwmingis, 2021-12-03:
Hi, 

Thanks for posting these examples igrr.  I have tried your suggestion but using the arduino platform in PlatformIO. (I realise this may be a bit of an ask mentioning Arduino, but i imagine my question is more ESP32 and C++ memory allocation related so not sure where to go next)

I've got stuck because the function I use to attach the interrupt does not allow arguments to be passed to the ISR (ie the 'arg' parameter).  Is there a variant of your example for functions without parameters?  Someting like the below:

`// header file
class InterruptButton{
public:
    InterruptButton();
protected:
    static void readButtonISR_static(VOID);
    void readButtonISR();
};

// source file
void InterruptButton::InterruptButton()
{
    attachInterrupt(digitalPinToInterrupt(m_pin), &InterruptButton::readButtonISR_static, CHANGE);
}

void IRAM_ATTR InterruptButton::readButtonISR_static(VOID)
{
    reinterpret_cast<InterruptButton*>()->readButtonISR();
}

void IRAM_ATTR InterruptButton::readButtonISR()
{
    // actual handler
}`

igrr, 2021-12-03:
@rwmingis in this case you can only have a single instance of InterruptButton class (signleton), since the interrupt handler won't know which InterruptButton instance to call the readButtonISR member on. To be honest this doesn't look very ESP32 specific to me, this is a general issue if the callback (ISR, in your case) doesn't pass any context or pointer along.

If in the ISR you can somehow figure out _which_ of the InterruptButton instances you need to call, for example by reading GPIO status, then you could lift this limitation, to some extent. For example, declare a static array of pointers to InterruptButton instances, one for each GPIO you are using. In the ISR, check the GPIO status and figure out which was toggled. Then find the corresponding InterruptButton instance from the array and call its member function.

rwmingis, 2021-12-03:
Ahhh, good man @igrr.  I totally forgot that static members more reside in the global deffinition of the class rather than in the instance itself.  Yes, you are reight, not ESP specific assuming the the IRAM_ATTR doesn't stop me. 

In my previous version, I had an external ISR in RAM and passed a pointer to that ISR to the object instance during the setup.  The object then sets up the interrupts to call that ISR when event occurs.   When the ISR was called, it then called the readbutton member function of the object, and I don't believe that readButton code sits in the ram even though the actual ISR does.  Was trying to fix that.

Cheers for your help!

Rob

rwmingis, 2022-01-12:
Hi all, just sharing a working example i made using @igrr's suggestion for creating ISR inside of class objects.  It's done by creating a static class handler function common along all instances, which is passed a reference to a particular instance to do the ISR dirtywork.  

It also uses as similar technique to attach an a class member function to a pin change interrupt using std::bind method. 

This was done to make my interrupt driven debounced button library work.  I found the standard non-interrupt (loop driven) button libraries lacking abit, so I hope this helps someone out.

[](https://github.com/rwmingis/InterruptButton)

Thank you @igrr for the lesson/tip.  :)

Cheers!

Rob

technoo10201, 2022-10-08:
> Hi @FoxKeys, indeed GCC does not support section attributes on lambdas. However, even if this could be made to work, it still doesn't guarantee that all the functions used to dispatch this lambda [...]

Thanks for your example, it helped me to solve an issue for a project.
Have a nice day!

T94T, 2023-02-09:
While searching for my problem, I came across this issue.

I am trying to attach a pin interrupt whose ISR is to call `xTaskResumeFromISR` or `xQueueSendFromISR`. The ISR gets called correctly, but the code to execute results in a _core 0 panic_.

Here are the implementation details.

**PlatformIO**: platform = espressif32 @ 6.0.1, framework = arduino, board = esp32dev

**Header file (Worker.h)**
```
#pragma once
#include <Arduino.h>

class Worker {
  public:
    Worker(uint8_t pinExtInterrupt);
    bool startTask(void);
  protected:
    // static wrapper for task creation
    static void  staticRun(void *arg) {
      reinterpret_cast<Worker *>(arg)->run();
    }

    // actual task's logic
    void run(void);

    // static interrupt handler
    static void staticIsrHandler(void* arg);

    // actual interrupt handler
    void isrHandler();
    
    TaskHandle_t _taskHandle = nullptr;
    uint8_t _pinExtInterrupt;
};
```
**Source file (Worker.cpp)**
```
#include "Worker.h"

Worker::Worker(uint8_t pinExtInterrupt) {
  _pinExtInterrupt = pinExtInterrupt;
  pinMode(pinExtInterrupt, INPUT);
}

bool Worker::startTask(void) {
  BaseType_t xReturned = xTaskCreate(staticRun, "Worker", 2048, this, 5, &_taskHandle);
  gpio_set_intr_type(static_cast<gpio_num_t>(_pinExtInterrupt), GPIO_INTR_NEGEDGE);
  gpio_install_isr_service(0);
  gpio_isr_handler_add(static_cast<gpio_num_t>(_pinExtInterrupt), staticIsrHandler, NULL);
  return true;
}

void Worker::run(void) {
  for (;;) {
    vTaskSuspend(NULL);
    // LOGIC
  }
}

void IRAM_ATTR Worker::staticIsrHandler(void* arg) {
  reinterpret_cast<Worker*>(arg)->isrHandler();
}

void IRAM_ATTR Worker::isrHandler() {
  xTaskResumeFromISR(_taskHandle); // ###### THIS LINE THROWS THE EXCEPTION ######
}
```

**Error**
 _Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
 0x400d1d00:0x3ffbeaac in Worker::isrHandler() at ..._

But what works is if you replace `xTaskResumeFromISR` e.g. with `digitalWrite(..)`.

rwmingis, 2023-02-09:
Hi T94T.  I think I ran into a similar problem when developing my interrupt button library. (this one: https://github.com/rwmingis/InterruptButton/blob/main/InterruptButton.cpp)

I think my solution was to not use the resumeFromISR bit.   From memory, I think i added a task to a queue in the ISR, and then let the normal handler do the queue stuff using the normal vTaskResume instead.

It has been 6 months and I am only an intermediate, so I may be completely off the mark.

Cheers.

T94T, 2023-02-10:
@rwmingis Thanks for you quick reply. I don't really get the idea of you proposal.

T94T, 2023-02-10:
Fixed now by replacing `NULL` with the `this` pointer:
```C++ 
gpio_isr_handler_add(static_cast<gpio_num_t>(_pinExtInterrupt), staticIsrHandler, this);
```

Also changed the `isrHandler` logic to:
```C++
void IRAM_ATTR Worker::isrHandler() {
  if(_taskHandle != NULL) {
    xTaskResumeFromISR(_taskHandle);
  }
}
```