espressif/esp-idf #2356: Make menuconfig not works on Ubuntu

augustocadini, 2018-09-01:
I'm follow this tutorial: https://dl.espressif.com/doc/esp-idf/latest/get-started/index.html#connect

But when I enter on the project folder and type `make menuconfig` apparentely the path is not configured, but when I type printenv I can see the paths.

[image]
I did followed all steps on the espressif official tutorial.
ps. Putty not works too, but in the Arduino windows I Can up firmware.

negativekelvin, 2018-09-01:
IDF_PATH should be one directory only and PATH should not have entry for project.mk

augustocadini, 2018-09-01:
Sorry, I don't got it. DO you can help me?

negativekelvin, 2018-09-01:
export IDF_PATH=/home/augusto/esp/esp-idf

augustocadini, 2018-09-01:
I updated, my comment, I don't known which file I must edit

projectgus, 2018-09-04:
If you use bash then you can put it in either .bashrc or .profile

However the setup guide recommends .profile. The full details are in the Getting Started guide for Linux:
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/linux-setup.html#toolchain-setup
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/add-idf_path-to-profile.html#add-idf-path-to-profile-linux-macos

Note that after editing .profile you will need to either log out and back in, or `source ~/.profile` in the current terminal, to get the new values.

closed 2018-09-11 by FayeY