espressif/esp-idf #2359: Zero reason code in case of wrong password while WiFi connection

laszlopataki, 2018-09-03:
## Environment

- Development Kit: ESP32-DevKitC
- Core (if using chip or module): ESP32
- IDF version : //a557e8cc
- Development Env:		Make
- Operating System:		Windows
- Power Supply:         USB

## Problem Description

Wrong reason code in SYSTEM_EVENT_STA_DISCONNECTED event when the pw is wrong.
The aim would be to detemine quick and respond to the user this case.
Now the only possiblity which I can see is to counting the number of '0' (zero) reason codes.

### Expected Behavior

Getting the correct reason codes in the event in case of the wrong password.

### Actual Behavior

The reason code many times just 0. (zero)

### Steps to reproduce

1. Turn on your PC's hotspot
2. Print the reason code in the disconnect event. (see example)
3. Config the 'scan' example with a wrong pw
4. Test it in different cases
    4.1 ESP boots and tries to connect with the wrong pw
    4.2 ESP is connected but the pw has changed after that
    
### Code to reproduce this issue (scan example + 1 row)

```cpp
/* Scan Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

/*
    This example shows how to use the All Channel Scan or Fast Scan to connect
    to a Wi-Fi network.

    In the Fast Scan mode, the scan will stop as soon as the first network matching
    the SSID is found. In this mode, an application can set threshold for the
    authentication mode and the Signal strength. Networks that do not meet the
    threshold requirements will be ignored.

    In the All Channel Scan mode, the scan will end only after all the channels
    are scanned, and connection will start with the best network. The networks
    can be sorted based on Authentication Mode or Signal Strength. The priority
    for the Authentication mode is:  WPA2 > WPA > WEP > Open
*/
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"

/*Set the SSID and Password via "make menuconfig"*/
#define DEFAULT_SSID CONFIG_WIFI_SSID
#define DEFAULT_PWD CONFIG_WIFI_PASSWORD

#if CONFIG_WIFI_ALL_CHANNEL_SCAN
[... 274 lines cut ...]
I (88630) scan: SYSTEM_EVENT_STA_DISCONNECTED
reason: 0
>>> Turn off Hotspot
I (88750) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (88750) wifi: state: init -> auth (b0)
I (88760) wifi: state: auth -> assoc (0)
I (88760) wifi: state: assoc -> run (10)
I (92910) wifi: state: run -> init (8c0)
I (92910) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (92920) scan: SYSTEM_EVENT_STA_DISCONNECTED
reason: 8 /* OK */
I (93040) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (93040) wifi: state: init -> auth (b0)
I (94040) wifi: state: auth -> init (200)
I (94040) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (94040) scan: SYSTEM_EVENT_STA_DISCONNECTED
reason: 2 /* OK */
I (96460) scan: SYSTEM_EVENT_STA_DISCONNECTED
reason: 201 /* OK */
I (98870) scan: SYSTEM_EVENT_STA_DISCONNECTED
reason: 201
I (101280) scan: SYSTEM_EVENT_STA_DISCONNECTED
reason: 201

```

liuzfesp, 2018-09-18:
@laszlopataki this is our known issue because of some historic reasons. We will add the 4-handshake timeout reason if the password is wrong or if 4-handshake fails in future release.

laszlopataki, 2018-09-19:
Hi @liuzfesp,

Thanks for the response.
Please let me know if you have any known release date or version number.

Regards

liuzfesp, 2018-09-25:
Hi @laszlopataki, I hope we can complete it in IDF v3.1/3.2 bugfix release and v4.0, I can't give you the exact date and it should be done in October or November.

zhangyanjiaoesp, 2018-10-12:
@laszlopataki I have tested the scan example as what you said. But I didn't reproduce your problem, I only get the reason 204 or 201. 
I find that when the reason is 0, the state change of the wifi is `wifi: state: run -> init (c0)` in your log, Then I check the wifi code, I find when the value is c0, it means that sta receive a deauth packet, so please check whether your AP has send deauth packet to station.

enricop, 2019-09-23:
Hi,

in our project we have to alert the user if the Wifi credentials are wrong.
We are working with the `release/v4.0` branch.

When connecting to the right SSID with the wrong password we are receiving `REASON 201` (`WIFI_REASON_NO_AP_FOUND`) on the `WIFI_EVENT_STA_DISCONNECTED` event.

Sholdn't it be `WIFI_REASON_AUTH_FAIL` ?
Can you fix this?

Thanks for your support

zhangyanjiaoesp, 2020-07-13:
hi @enricop,
 I have tested on the least release/v4.0 branch. 
When the AP is open, but the STA uses a password to connect the AP, the disconnect event reason is `WIFI_REASON_NO_AP_FOUND(201)`. 
When the AP is encrypted, and the STA uses a wrong password to connect the AP, the disconnect event reason is `WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT(15)`.

Alvin1Zhang, 2020-07-14:
Thanks for reporting and sharing updates. Feel free to reopen if the issue still happens, or please help file a new ticket for new issue, thanks.

closed 2020-07-14 by Alvin1Zhang