espressif/esp-idf #2363: Reversed RGB led GPIO in the documentation for the wrover kit ?

Navis-Raven, 2018-09-03:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4.1]
- Core (if using chip or module): [ESP32-Wrover]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[other]
- Operating System:		[Window]
- Power Supply:         [USB]

## Problem Description

It seems that the following table in the documentation have reversed led pins

https://docs.espressif.com/projects/esp-idf/en/latest/get-started/get-started-wrover-kit.html#related-documents

[image]

In my wrover kit when I put LED_BUILTIN 4 it's the blue led which blinks. and when I put LED_BUILTIN 2
it's the green led which blinks. It seems that in the documentation the blue and green are reversed. I have another question, how to make the LED D7 blinks ?

Code:
```
#include <Arduino.h>

#ifndef LED_BUILTIN
  // Set LED_BUILTIN if it is not defined by Arduino framework
  #define LED_BUILTIN 4

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  // wait for a second
  delay(1000);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
   // wait for a second
  delay(1000);
}
````

krzychb, 2018-09-04:
Hi @Navis-Raven,

Thank you for noting this issue. I will correct it in documentation.
The getting started guide you refer to is for V3 of the ESP-WROVER-KIT.
Similar document for V4.1 of the ESP-WROVER-KIT is in review and should be released soon.

closed 2018-09-11 by igrr

Navis-Raven, 2018-09-11:
excellent