espressif/esp-idf #2364: python command not found

Oseihie, 2018-09-03:
Hi,
i have been trying to set up the bare metal development toolchain for windows as directed on this website: https://exploreembedded.com/wiki/Setting_up_Bare_Metal_Development_Toolchain_for_ESP32
and it has worked so far until the last point where you compile the project.
I get the error Python: command not found. i have attached a screenshot of the error message.

[image]

Please, any solution to this?

chegewara, 2018-09-03:
You have to start msys from file name: mingw32.exe not msys.exe or mingw64.exe

projectgus, 2018-09-04:
I think following @chegewara's advice will fix this, but if not then please try following the official setup guide and then let us know if you still experience problems:

closed 2018-09-04 by projectgus