espressif/esp-idf #2365: ESP-32 WROVER KIT what are the role of the LED ? How to male D7 blinks ?

Navis-Raven, 2018-09-04:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP32-Wrover]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[VSCode]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

The ESP-WROVER-KIT starter guide describe that D7 (which is a LED on the board) can be toggled by switching GPIO35 but it does not work

https://github.com/espressif/esp-idf/blob/master/docs/en/get-started/get-started-wrover-kit.rst

### Expected Behavior
I think that it is a documentation issue. The documentation does not precise what are the role the LED. And whatever, the LED 7 should blinks if it is GPIO35 according to the documentation. 
[image]

Furthemore there is no blinky code example on the web (except for the RGB LED)

### Actual Behavior
Nothing happens

### Code to reproduce this issue

```cpp

#include <Arduino.h>

#ifndef LED_BUILTIN
  // Set LED_BUILTIN if it is not defined by Arduino framework
  #define LED_BUILTIN 7
#endif

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  // wait for a second
  delay(1000);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
   // wait for a second
  delay(1000);
}

```

me-no-dev, 2018-09-04:
D7 is DATA BIT 7 for the Camera interface. RGB LEDs are defined further down https://github.com/espressif/esp-idf/blob/master/docs/en/get-started/get-started-wrover-kit.rst#rgb-led Where in the documentation did you see that GPIO35 is connected to LED?

projectgus, 2018-09-04:
I think the root cause is that the red power LED on the board has component name D7. (Diode 7.)

Unfortunately the D7 power LED and the D7 on the camera connector are different D7s.

Navis-Raven, 2018-09-04:
@me-no-dev you see this D7 directly on the board 
@projectgus ok so, there is a single name for two different things. So the problem is from the electronic design. At least a note in the documentation that precise to not to confuses these things is highly recommandable.

Navis-Raven, 2018-09-04:
@projectgus so the only one drivable led is the RGB isn't it ? 

As for today there is nowhere in any documentation, that precise the rôle of each LED: D5 to D12

The documentation guide should be updated to include these informations.

me-no-dev, 2018-09-04:
Functionality of those leds is shown in the schematic of the board. While I can understand the need for better silk screen maybe, general rule of thumb is that unmarked LEDs are unaccessible. You can guess the functionality of the LEDS (if schematic is not provided) by their position. D9-12 for example are right next to the FTDI chip and D7 is close to the power.

[image]
[image]

Navis-Raven, 2018-09-04:
@me-no-dev 

> general rule of thumb is that unmarked LEDs are unaccessible

Admitting this rule, a table which explicit the rôle of each of these unaccessible LED would be welcomed. 
And also a note that warns the user to not confuse the D7 (or any other Dx) on the silkscreen and the other D7. I don't know how is in charge of the wrover kit tutorial, it's juste one table and one phrase. Please notice, that as of today, the only documentations of the Wrover Kit is the pdf schematic above, and the getting started guide there is nothing else, no description document nor bill of material or gerber files. I guess that it is because it is still a new product...

krzychb, 2018-09-04:
Hi @Navis-Raven,
Thank you for taking time to document issues you had with documentation. I will address them in upcoming release of Getting Started Guide for V4.1 of the ESP-WROVER-KIT. The document you refer to is for V3 of this board.

closed 2018-09-11 by igrr

SDS-Emb, 2018-09-11:
What is this ? 
[image]

krzychb, 2018-09-12:
Hi @SDS-Emb, this is to hide the older version of documents  in main menu, so orphan files are not reported, while keeping them available for linking in other places. These files are accessible under [Previous Versions of ESP32 Modules and Boards](https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/modules-and-boards-previous.html#previous-versions-of-esp32-modules-and-boards)