espressif/esp-idf #2368: Issue with blufi example on feature/cmake

lucascoxBAF, 2018-09-04:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Core (if using chip or module): ESP-WROOM-32D
- IDF version: e7faa1d8
- Development Env:	 CMAKE
- Operating System: Ubuntu
- Power Supply: USB

## Problem Description

I have been trying to use the feature/cmake branch for some initial development on a project. I tried running the bluefi example (in esp-idf/examples/bluetooth/blufi) and got the following output: [notWorking.txt](https://github.com/espressif/esp-idf/files/2349582/notWorking.txt). It appears the wifi stack gets up and running perfectly fine, but the bluetooth stack doesn't fully initialize. 

I then tried the same example on master. It seems to work quite well. See the output: 
[working.txt](https://github.com/espressif/esp-idf/files/2349598/working.txt)

### Expected Behavior
    Both bluetooth and wifi get up and running

### Actual Behavior
    Wifi appears to work, but bluetooth gets stuck initializing

### Steps to repropduce

1. checkout feature/cmake
2. run idf.py flash monitor
3. Look at the monitor output

projectgus, 2018-09-05:
> Note: on master, I know for certain that bluetooth is advertising and I know for certain that on feature/cmake it is not advertising.

I'm not sure if this is a bug in CMake build or in the feature/cmake branch. We have an updated feature/cmake which we plan to merge to master in the next week or so, and this bug doesn't seem to occur there. Sorry I don't have better answer apart from "sit tight and this should be fixed" :).

> Another note: on the cmake branch, the app size is 981328 bytes, and on master it is 852848 bytes. Any idea why there might be a ~130k size difference?

On master branch, the blufi demo now has a `sdkconfig.defaults` file. These example-specific config settings strip out a number of unneeded Bluetooth features to shrink the final binary size.

> Yet another note: I noticed there have been a ton of change to bt since the last time master was merged into feature/cmake. Do you think merging master in may fix the issue?

It does seem that merging master does fix. I don't recommend trying this by hand, as there are a lot of changes! We should be updating master with cmake support quite soon.

lucascoxBAF, 2018-09-05:
Thank you for the quick reply! I'll wait for the merge.

closed 2018-09-11 by FayeY