espressif/esp-idf #2370: WiFi disconnect and lockup in "wifi" task.

vonnieda, 2018-09-04:
## Environment

- Development Kit: Custom Board
- Kit version (for WroverKit/PicoKit/DevKitC): n/a
- Core (if using chip or module): ESP32-Wrover
- IDF version: release/3.0.3
- Development Env: Make
- Operating System: MacOS
- Power Supply: external 3.3V

## Problem Description

Board suddenly disconnected from WiFi and the "wifi" task consumed all CPU, requiring a power cycle.

```
I (684285) wifi: bcn_timout,ap_probe_send_start
I (686785) wifi: ap_probe_send over, resett wifi status to disassoc
I (686785) wifi: state: run -> init (1)
I (686795) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (686795) wifi: pm stop, total sleep time: 215792953/678005413

S (686805)  xx_wifi.c:224  (0x0000) SYSTEM_EVENT_STA_DISCONNECTED ssid "xx", bssid 80:2a:a8:03:0d:5b, reason WIFI_REASON_BEACON_TIMEOUT.
E (686815)  xx_mqtt.c:258  (0xffffffff) MQTTYield error
W (686845) wifi: Haven't to connect to a suitable AP now!
S (686865)  xx_wifi.c:331  (0x0000) WiFi connecting to "xx".
I (686915)  xx_mqtt.c:232  (0x0000) WiFi not connected, waiting for it.
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
```

The above watchdog messages just continued for several minutes until I power cycled the board. 

This is the first time that I've seen this issue, but there have been a handful of instances of the device locking up previously, so I do wonder if this could be related.

At the time that the code emitted the message "WiFi connecting to "xx"" it would have been calling esp_wifi_connect().

Thanks,
Jason

NOA-Vernon, 2018-09-05:
Yes, this issues always exist until the newest IDF. I tested it just now, it still happened again. In my case, I use Bluetooth and wifi coexist and enable the coexist item in the menuconfig option. following is my looping log:

Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE0 (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE1
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE0 (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE1

any suggestion or point out the problem is appreciated.
Thanks,
Vernon

vbvchauthmal, 2018-09-09:
Facing same issue after updating my esp-idf version from **release/v3.0** (commit id : [0e640c61bd10ac848fc7114199fc7b30381b5332](https://github.com/espressif/esp-idf/tree/0e640c61bd10ac848fc7114199fc7b30381b5332)) to **release/v3.1** (commit id : [d91c425178d4e37b72178c231b5cc50b7ab157e2](https://github.com/espressif/esp-idf/tree/d91c425178d4e37b72178c231b5cc50b7ab157e2)). With earlier version 3.0, didn't faced this issue.
Currently WiFi/Bluetooth performance preference is set to Balance in menuconfig. Tried executing wifi task on core 1, but didn't solve this issue. 

**Its looping with below log, until I reset the board.**
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: wifi
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: wifi
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: wifi
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: wifi
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: wifi

Thanks,
Vaibhav

NOA-Vernon, 2018-09-11:
Is there any updates on this thread?

sebirdman, 2018-09-12:
I hit this problem today when i updated from 3.0 to 3.1 as well

Currently my workaround is to set the power management to 

`    
ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_MAX_MODEM));
`

and setting the listen value very very low.

```
wifi_config_t sta_config = {
    .sta = {
        .scan_method = WIFI_AUTH_WPA2_PSK,
        .sort_method = WIFI_CONNECT_AP_BY_SIGNAL,
        .threshold.rssi = -127,
        .threshold.authmode = WIFI_AUTH_WPA2_PSK,
        .listen_interval = 3, // this is the main setting, 3 is lowest
    }};

```

I tried just using power save none (and the default MIN option) but neither of those seemed to properly solve the issue.

sebirdman, 2018-09-12:
I've been trying to change my settings to more closely match the example https://github.com/espressif/esp-idf/blob/release/v3.1/examples/wifi/power_save/sdkconfig.defaults to see if any of those options fixed it (i figure, the examples can't be exhibiting this or it would have been fixed earlier right?)

I've been running for about an hour now (longer than ever before) after changing my CPU to 80mhz

obviously this is less than ideal, but hopefully this helps the espressif devs figure out what's going on.

@NOA-Vernon @vonnieda @vbvchauthmal if you change your CPU to:

CONFIG_ESP32_DEFAULT_CPU_FREQ_80=y

does this problem still occur?

(mine was at CONFIG_ESP32_DEFAULT_CPU_FREQ_160 by default)

NOA-Vernon, 2018-09-13:
@sebirdman Yes, that did not happen again in 2 hours after I changed the CPU to 80 MHz, But the I2S output seems not fast enough after changed and the sound sounded not clear. why 80 MHz runs normally and that problems always occurred while CPU runs at 160MHz and 240MHz?

sebirdman, 2018-09-13:
@NOA-Vernon Sounds like your application needs the cpu to be higher than 80MHz. I'm not sure exactly why this bug isn't happening at 80MHz (since the code isn't readily available for the wifi part of the esp32) however, I can speculate (this might be very wrong...):

When running a FreeRTOS task, you have to yield to other tasks occasionally with something like:

`    vTaskDelay(100 / portTICK_PERIOD_MS);`

without this, the WatchDog timeout will trigger and you'll get a message like:

```
Tasks currently running:
CPU 0: IDLE
CPU 1: wifi
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
```

You can actually cause this output yourself, setup a FreeRTOS task with a loop and don't do any kind of vTaskDelay and you'll get something like:

```
Tasks currently running:
CPU 0: IDLE
CPU 1: <YOUR TASK NAME>
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
```

NOA-Vernon, 2018-09-14:
@sebirdman tha maximum value of CONFIG_TASK_WDT_TIMEOUT_S tent to be 30 seconds, I attemp to increase this timeout to 60 secnods, but it does't work and the interval of print out the wdt timeout log is 30 seconds,  the watchdog got triggered always even whaterver the value of CONFIG_TASK_WDT_TIMEOUT_S to be set.

Task watchdog got triggered. The following tasks did not reset the watchdog in time:[2018-09-14 11:05:35.078]
 - IDLE (CPU 0)[2018-09-14 11:05:35.078]
Tasks currently running:[2018-09-14 11:05:35.078]
CPU 0: wifi[2018-09-14 11:05:35.078]
CPU 1: IDLE[2018-09-14 11:05:35.078]
Task watchdog got triggered. The following tasks did not reset the watchdog in time:[2018-09-14 11:06:05.077]
 - IDLE (CPU 0)[2018-09-14 11:06:05.077]
Tasks currently running:[2018-09-14 11:06:05.077]
CPU 0: wifi[2018-09-14 11:06:05.077]
CPU 1: IDLE[2018-09-14 11:06:05.077]
Task watchdog got triggered. The following tasks did not reset the watchdog in time:[2018-09-14 11:06:35.077]
 - IDLE (CPU 0)[2018-09-14 11:06:35.077]
Tasks currently running:[2018-09-14 11:06:35.077]
CPU 0: wifi[2018-09-14 11:06:35.077]
CPU 1: IDLE[2018-09-14 11:06:35.077]
Task watchdog got triggered. The following tasks did not reset the watchdog in time:[2018-09-14 11:07:05.078]
 - IDLE (CPU 0)[2018-09-14 11:07:05.078]
Tasks currently running:[2018-09-14 11:07:05.078]
CPU 0: wifi[2018-09-14 11:07:05.078]
CPU 1: IDLE[2018-09-14 11:07:05.078]
Task watchdog got triggered. The following tasks did not reset the watchdog in time:[2018-09-14 11:07:35.078]
 - IDLE (CPU 0)[2018-09-14 11:07:35.078]
Tasks currently running:[2018-09-14 11:07:35.078]
CPU 0: wifi[2018-09-14 11:07:35.078]
CPU 1: IDLE[2018-09-14 11:07:35.078]

sebirdman, 2018-09-14:
@NOA-Vernon drat, i think we will have to wait for espressif to respond with a real fix then.

NOA-Vernon, 2018-09-18:
Hi @liuzfesp , just do the steps you provided above and the unexpected event occurred again, following is the log regarding "Task WDT" issue. 

Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Aborting.
abort() was called at PC 0x400d3483 on core 0

Backtrace: 0x400955a0:0x3ffc05b0 0x40095777:0x3ffc05d0 0x400d3483:0x3ffc05f0 0x40082731:0x3ffc0610 0x4008ee3e:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5728
ho 0 tail 12 room 4
load:0x40078000,len:7684
load:0x40080000,len:7392
entry 0x4008039c
[0;32mI (30) boot: ESP-IDF v3.2-dev-57-ge75628b 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 16:42:21[0m
[0;32mI (31) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) boot: SPI Speed      : 40MHz[0m
[0;32mI (41) boot: SPI Mode       : DIO[0m
[0;32mI (45) boot: SPI Flash Size : 16MB[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (67) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (75) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (82) boot:  3 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (90) boot:  4 ota_0            OTA app          00 10 00210000 00200000[0m
[0;32mI (97) boot:  5 ota_1            OTA app          00 11 00410000 00200000[0m
[0;32mI (105) boot: End of partition table[0m
[0;32mI (109) boot: Defaulting to factory image[0m
[0;32mI (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x718f4 (465140) map[0m
[0;32mI (285) esp_image: segment 1: paddr=0x0008191c vaddr=0x3ffc0000 size=0x05f98 ( 24472) load[0m
[0;32mI (295) esp_image: segment 2: paddr=0x000878bc vaddr=0x3ffc5f98 size=0x00000 (     0) load[0m
[0;32mI (296) esp_image: segment 3: paddr=0x000878c4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (304) esp_image: segment 4: paddr=0x00087ccc vaddr=0x40080400 size=0x08344 ( 33604) load[0m
[0;32mI (326) esp_image: segment 5: paddr=0x00090018 vaddr=0x400d0018 size=0x11a364 (1155940) map[0m
[0;32mI (731) esp_image: segment 6: paddr=0x001aa384 vaddr=0x40088744 size=0x10374 ( 66420) load[0m
[0;32mI (759) esp_image: segment 7: paddr=0x001ba700 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (759) esp_image: segment 8: paddr=0x001ba708 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (781) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (781) boot: Disabling RNG early entropy source...[0m
[0;32mI (782) cpu_start: Pro cpu up.[0m
[0;32mI (785) cpu_start: Starting app cpu, entry point is 0x400811f4[0m
[0;32mI (776) cpu_start: App cpu up.[0m
[0;32mI (796) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (803) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (809) heap_init: At 3FFD8670 len 00007990 (30 KiB): DRAM[0m
[0;32mI (815) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (821) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (827) heap_init: At 40098AB8 len 00007548 (29 KiB): IRAM[0m
[0;32mI (834) cpu_start: Pro cpu start user code[0m
[0;32mI (181) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m

PegasYura, 2018-09-21:
Hello
I also encountered this problem.
My log

> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> - IDLE (CPU 0)
>Tasks currently running:
>CPU 0: wifi
>CPU 1: IDLE
>Aborting.
>abort() was called at PC 0x400d349f on core 0
>0x400d349f: task_wdt_isr at /home/osboxes/esp/esp-idf/components/esp32/task_wdt.c:236
>
>Backtrace: 0x400975fa:0x3ffc0580 0x40097788:0x3ffc05b0 0x400d349f:0x3ffc05e0 >0x40082ac1:0x3ffc0610 0x4009036b:0x00000000
>0x400975fa: invoke_abort at /home/osboxes/esp/esp-idf/components/esp32/panic.c:653
>
>0x40097788: abort at /home/osboxes/esp/esp-idf/components/esp32/panic.c:653
>
>0x400d349f: task_wdt_isr at /home/osboxes/esp/esp-idf/components/esp32/task_wdt.c:236
>
>0x40082ac1: _xt_lowint1 at /home/osboxes/esp/esp-idf/components/freertos/xtensa_vectors.S:1105
>
>0x4009036b: coex_force_wifi_mode at ??:?
>
>
>E (5630) esp_core_dump: Skipped 1 tasks with bad TCB!
>E (5630) esp_core_dump: Crashed task has been skipped!
>Rebooting...

I can provide "сoredump". Can this help you?

liuzfesp, 2018-09-25:
@PegasYura @davctv @NOA-Vernon I think your issues are same as the one we are debugging. @TianHao-Espressif now is debugging it.

jkhax0r, 2018-10-01:
@liuzfesp @TianHao-Espressif Same issue here.  Any updates?

antegulin, 2018-10-01:
I'm facing the same issue, it's a roadblock for me. Really looking forward to any progress on this!

PegasYura, 2018-10-11:
Hello, is there any news?

sebirdman, 2018-10-15:
@liuzfesp @TianHao-Espressif any update or workaround besides clocking to 80Mhz? This is very critical to our application.

kluverp, 2018-10-16:
I'am getting the same coredump as @PegasYura. I'am also on IDF version 3.1.
It seems to happen when a WiFi connection is 'disconnected'. But the WiFi library seems to 'stop' the WiFi. Then the 'Disconnected' event fires, and tries to connect again, as in the example. But then an error occurs: 'wifi not started'. Then we start WiFi, and the system crashes (Panic Handler starts and reboots system).

TianHao-Espressif, 2018-10-25:
related to https://github.com/espressif/esp-idf/issues/2489

It's resolved.

Milan991, 2018-10-25:
I just tested it with lattest ESP-IDF commit. I keep getting:

W (34833) EM: STA DISCONNECTED
I (34833) EM: SYSTEM_EVENT_STA_START NOW
W (37253) EM: STA DISCONNECTED
I (37253) EM: SYSTEM_EVENT_STA_START NOW
.
.
.
W (39693) EM: STA DISCONNECTED
I (39693) EM: SYSTEM_EVENT_STA_START NOW

In loop, it just can't connect to the WiFi. And with older version of ESP-IDF same source code connect successfully.

TianHao-Espressif, 2018-11-07:
@Milan991 , do you use BLE or BR/EDR while wifi is connecting?

Milan991, 2018-11-07:
@TianHao-Espressif I'm using BLE.

Milan991, 2018-11-17:
Is there any update on this? Same source code is working fine with older version of ESP-IDF.

NateBowen, 2019-02-01:
We've run into the same issue while running on `v3.0-rc1`. Bluetooth is not enabled.

Here's a log that shows the behavior after the WiFi network is dropped:

```
I (35194876) wifi: ap_probe_send over, resett wifi status to disassoc
I (35194877) wifi: state: run -> init (1)
I (35194878) wifi: pm stop, total sleep time: 0/827807509

I (35194878) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
D (35194886) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:sanitized, ssid_len:13, bssid:0a:18:d6:87:3d:92, reason:200
D (35194897) tcpip_adapter: check: local, if=0 fn=0x4013b564

D (35194902) tcpip_adapter: if0 start ip lost tmr: enter
D (35194907) tcpip_adapter: if0 start ip lost tmr: interval=120
D (35194913) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (35194919) tcpip_adapter: check: remote, if=0 fn=0x4013b564

I (35194925) wifiEventHandler: SYSTEM_EVENT_STA_DISCONNECTED - reason: 200
I (35195033) networkConnected: wifi disconnected - retrying network connect
```

The last line in that log represents a call to `esp_wifi_connect()`. Five seconds later the watchdog fires:

```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
```

At this point, every 5 seconds the watchdog is triggered.

We've updated the IDF to `v3.1` and have enabled the panic handler to trigger on the watchdog. Currently testing to see if we can catch the backtrace.

It's a very rare bug, and we can't seem to force its behavior. Any information on the root cause? Are there any confirmed work-arounds?

liuzfesp, 2019-02-02:
HI @NateBowen for latest IDF v3.1/3.2/3.3, the watchdog issue in WiFi/BT coexist should have been fixed. 

For old v3.0-rc, could you help to check the backtrace:

- Enable panic when WDT happen via menuconfig: make menuconfig => Component config => ESP32-specific => Invoke panic handler on Task Watchdog timeout
- make flash monitor
- When the problem is reproduced, paste the crash log (only the backtrace part) here and I will check whether it's the same issue with our debugging one

NateBowen, 2019-02-02:
@liuzfesp Thank you for getting back so quickly. I'm confused why this would be a WiFi/BT coexist bug when we haven't enabled Bluetooth in the sdkconfig?

liuzfesp, 2019-02-02:
Hi @jinsukseo, you can run following commands to get the backtrace, the *elf is your elf file:
xtensa-esp32-elf-addr2line -piaf -e *elf 0x400938d8:0x3ffbe2b0 0x40093aa9:0x3ffbe2d0 0x400d2767:0x3ffbe2f0 0x400825d5:0x3ffbe310 0x4008e08b:0x00000000

NateBowen, 2019-02-28:
Hi @liuzfesp. We've finally recorded this event since updating to the `v3.1` tag. Unfortunately, this device didn't have the panic handler enabled for the task watchdog.

At bootup, we see the following information for the wifi library:

```
I (483) wifi: wifi driver task: 3ffdd27c, prio:23, stack:3584, core=0
I (488) wifi: wifi firmware version: 54ef8ed
```

And after many hours of running, the following occurs:

```
I (19921486) wifi: bcn_timout,ap_probe_send_start

I (19923989) wifi: ap_probe_send over, resett wifi status to disassoc
I (19923989) wifi: state: run -> init (100)
I (19923990) wifi: pm stop, total sleep time: 18437258724 us / 19758404387 us

I (19923995) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
D (19924000) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:MEAT, ssid_len:4, bssid:14:ae:db:57:c4:fd, reason:200
D (19924012) tcpip_adapter: if0 start ip lost tmr: enter

D (19924015) tcpip_adapter: if0 start ip lost tmr: interval=120 sz: 5

I (19924034) wifiEventHandler: SYSTEM_EVENT_STA_DISCONNECTED - reason: 200
I (19924081) networkConnected: wifi disconnected - retrying network connect

Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
```

Which is the same basic flow as we saw on `v3.0-rc1` above.

liuzfesp, 2019-03-04:
Hi @NateBowen, I think it's not easy for us to analyze the watchdog log without backtrace. Moreover v3.0-rc1 is a very old release, suggest your to upgrade your IDF to latest v3.0 (3.0.7 or 3.0.8), or to v3.1/3.2.

igrr, 2019-03-04:
To add to what @liuzfesp has said, you can get the task watchdog to trigger a restart (printing the backtrace) if you enable [CONFIG_TASK_WDT_PANIC](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-task-wdt-panic). Updating IDF to v3.1.3 might also help.

NateBowen, 2019-03-04:
@liuzfesp I understand the backtrace issue, and I wish I had that information for you. However, I mentioned in my last post that this is happening on `v3.1`, with `wifi: wifi firmware version: 54ef8ed`.

NateBowen, 2019-03-04:
@igrr Is there a known issue in `v3.1`? I've tried updating to `v3.1.2` and `v3.1.2`, but something breaks in our connection to a mobile app.

liuzfesp, 2019-03-04:
HI @NateBowen 54ef8ed still too old for the interrupt watchdog issue, you can update to latest v3.1 (not v3.1.3) to check whether it breaks connection to mobile app.

NateBowen, 2019-03-07:
Hi @liuzfesp. I've been able update to `7e5ac34be` on `release/v3.1` and it looks like the mobile app is happy again. This puts us at `wifi firmware version: 5430168`. I'll enable the task watchdog panic and add come back with that information if we see the issue pop up again.

Alvin1Zhang, 2019-04-19:
@vonnieda @NateBowen Hi, has your issue been resolved? Could you help share if any updates for this issue? Thanks.

NateBowen, 2019-04-23:
Hi @Alvin1Zhang. We haven't seen any issues since my last comment. Thank you.

vonnieda, 2019-04-23:
@Alvin1Zhang I just saw this again yesterday, for the first time in months. Here is what I saw in the logs:

I (2685015) wifi: bcn_timout,ap_probe_send_start
I (2687525) wifi: ap_probe_send over, resett wifi status to disassoc
I (2687525) wifi: state: run -> init (1)
I (2687525) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (2687525) wifi: pm stop, total sleep time: 418304260/-1613067069

I (2687535)  fb_wifi.c:224  (0x0000) SYSTEM_EVENT_STA_DISCONNECTED ssid "xxxxxxxx", bssid fc:ec:da:a4:da:a7, reason WIFI_REASON_BEACON_TIMEOUT.
E (2687555)  xx_mqtt.c:260  (0xffffffff) MQTTYield error
I (2687585)  xx_wifi.c:331  (0x0000) WiFi connecting to "xxxxxxxxxx".
I (2687605)  xx_mqtt.c:234  (0x0000) WiFi not connected, waiting for it.
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi

The wifi task never recovered and I had to restart the device.

liuzfesp, 2019-05-24:
HI @vonnieda, We also saw similar watchdog in our recently release test, but unfortunately we didn't enable that watchdog panic and didn't get the backtrace. We are now reproducing this issue. Will update you once fixing it.

vonnieda, 2019-05-24:
Glad to hear it, thank you!

chegewara, 2019-07-29:
Hi,
i am having the same issue with arduino as component (esp-idf v3.2) when i dont set power mode to NON `esp_wifi_set_ps(WIFI_PS_NONE);` and this is happening in middle of downloading file (sockets are opened and closed all the time before that event).

Here is log:
```
I (220832) wifi: bcn_timout,ap_probe_send_start
I (223332) wifi: ap_probe_send over, resett wifi status to disassoc
I (223342) wifi: state: run -> init (c800)
I (223342) wifi: pm stop, total sleep time: 95335777 us / 217126450 us

I (223342) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (343352) event: station ip lost
```

liuzfesp, 2019-07-30:
Hi @chegewara, could you help to provide following info:

- The AP type to which the ESP32 is connected
- Is this issue easy to be reproduced?
- The detailed steps to replicate this issue
- The demo code to replicate this issue

chegewara, 2019-07-30:
Hi,
i have to repeat tests to confirm that issue and all other. I made a mistake and appended `0x00` in all requests, so it caused some strange problems i got(most likely not related to this one). 

I can provide test code, but it requires to use sipeed maixduino board and this code on esp32:
https://github.com/sipeed/MaixDuino_ESP32_SpiSlave

milamber-ls, 2019-08-10:
is there an update in this? I have the same problem

Octogonapus, 2019-10-18:
Still having this issue on the latest release

liuzfesp, 2019-10-28:
@Octogonapus do you mean still have the task watchdog issue on latest master?

Octogonapus, 2019-10-28:
> do you mean still have the task watchdog issue on latest master?

Not on the latest master. I am using v3.3.0.

liuzfesp, 2019-10-30:
Hi @Octogonapus does your v3.3 IDF contains fix: 2d5ee438, this commit fix a bug that WiFi stop leads to watchdog. For other releases, the fix commits are:

- master - 53d57dd7
- v4.0 - d9cfb05e
- v3.1 - 907878d6
- v3.2 - cdb224c1 (not merged yet)

Octogonapus, 2019-10-30:
I don't think I can check that because I am using PlatformIO, so there is no git repo on disk I can look in. I compared the crc32 hashes of a few of those binaries, though; they are different.

liuzfesp, 2019-11-01:
@Octogonapus that's OK. What I suggest is:

- Keep your git commit ID each time you upgrade your IDF, it's helpful for us to check whether your IDF already contains the fix or NOT
- Always enable watchdog panic via: menuconfig => component config => Common ESP related => Invoke panic handler on Task Watchdog timeout. Once the watchdog panic is enabled, the crash backtrace will be displayed, then we can find out the crash point that causes the task watchdog according to the backtrace and *.elf file. It means you also need to keep the *elf file. Based on our investigation, the task watchdog happens very rare, so you can keep the *elf file if you want to do stress time (long time test).

liuzfesp, 2019-11-26:
Hi @Octogonapus, any update about this issue?

Alvin1Zhang, 2020-01-21:
@vonnieda Thanks for reporting, would you please help try the suggestions and commits by @liuzfesp and help share if the issue gets resolved at your side? Thanks.

vonnieda, 2020-01-22:
@Alvin1Zhang I have not seen this issue in quite some time, so I suspect it was resolved by updating to v3.3.1 at some point.

Alvin1Zhang, 2020-01-29:
Thanks for the updates. Would close the ticket for now, feel free to reopen the issue if it happens again. Thanks.

closed 2020-01-29 by Alvin1Zhang

gadget-man, 2020-07-21:
I'm seeing similar issues on ESP-IDF v3.3-r4 via Mongoose-os. My device connects fine, but after some time (usually 1-3 hours) it gets stuck into a disconnect loop. I'm not seeing the watchdog timeout but the device disconnects with the same logging shown above, then reconnects, gets an IP address but then a few seconds later disconnects again with the same error. Here is a log extract:
`sleep.js:7              Setting auto sleep mode
[0;32mI (43855) pm_esp32: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 10, Light sleep: ENABLED[0m
I (43865) wifi: Set ps type: 2

init.js:504             NETWORK: "GOT_IP"
mgos_event.c:135        ev NET3 triggered 4 handlers
mg_net.c:794            0x3ffef2f8 udp://192.168.0.1:53 -> 0
mgos_mongoose.c:66      New heap free LWM: 4193416
mg_rpc.c:470            0x3ffdd108 CHAN OPEN (loopback)
mgos_event.c:135        ev RPC0 triggered 0 handlers
mg_resolv.c:209         Failed to resolve 'a24buxh69lukw9-ats.iot.eu-west-1.amazonaws.com', server 192.168.0.1
mgos_mqtt_conn.c:227    MQTT0 TCP connect error (-1)
mgos_mqtt_conn.c:253    MQTT0 Disconnect
mgos_event.c:135        ev MOS5 triggered 0 handlers
init.js:511             MQTT Disconnect
init.js:519             Starting MQTT Disconnect Timer: 1073669700
mgos_mqtt_conn.c:546    MQTT0 connecting after 1948 ms
mg_net.c:150            0x3ffe6c14 0x10 1
mg_resolv.c:209         Failed to resolve 'y6tp1eivk3.execute-api.eu-west-1.amazonaws.com', server 192.168.0.1
init.js:963             
mg_net.c:150            0x3ffde208 0x10 1
mg_resolv.c:209         Failed to resolve 'time.google.com', server 192.168.0.1
mgos_sntp.c:78          SNTP close
mg_net.c:150            0x3ffe8b4c 0x802 1
mgos_mqtt_conn.c:468    MQTT0 connecting to a24buxh69lukw9-ats.iot.eu-west-1.amazonaws.com:8883
mgos_event.c:135        ev MOS6 triggered 0 handlers
mg_net.c:911            0x3ffe8b1c a24buxh69lukw9-ats.iot.eu-west-1.amazonaws.com:8883 aws-iParcelBox-30AEA.crt.pem,aws-iParcelBox-30AEA.key.pem,ca.pem
mgos_vfs.c:283          aws-iParcelBox-30AEA.crt.pem -> /aws-iParcelBox-30AEA.crt.pem pl 1 -> 1 0x3ffcdb3c (refs 1)
mgos_vfs_fs_spiffs.:838 aws-iParcelBox-30AEA.crt.pem -> TmToIfHxtOJoQOnpezBpzyPTw9CtOtMkjjTVXB2kNyY=
mgos_vfs.c:377          open aws-iParcelBox-30AEA.crt.pem 0x0 0x1b6 => 0x3ffcdb3c aws-iParcelBox-30AEA.crt.pem 1 => 257 (refs 1)
mgos_vfs.c:536          fstat 257 => 0x3ffcdb3c:1 => 0 (size 1162)
mgos_vfs.c:536          fstat 257 => 0x3ffcdb3c:1 => 0 (size 1162)
mgos_vfs.c:564          lseek 257 0 1 => 0x3ffcdb3c:1 => 0
mgos_vfs.c:564          lseek 257 1024 0 => 0x3ffcdb3c:1 => 1024
mgos_vfs.c:564          lseek 257 1024 0 => 0x3ffcdb3c:1 => 1024
mgos_vfs.c:564          lseek 257 0 2 => 0x3ffcdb3c:1 => 1162
mgos_vfs.c:564          lseek 257 0 0 => 0x3ffcdb3c:1 => 0
mgos_vfs.c:410          close 257 => 0x3ffcdb3c:1 => 0 (refs 0)
mgos_vfs.c:283          aws-iParcelBox-30AEA.key.pem -> /aws-iParcelBox-30AEA.key.pem pl 1 -> 1 0x3ffcdb3c (refs 1)
mgos_vfs_fs_spiffs.:838 aws-iParcelBox-30AEA.key.pem -> TmToIfHxtOJoQOnpezBpz/bqyJjCxE4p31aL00zF8bY=
mgos_vfs.c:377          open aws-iParcelBox-30AEA.key.pem 0x0 0x1b6 => 0x3ffcdb3c aws-iParcelBox-30AEA.key.pem 1 => 257 (refs 1)
mgos_vfs.c:536          fstat 257 => 0x3ffcdb3c:1 => 0 (size 227)
mgos_vfs.c:536          fstat 257 => 0x3ffcdb3c:1 => 0 (size 227)
mgos_vfs.c:564          lseek 257 0 1 => 0x3ffcdb3c:1 => 0
mgos_vfs.c:564          lseek 257 0 0 => 0x3ffcdb3c:1 => 0
mgos_vfs.c:564          lseek 257 0 0 => 0x3ffcdb3c:1 => 0
mgos_vfs.c:564          lseek 257 0 2 => 0x3ffcdb3c:1 => 227
mgos_vfs.c:564          lseek 257 0 0 => 0x3ffcdb3c:1 => 0
mgos_vfs.c:410          close 257 => 0x3ffcdb3c:1 => 0 (refs 0)
mgos_vfs.c:283          ca.pem -> /ca.pem pl 1 -> 1 0x3ffcdb3c (refs 1)
mgos_vfs_fs_spiffs.:838 ca.pem -> O1O6d3Qy2IXXZpXodrJvI3fIjCBzxGJFekH2o9cnQhA=
mgos_vfs.c:377          open ca.pem 0x0 0x1b6 => 0x3ffcdb3c ca.pem 1 => 257 (refs 1)
mgos_vfs.c:410          close 257 => 0x3ffcdb3c:1 => 0 (refs 0)
mg_net.c:911            0x3ffe7f60 udp://192.168.0.1:53 -,-,-
mg_net.c:779            0x3ffe7f60 udp://192.168.0.1:53
mg_net.c:794            0x3ffe7f60 udp://192.168.0.1:53 -> 0
mg_net.c:911            0x3ffef3a8 udp://time.google.com:123 -,-,-
mg_net.c:911            0x3ffe6888 udp://192.168.0.1:53 -,-,-
mg_net.c:779            0x3ffe6888 udp://192.168.0.1:53
mgos_sntp.c:95          SNTP query to time.google.com
mgos_sntp.c:127         SNTP next query in 15908 ms
mg_net.c:794            0x3ffe6888 udp://192.168.0.1:53 -> 0
I (49855) wifi: bcn_timout,ap_probe_send_start
I (52355) wifi: ap_probe_send over, resett wifi status to disassoc
I (52365) wifi: state: run -> init (c800)
I (52365) wifi: pm stop, total sleep time: 765462 us / 9990041 us

I (52375) wifi: new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
mgos_wifi.c:119         WiFi STA: Disconnected, reason: 200
[... 48 lines cut ...]
mg_net.c:150            0x3ffe6b0c 0x10 1
mg_resolv.c:209         Failed to resolve 'a24buxh69lukw9-ats.iot.eu-west-1.amazonaws.com', server 192.168.0.1
mgos_mqtt_conn.c:227    MQTT0 TCP connect error (-1)
mgos_mqtt_conn.c:253    MQTT0 Disconnect
mgos_event.c:135        ev MOS5 triggered 0 handlers
init.js:511             MQTT Disconnect
init.js:519             Starting MQTT Disconnect Timer: 1073602968
mgos_mqtt_conn.c:546    MQTT0 connecting after 4051 ms
mg_net.c:150            0x3ffe8b1c 0x10 1
I (59475) wifi: bcn_timout,ap_probe_send_start
[... cut ...]
I (61985) wifi: ap_probe_send over, resett wifi status to disassoc
I (61985) wifi: state: run -> init (c800)
I (61995) wifi: pm stop, total sleep time: 226405 us / 9171684 us

I (62005) wifi: new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
mgos_wifi.c:119         WiFi STA: Disconnected, reason: 200
mgos_event.c:135        ev WFI0 triggered 0 handlers
mgos_net.c:82           WiFi STA: disconnected
init.js:436             timer_Disconnect:1073602968
init.js:504             NETWORK: "DISCONNECTED"
mgos_event.c:135        ev NET0 triggered 3 handlers
`