espressif/esp-idf #2372: Behaviour on change in IP address no SYSTEM_EVENT_STA_LOST_IP

ammaree, 2018-09-05:
## Environment
- Core (if using chip or module): ESP-WROOM32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): absolute latest from master
- Development Env:		Eclipse
- Operating System:		Windows
- Power Supply:         external 24V to 3.3V

## Problem Description
When deploying devices in the field we go through a process where the device is allocated a DHCP address and, depending on the address allocated we might change the IP address to fit into a certain organisational structure.
We primarily use MikroTik APs managed by CAPSMan so this change is achieved by:
a) converting the dynamic lease into a pseudo "static" lease;
b) changing the assigned "static" IP address to the required address
c) Waiting for the lease time to expire for the IP address to change

### Expected Behavior
We expect to see 2 events at this time being:
a) SYSTEM_EVENT_STA_LOST_IP
b) SYSTEM_EVENT_STA_GOT_IP
If both events are received our code will close all client & server sockets opened with the old address and reopen them with the new address.

### Actual Behavior
All we see is SYSTEM_EVENT_STA_GOT_IP and the application either hangs or spews out a long string of Telnet, Syslog or MQTT error messages

### Steps to reproduce
Just follow step a-c above

### Code to reproduce this issue

## Debug Logs

## Other items if possible

negativekelvin, 2018-09-05:
SYSTEM_EVENT_STA_GOT_IP has an `ip_changed` property.  So you are saying it crashes on dhcp renew?

ammaree, 2018-09-06:
If my assumption is correct the ip_changed flag would indicate the following logic:
a) On first SYSTEM_EVENT_STA_GOT_IP when IP goes from 0/invalid to valid1 the flag should be false
b) On subsequent SYSTEM_EVENT_STA_GOT_IP when IP goes from validX to validY the flag should be true to indicate that it did effectively "change"

If the above assumption is correct then I think we should re-look the code because the flag currently ALWAYS is true.

negativekelvin, 2018-09-06:
That is something to consider whether the flag should be set when going from an invalid address to a valid one.  But I don't know if it should really disrupt application logic since your app should know it doesn't have any connections to tear down.

ammaree, 2018-09-07:
Have added the logic to check if my existing IP is non-zero, in which case the change is a new address as opposed to the first address.

Happy to close the issue, but think the exact reasoning should be clarified in documentation.

closed 2018-09-07 by ammaree