espressif/esp-idf #2374: tcp server can not send keepalive

1163167506, 2018-09-05:
Esp32 can't send keepalive messages as tcpserver?

code:
Int keepAlive = 1;
Int keepIdle = 20;
Int keepInterval = 3;
Int keepCount = 3;

Setsockopt (listenfd, SOL_SOCKET, SO_KEEPALIVE, (void*) &keepAlive, sizeof (keepAlive));
Setsockopt (listenfd, IPPROTO_TCP, TCP_KEEPIDLE, (void*) &keepIdle, sizeof (keepIdle));
Setsockopt (listenfd, IPPROTO_TCP, TCP_KEEPINTVL, (void*) &keepInterval, sizeof (keepInterval));
Setsockopt (listenfd, IPPROTO_TCP, TCP_KEEPCNT, (void*) &keepCount, sizeof (keepCount));

I set esp32 as TCP client to send keepalive messages success.

But setting esp32 as TCP server can't send keepalive messages.

liuzfesp, 2018-09-18:
@1163167506 for TCP server, you need to set the keep alive options to each accepted socket, not set to the listen socket.

closed 2018-09-19 by FayeY

asidika, 2019-07-26:
Hi, currently i m facing with this issue, I could not set keepalive option. 
can anyone share a code snip which enables keep alive socket option?

liuzfesp, 2019-07-30:
Hi @asidika I think @1163167506 already give the demo code.