espressif/esp-idf #2375: event_handler  SYSTEM_EVENT_AP_STADISCONNECTED sometimes not received

francescofcf, 2018-09-05:
**SO:** Windows 10
**IDE:** Eclipse IDE for C/C++ Developers  Version: Photon Release (4.8.0) Build id: 20180619-1200
**environment:** esp32_win32_msys2_environment_and_toolchain-20180110
**ESP-IDF version:**  v3.2-dev-362-g1c7a8b3b
**EspTools :** esptool.py v2.5.0
**Source code :** "Simple WiFi Example" in "esp-idf \ examples \ wifi \ simple_wifi \ main"
**Source Code config :** AP mode
**Client :** Iphone 6s
**Power :** external 3.3V

**DevBoard1 : ESP32 DEVKITV1**
Chip is ESP32D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core
MAC: 24:0a:c4:80:32:e8

**DevBoard2: my board with**
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: 80:7d:3a:82:1c:50

### Hi, I have an problem about event_handler(void *ctx, system_event_t *event)
I have chek the problem also in "Simple WiFi Example" in "esp-idf \ examples \ wifi \ simple_wifi \ main" whitout modify. 
The simple project **"Simple WiFi Example"**  has the same problem!
Below there is the output about "Simple WiFi Example", see output on UART at **“57148 “** and **“127508”**

I have add as first row the below  in event_handler:
`ESP_LOGI(TAG, "EVENT: %d",event->event_id);`
**The problem  is:**
_At startup the  event_handler receive:_
Event 13 : SYSTEM_EVENT_AP_START
_after connected_
Event 15 : SYSTEM_EVENT_AP_STACONNECTED
Event 17 : SYSTEM_EVENT_AP_STAIPASSIGNED
_but after disconnected sometimes (often the first time) the_
Event 16 : SYSTEM_EVENT_AP_STADISCONNECTED
**it is not received.**

_After If I add a task and print this information:_
```
esp_wifi_ap_get_sta_list(&wifi_sta_list);
tcpip_adapter_get_sta_list(&wifi_sta_list, &adapter_sta_list);
```
the old station now disconnected is  given as connected
The procedure that I use to disconnect the station (Apple smartphone) is to turn off the your  wifi.

### Output on UART

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)                         
ets Jun  8 2016 00:22:57                                                      
                                                                            
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)                     
configsip: 0, SPIWP:0xee                                                        
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00         
mode:DIO, clock div:1                                                           
load:0x3fff0018,len:4                                                           
load:0x3fff001c,len:5916                                                        
load:0x40078000,len:9076                                                        
load:0x40080400,len:6140                                                        
entry 0x40080748                                                                
I (29) boot: ESP-IDF v3.2-dev-362-g1c7a8b3b-dirty 2nd stage bootloader          
I (29) boot: compile time 14:32:38                                              
I (30) boot: Enabling RNG early entropy source...                               
I (35) boot: SPI Speed      : 80MHz                                             
I (39) boot: SPI Mode       : DIO                                               
I (43) boot: SPI Flash Size : 4MB                                               
I (47) boot: Partition Table:                                                   
I (51) boot: ## Label            Usage          Type ST Offset   Length         
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000       
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000       
[... 145 lines cut ...]
                                                                                
I (136808) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1                      
I (136808) wifi: station: f4:31:c3:c3:41:f9 join, AID=1, bg, 20                 
I (136808) br_10: EVENT: 15                                                     
I (136808) br_10: station:f4:31:c3:c3:41:f9 join, AID=1                         
                                                                                
I (136948) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2        
I (136948) br_10: EVENT: 17                                                     
I (139498) wifi: station: f4:31:c3:c3:41:f9 leave, AID = 1                      
I (139498) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1                      
I (139498) br_10: EVENT: 16                                                     
I (139498) br_10: station:f4:31:c3:c3:41:f9leave, AID=1                         
                                                                                
I (145588) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1                      
I (145588) wifi: station: f4:31:c3:c3:41:f9 join, AID=1, bg, 20                 
I (145588) br_10: EVENT: 15                                                     
I (145588) br_10: station:f4:31:c3:c3:41:f9 join, AID=1                         
                                                                                
I (145728) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2        
I (145728) br_10: EVENT: 17                                                     
I (166348) wifi: station: f4:31:c3:c3:41:f9 leave, AID = 1                      
I (166348) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1                      
I (166348) br_10: EVENT: 16                                                     
I (166348) br_10: station:f4:31:c3:c3:41:f9leave, AID=1    
```

akshar001, 2018-09-23:
Hello can somebody tell us when this is thing is going to get fixed?

francescofcf, 2018-09-28:
I would like to know if it is a problem that can be solved, even using a workaround. 
Thank you very much

vonnieda, 2018-09-28:
@francescofcf In my project I have had to use several different checks to see if the WiFi connection is still active. I ping a number of servers and after a certain number of pings I try to reconnect. In addition, when starting a connection I set a timer and if I don't get either a CONNECTED or DISCONNECTED message within several seconds I abort and try again. This works, but it's certainly not ideal.

francescofcf, 2018-09-28:
Unfortunately, I can't do anything like that. I open only a communication and do not do traffic.
I am a fake server and work as an access point, for my purpose I need only about the keepAlive and know if it is connected  :(

XinDeng11, 2018-10-24:
@francescofcf hi, can you offer your reproduce step, because I can always receive AP_STADISCONNECTED event after connected event, maybe there is unknow gap between us

francescofcf, 2018-10-24:
only simple_wifi example, connect and diaconnect repeataly and loose the disconnect event.
Read first post and output on uart terminal.

akshar001, 2018-10-31:
As i have understood throughout reading the problem over and over again , i don't think this event can happen at exact time as we wanted because this is how all the wifi works, for continuous checking that the station is available or not there should be some ping a like (AP pings every station connected to it to check it is still there) method in wifi standards , there is one but it is slow , it will took roughly 2-10 minutes to get to know that station is disconnected in every router ( when station just stopped working without notifying ap ) so if anyone who wants to check whether station is disconnected or not (fast) should have to implement a ping  a like mechanism on your own in the code from AP side and if you handle the station code as well you can send message to AP in regular interval and if it didn't get received in AP side in some time (using some timer mechanism) you can get to know that station is disconnected!

zhangyanjiaoesp, 2020-01-19:
@akshar001 @francescofcf 
We have two different scenario to trigger the `SYSTEM_EVENT_AP_STADISCONNECTED` event:

* It will be triggered immediately when softAP receives the deauth/disassociation frame from the STA.
* It will be triggered when the 5 minutes keep-alive timer expires, the timer expire if the softAP doesn't receive any packets from STA for 5 minutes. Generally this scenario happens when the STA do hardware reset/power-off, or when the disassociation/de-auth frame is failed received by softAP. Currently we don't have an API to configure the keep-alive timer,but we will consider to add such API to make the keep-alive timer can be user configured.

nopnop2002, 2020-01-21:
Maybe Wifi power savings is involved.

Try this

https://github.com/espressif/esp-idf/issues/2183#issuecomment-422246898

liuzfesp, 2020-07-13:
Same one as https://github.com/espressif/esp-idf/issues/4987, we will close this one and keep https://github.com/espressif/esp-idf/issues/4987.

closed 2020-08-20 by projectgus