espressif/esp-idf #2379: Failed to sniffer all channel with AP connected in single esp32 module

outermanle, 2018-09-06:
[sinffer.zip](https://github.com/espressif/esp-idf/files/2355549/sinffer.zip)

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32D]
- IDF version (53509c7b) release 3.1 and other version has the same problem
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
We want to do sniffer all channel with AP connected in single esp32 module.
During data transfer we change the channel which AP works in,but we get failed when we do stable test.

we change channel every 300-500ms,  ping no response and http post get failed after 1-5minutes,
and no any exception in wifi log, even ap disconnect info is not found in log.

esp_wifi_set_promiscuous to ture/false can't be set frequently, orelse esp32 will crash, so we just keep sniffer all the time.

[similar issue](https://github.com/espressif/esp-idf/issues/1045)

#### For example
##### thread1: wifi chanel change thread
1. mutex protect and change other channel stay in about 300ms;
2. mutex protect and change to AP channel stay in about 300ms;
3. goto 1;

##### thread2: transfer data thread
1. mutex protect and change to AP channel until data transfer complete.
2. sleep 2s;
3. goto 1;

### Expected Behavior
Works fine@sniffer mode in all channel with AP connected.

### Actual Behavior
Just works fine@sniffer mode in the fixed channel of  AP.
NOT work@sniffer mode in all channel with AP connected.

### Steps to repropduce
0. config wifi info then make all and make flash monitor;
1. start and connect to AP;
2. get esp32 ip and ping the esp32 ip in another window;
3. test about 1-5minutes;
4. then ping no response and http post failed.

### Code to reproduce this issue
the test code [sinffer.zip](https://github.com/espressif/esp-idf/files/2355549/sinffer.zip)

```cpp

/* lock channel during send data */
static void http_task(void *p)
{
    uint8_t channel;
    esp_err_t err;
    esp_http_client_config_t config = {
        .url = "http://httpbin.org/post",
        .event_handler = _http_event_handler,
        .port = 80,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_http_client_set_method(client, HTTP_METHOD_POST);

	while (true) {
        mutex_lock(channel_mutex);
        wifi_sniffer_get_channel(&channel);
        if (channel != cur_channel) {
            wifi_sniffer_set_channel(cur_channel);
        }
        const char *post_data = "field1=value1&field2=value2";
        esp_http_client_set_post_field(client, post_data, strlen(post_data));
        err = esp_http_client_perform(client);
        if (err == ESP_OK) {
            ESP_LOGI(TAG, "HTTP POST Status = %d, content_length = %d",
                    esp_http_client_get_status_code(client),
                    esp_http_client_get_content_length(client));
        } else {
            ESP_LOGE(TAG, "HTTP POST request failed: %s", esp_err_to_name(err));
        }
        mutex_unlock(channel_mutex);

        vTaskDelay(post_freq / portTICK_PERIOD_MS);
    }

    vTaskDelete(NULL);
    return;
}

static void wifi_channel_task(void *p)
{
    uint8_t channel  = cur_channel;
    uint8_t tmp_chan =1;

	while (true) {
        if (channel == cur_channel) {
            goto IGNORE;
        }
        // not allow change channel if we are sending data to server
        mutex_lock(channel_mutex);
        wifi_sniffer_get_channel(&tmp_chan);
        if (tmp_chan != channel) {
            wifi_sniffer_set_channel(channel);
        }
        vTaskDelay(WIFI_CHANNEL_SWITCH_INTERVAL / portTICK_PERIOD_MS);
        mutex_unlock(channel_mutex);

        mutex_lock(channel_mutex);
        wifi_sniffer_get_channel(&tmp_chan);
        if (tmp_chan != cur_channel) {
            wifi_sniffer_set_channel(cur_channel);
        }
        vTaskDelay(WIFI_CHANNEL_SWITCH_INTERVAL / portTICK_PERIOD_MS);
        mutex_unlock(channel_mutex);
    IGNORE:
        channel++;
        if(channel > WIFI_CHANNEL_MAX) {
            channel = 1;
        }
    }

    vTaskDelete(NULL);
    return;
}
```

## Debug Logs

```
I (2571) event: sta ip: 172.26.58.79, mask: 255.255.255.0, gw: 172.26.58.254
I (2571) sniffer: SYSTEM_EVENT_STA_GOT_IP
I (2571) sniffer: enable wifi sniffer mode.
I (2571) wifi: ic_enable_sniffer
I (2581) sniffer: current channel=[1]
I (2581) sniffer: promiscuous filter mask = 0x1
I (5951) sniffer: Current device num = 1

I (5951) sniffer: MAC: ************, The time is: 5910, The rssi = -91, chan=1

{
  "args": {}, 
  "data": "", 
  "files": {}, 
  "form": {
    "field1": "value1", 
    "field2": "value2"
  }, 
  "headers": {
    "Connection": "close", 
    "Content-Length": "27", 
    "Content-Type": "application/x-www-form-urlencoded", 
    "Host": "httpbin.org", 
    "User-Agent": "ESP32 HTTP Client/1.0"
  }, 
  "json": null, 
  "origin": "110.122.57.200", 
  "url": "http://httpbin.org/post"
}
I (11531) sniffer: HTTP POST Status = 200, content_length = 405
I (13401) sniffer: Current device num = 2
I (13411) sniffer: MAC: ************, The time is: 13360, The rssi = -79, chan=1
I (995731) sniffer: MAC: ************, The time is: 995690, The rssi = -86, chan=1
{
  "args": {}, 
  "data": "", 
  "files": {}, 
  "form": {
    "field1": "value1", 
    "field2": "value2"
  }, 
  "headers": {
    "Connection": "close", 
    "Content-Length": "27", 
    "Content-Type": "application/x-www-form-urlencoded", 
    "Host": "httpbin.org", 
    "User-Agent": "ESP32 HTTP Client/1.0"
  }, 
  "json": null, 
  "origin": "110.122.57.200", 
  "url": "http://httpbin.org/post"
}
I (998351) sniffer: HTTP POST Status = 200, content_length = 405
I (999171) sniffer: Current device num = 169
I (999171) sniffer: MAC: *************, The time is: 999130, The rssi = -92, chan=4
I (1003811) sniffer: Current device num = 170
I (1003811) sniffer: MAC: *************, The time is: 1003770, The rssi = -82, chan=1
I (1005841) sniffer: Current device num = 171
I (1005841) sniffer: MAC:*************, The time is: 1005800, The rssi = -92, chan=1
E (1006561) sniffer: HTTP POST request failed: ESP_ERR_HTTP_FETCH_HEADER
E (1014761) sniffer: HTTP POST request failed: ESP_ERR_HTTP_FETCH_HEADER
I (1020001) sniffer: Current device num = 172
I (1020001) sniffer: MAC: *************, The time is: 1019960, The rssi = -92, chan=1
E (1022961) sniffer: HTTP POST request failed: ESP_ERR_HTTP_FETCH_HEADER
I (1023651) sniffer: Current device num = 173
I (1023651) sniffer: MAC: 0xD2.0x21.0x2D.0x5C.0x1C.0x78, The time is: 1023610, The rssi = -89, chan=1
I (1026101) sniffer: Current device num = 174
I (1026101) sniffer: MAC: 0xDA.0xA1.0x19.0xFB.0x2C.0xE0, The time is: 1026060, The rssi = -83, chan=1
I (1026611) sniffer: Current device num = 175
I (1026611) sniffer: MAC: 0x36.0x76.0x36.0xF3.0x52.0x5B, The time is: 1026570, The rssi = -91, chan=1
I (1029251) sniffer: Current device num = 176
I (1029251) sniffer: MAC: *************, The time is: 1029210, The rssi = -85, chan=1
I (1029981) sniffer: Current device num = 177
I (1029981) sniffer: MAC: 0x2A.0x27.0x9F.0xD1.0x2C.0x39, The time is: 1029940, The rssi = -92, chan=1
E (1031161) sniffer: HTTP POST request failed: ESP_ERR_HTTP_FETCH_HEADER
I (1033291) sniffer: Current device num = 178
I (1033291) sniffer: MAC: 0xCA.0x91.0xEE.0xAE.0x1E.0x89, The time is: 1033250, The rssi = -91, chan=1
I (1033631) sniffer: Current device num = 179
I (1033631) sniffer: MAC: 0xD2.0x4D.0x8D.0x7E.0x4F.0x9A, The time is: 1033590, The rssi = -93, chan=10
I (1036651) sniffer: Current device num = 180
I (1036651) sniffer: MAC: *************,, The time is: 1036610, The rssi = -86, chan=1
E (1039361) HTTP_CLIENT: Error write request
E (1039361) sniffer: HTTP POST request failed: ESP_ERR_HTTP_WRITE_DATA
I (1044311) sniffer: Current device num = 181
I (1044311) sniffer: MAC: 0x8A.0xA9.0xC7.0x5F.0x8D.0xE4, The time is: 1044270, The rssi = -86, chan=1
I (1048401) sniffer: Current device num = 182
I (1048401) sniffer: MAC: 0xDA.0xA1.0x19.0x36.0x54.0x3F, The time is: 1048360, The rssi = -84, chan=1
I (1048571) sniffer: Current device num = 183
I (1048571) sniffer: MAC: 0x2A.0xA2.0x92.0x52.0x6F.0x5F, The time is: 1048530, The rssi = -90, chan=1
I (1049861) sniffer: Current device num = 184
I (1049861) sniffer: MAC:*************, The time is: 1049820, The rssi = -94, chan=1
E (1055481) HTTP_CLIENT: Connection failed, sock < 0
E (1055481) sniffer: HTTP POST request failed: ESP_ERR_HTTP_CONNECT
I (1056991) sniffer: 

```

Thanks a lot.

outermanle, 2018-09-07:
@TimXia 
esp32 wifi driver can do time division multiplexing to prevent this happen? and make station mode concurrence with sniffer mode. our network traffic is not heavy, we can accept some latency.

TimXia, 2018-09-12:
@outermanle No. Now ESP32 which is in station mode can not prevent AP from sending data to it after it switches to another channel. As a resutlt, ESP32 will fail to receive data from AP. We do not suggest to use sniffer mode on other channel when ESP32 is connected with AP.

outermanle, 2018-09-12:
I got it,thank you!

Alvin1Zhang, 2018-09-18:
@outermanle Does our reply help resolve your issue? If yes, would you help close this issue? Thanks.

outermanle, 2018-09-30:
@Alvin1Zhang yeah, please close this issue.Thanks a lot.

closed 2018-09-30 by Alvin1Zhang