espressif/esp-idf #2381: wifi: wpa2_enterprise with AES128-SHA cipher suite fails with HW crypto

mikaelkanstrup, 2018-09-06:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [-]
- Core (if using chip or module): [ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 599da58
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [external 5V]

## Problem Description

Connecting to wpa2_enterprise networks that selects TLS_RSA_WITH_AES_128_CBC_SHA cipher suite does not work when HW crypto/fast_crypto is used. Only software crypto works.

### Expected Behavior
1. Connect to wpa2_enterprise network
2. Notice that network selects TLS_RSA_WITH_AES_128_CBC_SHA cipher suite
3. Successful EAP handshake
4. Successful 4-way handshake
5. Wifi connected

### Actual Behavior
1. Connect to wpa2_enterprise network
2. Notice that network selects TLS_RSA_WITH_AES_128_CBC_SHA cipher suite
3. During EAP handshake server side rejects connection attempt due to decryption failure.
4. Wifi disconnected

### Steps to repropduce

1. Setup a enterprise test network with hostapd
2. Configure network to only allow cipher suite TLS_RSA_WITH_AES_128_CBC_SHA. i.e. hostapd.conf option 'openssl_cipher=AES128-SHA'
3. Connect to network
4. Notice connect failure

### Code to reproduce this issue

Use the esp-idf wpa2_enterprise example code to connect to test network

## Debug Logs

Relevant ESP32 logs
```
D (605) event: SYSTEM_EVENT_STA_START
V (615) event: enter default callback
V (615) tcpip_adapter: check: local, if=0 fn=0x400e5090
0x400e5090: tcpip_adapter_start_api at esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1082

V (615) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (625) tcpip_adapter: check: remote, if=0 fn=0x400e5090
0x400e5090: tcpip_adapter_start_api at esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1082

V (635) event: exit default callback
I (3045) wifi: n:3 0, o:1 0, ap:255 255, sta:3 0, prof:1
I (4025) wifi: state: init -> auth (b0)
I (4035) wifi: state: auth -> assoc (0)
I (4035) wifi: state: assoc -> run (10)
I (4035) wpa: wpa2_task prio:2, stack:6656

I (4045) wpa: EAP-TLS: Private key not configured
E (4045) wpa: Method private structure allocated failure

D (4055) wpa: TLS: using phase1 config options
D (4055) wpa: SSL: Received packet(len=6) - Flags 0x21
D (4055) wpa: EAP-PEAP: Start (server ver=1, own ver=1)
D (4065) wpa: EAP-PEAP: Using PEAP version 1
D (4065) wpa: TLSv1: Send ClientHello
D (4075) wpa: SSL: 62 bytes left to be sent out (of total 62 bytes)
D (4095) wpa: SSL: Received packet(len=1215) - Flags 0x01
D (4095) wpa: TLSv1: Received content type 22 version 3.1 length 42
D (4095) wpa: TLSv1: Received ServerHello
D (4095) wpa: TLSv1: Using TLS v1.0
D (4095) wpa: TLSv1: Selected cipher suite: 0x002f
D (4105) wpa: TLSv1: Received content type 22 version 3.1 length 1148
D (4115) wpa: TLSv1: Received Certificate (certificate_list len 1144)
D (4115) wpa: TLSv1: Certificate 0 (len 1138)
D (4125) wpa: X509: Version X.509v3
D (4125) wpa: X509: serialNumber 4
D (4125) wpa: X509: issuer CN=WiFi-Intermediate-CA-srv
D (4135) wpa: X509: Validity: notBefore: 0 notAfter: 0
D (4135) wpa: X509: subject CN=wifi-server
D (4145) wpa: X509: Extension: extnID=2.5.29.35 critical=0
D (4145) wpa: X509: Extension: extnID=2.16.840.1.113730.1.4 critical=0
D (4155) wpa: X509: Extension: extnID=2.5.29.31 critical=0
D (4165) wpa: X509: Extension: extnID=1.3.6.1.5.5.7.1.1 critical=0
D (4165) wpa: X509: Extension: extnID=2.5.29.18 critical=0
D (4175) wpa: X509: IssuerAltName
D (4175) wpa: X509: Extension: extnID=2.5.29.17 critical=0
D (4185) wpa: X509: SubjectAltName
D (4185) wpa: X509: Extension: extnID=2.5.29.15 critical=0
D (4195) wpa: X509: KeyUsage 0x5
D (4195) wpa: X509: Extension: extnID=2.5.29.37 critical=0
D (4195) wpa: X509: Extension: extnID=2.5.29.14 critical=0
D (4205) wpa: X509: Version X.509v3
D (4205) wpa: X509: serialNumber 4
D (4215) wpa: X509: issuer CN=WiFi-Intermediate-CA-srv
D (4215) wpa: X509: Validity: notBefore: 0 notAfter: 0
D (4225) wpa: X509: subject CN=wifi-server
D (4225) wpa: X509: Extension: extnID=2.5.29.35 critical=0
D (4235) wpa: X509: Extension: extnID=2.16.840.1.113730.1.4 critical=0
D (4235) wpa: X509: Extension: extnID=2.5.29.31 critical=0
D (4245) wpa: X509: Extension: extnID=1.3.6.1.5.5.7.1.1 critical=0
D (4255) wpa: X509: Extension: extnID=2.5.29.18 critical=0
D (4255) wpa: X509: IssuerAltName
D (4255) wpa: X509: Extension: extnID=2.5.29.17 critical=0
D (4265) wpa: X509: SubjectAltName
D (4265) wpa: X509: Extension: extnID=2.5.29.15 critical=0
D (4275) wpa: X509: KeyUsage 0x5
D (4275) wpa: X509: Extension: extnID=2.5.29.37 critical=0
D (4285) wpa: X509: Extension: extnID=2.5.29.14 critical=0
D (4285) wpa: X509: Validate certificate chain
D (4295) wpa: X509: 0: CN=wifi-server
D (4295) wpa: X509: Did not find any of the issuers from the list of trusted certificates
D (4305) wpa: X509: Certificate chain validation disabled - ignore unknown CA issue
D (4315) wpa: X509: Certificate chain valid
D (4315) wpa: TLSv1: Received content type 22 version 3.1 length 4
D (4325) wpa: TLSv1: Received ServerHelloDone
D (4325) wpa: TLSv1: Send ClientKeyExchange
D (4545) wpa: TLSv1: Send ChangeCipherSpec
D (4545) wpa: TLSv1: Record Layer - New write cipher suite 0x002f
D (4545) wpa: TLSv1: Send Finished
D (4545) wpa: SSL: 326 bytes left to be sent out (of total 326 bytes)
I (4565) wpa: >>>>>wpa2 FAILED

D (4565) wpa: TLSv1: Selected cipher suite: 0x0000
D (4565) wpa: TLSv1: Record Layer - New write cipher suite 0x0000
D (4565) wpa: TLSv1: Record Layer - New read cipher suite 0x0000 

I (4575) wifi: state: run -> init (17c0)
I (4575) wifi: n:3 0, o:3 0, ap:255 255, sta:3 0, prof:1
D (4585) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:Kanstrup, ssid_len:8, bssid:xx:xx:xx:xx:xx:xx, reason:23
V (4595) event: enter default callback
V (4595) tcpip_adapter: check: local, if=0 fn=0x400e569c
0x400e569c: tcpip_adapter_down_api at esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1082

D (4605) tcpip_adapter: if0 start ip lost tmr: enter
D (4605) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc6594 interval=120 ip=0
V (4615) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (4625) tcpip_adapter: check: remote, if=0 fn=0x400e569c
0x400e569c: tcpip_adapter_down_api at esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1082
```

Relevant hostapd logs:
```
Configuration file: peap-mschapv2.conf
Using interface wlan0 with hwaddr xx:xx:xx:xx:xx:xx and ssid "Kanstrup"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 30:ae:a4:22:6a:1c IEEE 802.11: authenticated
wlan0: STA 30:ae:a4:22:6a:1c IEEE 802.11: associated (aid 1)
wlan0: CTRL-EVENT-EAP-STARTED 30:ae:a4:22:6a:1c
wlan0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
wlan0: CTRL-EVENT-EAP-STARTED 30:ae:a4:22:6a:1c
wlan0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
wlan0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=13
wlan0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=25
SSL: SSL3 alert: write (local SSL3 detected an error):fatal:bad record mac
OpenSSL: openssl_handshake - SSL_connect error:1408F119:SSL routines:SSL3_GET_RECORD:decryption failed or bad record mac
wlan0: CTRL-EVENT-EAP-FAILURE 30:ae:a4:22:6a:1c
wlan0: STA 30:ae:a4:22:6a:1c IEEE 802.1X: authentication failed - EAP type: 0 (unknown)
wlan0: STA 30:ae:a4:22:6a:1c IEEE 802.1X: Supplicant used different EAP type: 25 (PEAP)
```

mikaelkanstrup, 2018-09-06:
The root cause for this problem is that the crypto engine is provided with an invalid key. key_len is incorrectly hard coded to 256 bits which it's not in this case. Problem fixed by modifying fast_crypto_internal-cipher.c function fast_crypto_cipher_init as shown below:

```
         case CRYPTO_CIPHER_ALG_AES:                
             mbedtls_aes_init(&(ctx->u.aes.ctx_enc));
-            mbedtls_aes_setkey_enc(&(ctx->u.aes.ctx_enc), key, 256);
+            mbedtls_aes_setkey_enc(&(ctx->u.aes.ctx_enc), key, key_len * 8);
             mbedtls_aes_init(&(ctx->u.aes.ctx_dec));
-            mbedtls_aes_setkey_dec(&(ctx->u.aes.ctx_dec), key, 256);               
+            mbedtls_aes_setkey_dec(&(ctx->u.aes.ctx_dec), key, key_len * 8);
```

negativekelvin, 2018-09-06:
Nice you probably just solved this issue too https://github.com/espressif/esp-idf/issues/2152

XinDeng11, 2018-09-14:
hi@mikaelkanstrup，thank you for your kindly remind, we have submit a merge request to fix the issue. I want to reproduce the phemenon in our site, but I can't find the openssl_cipher in my hostapd.conf, and if I add it by myself, when I run the hostapd, it reported a error which can not find the openssl_cipher choice

mikaelkanstrup, 2018-09-14:
@XinDeng11 The option is called 'openssl_ciphers'. I had got the name wrong in the issue description. Now edited and corrected. If still this does not work make sure you run a new enough hostapd. At least v2.4 is needed.

closed 2018-09-24 by igrr