espressif/esp-idf #2382: [Issue] ESP32 WPS on mikrotik hap lite router

farizadp, 2018-09-06:
Anyone has ever tried ESP32 WPS to Mikrotik hap lite router?
Using wps_example from esp-idf 599da58 (master), I got panic message, see debug log below. I would like to point out whether it's sdk or router issue.

```
I (680) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (701) wifi: Init dynamic tx buffer num: 16
I (702) wifi: Init data frame dynamic rx buffer num: 32
I (702) wifi: Init management frame dynamic rx buffer num: 32
I (706) wifi: Init static rx buffer size: 1600
I (710) wifi: Init static rx buffer num: 10
I (714) wifi: Init dynamic rx buffer num: 32
I (778) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
I (780) wifi: mode : sta (d8:a0:1d:40:80:d0)
I (782) example_wps: SYSTEM_EVENT_STA_START
I (784) example_wps: start wps...
I (787) wpa: manufacturer: ESPRESSIF, model number: ESP32, model name: ESPRESSIF IOT, device name: ESP STATION
I (797) wpa: wifi_wps_enable

I (11268) wifi: n:3 1, o:1 0, ap:255 255, sta:3 1, prof:1
I (12958) wifi: state: init -> auth (b0)
I (12964) wifi: state: auth -> assoc (0)
I (12971) wifi: state: assoc -> run (10)
I (17970) wifi: state: run -> init (fc0)
I (17971) wifi: n:3 0, o:3 1, ap:255 255, sta:3 1, prof:1
I (17973) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (18096) wifi: n:3 1, o:3 0, ap:255 255, sta:3 1, prof:1
I (18097) wifi: state: init -> auth (b0)
I (18125) wifi: state: auth -> assoc (0)
I (18136) wifi: state: assoc -> run (10)
I (23129) wifi: state: run -> init (fc0)
I (23130) wifi: n:3 0, o:3 1, ap:255 255, sta:3 1, prof:1
I (23132) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (23254) wifi: n:3 1, o:3 0, ap:255 255, sta:3 1, prof:1
I (23256) wifi: state: init -> auth (b0)
I (23259) wifi: state: auth -> assoc (0)
E (23262) wifi: Set status to INIT.

I (23263) wifi: state: assoc -> init (100)
I (23263) wifi: n:3 0, o:3 1, ap:255 255, sta:3 1, prof:1
I (23270) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (23395) wifi: n:3 1, o:3 0, ap:255 255, sta:3 1, prof:1
I (23397) wifi: state: init -> auth (b0)
I (23401) wifi: state: auth -> assoc (0)
E (23405) wifi: Set status to INIT.

I (23405) wifi: state: assoc -> init (100)
I (23405) wifi: n:3 0, o:3 1, ap:255 255, sta:3 1, prof:1
I (23411) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (23536) wifi: n:3 1, o:3 0, ap:255 255, sta:3 1, prof:1
I (23538) wifi: state: init -> auth (b0)
I (23541) wifi: state: auth -> assoc (0)
E (23545) wifi: Set status to INIT.

I (23546) wifi: state: assoc -> init (100)
I (23546) wifi: n:3 0, o:3 1, ap:255 255, sta:3 1, prof:1
I (23552) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (23678) wifi: n:3 1, o:3 0, ap:255 255, sta:3 1, prof:1
I (23679) wifi: state: init -> auth (b0)
I (23682) wifi: state: auth -> assoc (0)
E (23686) wifi: Set status to INIT.

I (23686) wifi: state: assoc -> init (100)
I (23687) wifi: n:3 0, o:3 1, ap:255 255, sta:3 1, prof:1
I (23693) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (23819) wifi: n:3 1, o:3 0, ap:255 255, sta:3 1, prof:1
I (23820) wifi: state: init -> auth (b0)
I (23824) wifi: state: auth -> assoc (0)
E (23827) wifi: Set status to INIT.

I (23828) wifi: state: assoc -> init (100)
I (23828) wifi: n:3 0, o:3 1, ap:255 255, sta:3 1, prof:1
I (23834) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (23960) wifi: n:3 1, o:3 0, ap:255 255, sta:3 1, prof:1
I (23962) wifi: state: init -> auth (b0)
I (23969) wifi: state: auth -> assoc (0)
E (23972) wifi: Set status to INIT.

I (23973) wifi: state: assoc -> init (100)
I (23973) wifi: n:3 0, o:3 1, ap:255 255, sta:3 1, prof:1
I (23976) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (24101) wifi: n:3 1, o:3 0, ap:255 255, sta:3 1, prof:1
I (24103) wifi: state: init -> auth (b0)
I (24110) wifi: state: auth -> assoc (0)
E (24114) wifi: Set status to INIT.

I (24114) wifi: state: assoc -> init (100)
I (24115) wifi: n:3 0, o:3 1, ap:255 255, sta:3 1, prof:1
I (24117) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c310  PS      : 0x00060230  A0      : 0x80111dcb  A1      : 0x3ffaf980
A2      : 0x3ffc7fd4  A3      : 0x00000000  A4      : 0x00000006  A5      : 0x3ffc7fd4
A6      : 0x3ffaf990  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffaf960
A10     : 0x3ffc7fe6  A11     : 0x3f40ad54  A12     : 0x00000000  A13     : 0x3ffc7fe6
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x4000c310:0x3ffaf980 0x40111dc8:0x3ffaf990 0x40111e3b:0x3ffaf9c0 0x400d1494:0x3ffaf9e0
```

Backtrace:
```
0x4000c310: ?? ??:0
0x40111dc8: wps_txStart at ??:?
0x40111e3b: wifi_station_wps_eapol_start_handle at ??:?
0x400d1494: timer_process_alarm at /data/projects/nsure/farpra/work_new_sdk/esp-idf/components/esp32/esp_timer.c:284
 (inlined by) timer_task at /data/projects/nsure/farpra/work_new_sdk/esp-idf/components/esp32/esp_timer.c:311
```

Mikrotik hap lite router information:
RouterOS v6.40.4 (stable)
Mikrotik hap lite wireless security setting:
- Mode: dynamic keys
- Authentication types: WPA PSK, WPA2 PSK
- Unicast ciphers: aes ccm
- Group Ciphers: aes ccm

Router logging:
```
21:05:02 wireless,info 00:0F:11:50:44:21@wlan1: connected 
21:05:07 wireless,info 00:0F:11:50:44:21@wlan1: disconnected, unicast key exchange timeout 
```

XinDeng11, 2018-09-18:
it is strange always fail when do assoc, please offer the air sniffer packet so that we may find some thing why it always fail

XinDeng11, 2018-09-25:
@farizadp Yes, but you  should make sure your wireless card's driver support the sniffer feature

farizadp, 2018-10-08:
@XinDeng11 : see attached pcap file. I use Wireshark version 2.6.3
I already filtered the packets down to:
- only in and outgoing packets between the esp32 and the access point
- broadcast packets from esp32 and the access point

i.e. :
(wlan.addr contains 80:d0 && wlan.addr contains e7:af) || ((wlan.sa contains 80:d0) && (wlan.da == ff:ff:ff:ff:ff:ff)) || ((wlan.sa contains e7:af) && (wlan.da == ff:ff:ff:ff:ff:ff))

[sniff_filtered.zip](https://github.com/espressif/esp-idf/files/2456809/sniff_filtered.zip)

Let me know if you're missing packets

XinDeng11, 2018-10-10:
@farizadp It looks very strange, AP start 4 way handshake rather than DH step at first time even both sides declare support WPS
[image]
because no correct pmk, it can not start 4 way.
After 4 way timeout, AP reject 32 at assoc step by status code 1. Can any other device success to connect with that router? If it that, please offer us the router's model

farizadp, 2018-10-10:
The following device is able to connect to the router using wps.
DELL Latitude 5480
Wireless NIC: Intel(R) Dual Band Wireless-AC 8260
OS: Windows 7

I've recorded again the air traffic of the esp32 and the the laptop for comparison.

[sniff_esp32_filtered.zip](https://github.com/espressif/esp-idf/files/2463817/sniff_esp32_filtered.zip)
[sniff_laptop_filtered.zip](https://github.com/espressif/esp-idf/files/2463818/sniff_laptop_filtered.zip)

Below is a link to the router's manufacturer website.
https://mikrotik.com/product/RB941-2nD

farizadp, 2018-10-12:
@XinDeng11 : Any news regarding this? Have you had time to look at both the laptop and esp32 air traffics?

farizadp, 2018-10-16:
@FayeY @XinDeng11 : Any update regarding this? Is there more information you need from us?

farizadp, 2018-10-18:
Can anyone please conclude whether this issue is at router side or sdk?

mattes, 2018-10-18:
Maybe this issue is related? Error log looks similar: https://github.com/espressif/esp-idf/issues/1188#issuecomment-430503903

farizadp, 2018-10-19:
@mattes: nope, that guy succeeded getting the IP address (which is some steps further than mine). Mine already fails during wireless association.

@igrr : perhaps you have an idea ?

XinDeng11, 2018-10-22:
@farizadp Sorry for later reply, I was asked to deal with another problem last week, the air packet can not find any difference between those two device, I prepare to apply buy the router in the link to verify in our offfice, maybe  it need another week to get the result

farizadp, 2018-10-22:
@XinDeng11 : Thanks for the respond, let me know when you have the router/result.

XinDeng11, 2018-10-25:
@farizadp The router's assoc response is against the spec, according to spec, it should include WSC IE when start WPS, however, this router only add WSC IE in probe response and beacon, but not include it in assoc response. And there is another thing make me confuse is that this router start 4 way handshake rather than RP even 32 has sent eapol start. I still need some time to check this router because this is not a standard WPS router

farizadp, 2018-10-25:
@XinDeng11 : thanks for reply. Still strange that the other device is able to handle it. Do you have a link where I can read the spec (honestly, I don't have knowledge regarding WPS) so that my understanding can be aligned with you?

XinDeng11, 2018-10-25:
[Wi-Fi_Simple_Configuration_Technical_Specification_v2.0.6.pdf](https://github.com/espressif/esp-idf/files/2514569/Wi-Fi_Simple_Configuration_Technical_Specification_v2.0.6.pdf)
this is the latest one

farizadp, 2018-10-26:
@XinDeng11 : ok.. I need some moment to understand those. But the obvious thing, why the DELL laptop does succeed WPS with this router?
If the router is not according to spec, then other device won't be able to WPS with this router, right? (was just blunt conclusion from me :) )

XinDeng11, 2018-10-26:
@farizadp because other device move the RSN IE and WPA IE before EAP-WSC success, and the spec just ask to add WPS IE in assoc request. What I want to say is the router don't follow the spec to check WPS IE when do connect. And I has done compatible to this router, it can success connect now. However, I don't this the router's check method is right and I still need discuss and test before merge the change

farizadp, 2018-10-29:
@XinDeng11 : Would it be an idea to first contact the manufacturer to confirm? Perhaps they have an explanation how to handle it if it's not according to spec

XinDeng11, 2018-10-30:
@farizadp Yeah, It had better to send a mail to the manufacturer. However, we have done a compatible to that sceanrio, you call try with the latest version in the master branch

farizadp, 2018-10-30:
@XinDeng11 : Thanks, do you have a commit id for it?

XinDeng11, 2018-11-01:
@farizadp you can check b8784a18 for verify

closed 2018-11-01 by igrr