espressif/esp-idf #2385: BLE Modem Sleep crash/Connection timeout

jakkra, 2018-09-07:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    cd78dc632a0ed6f13790c9e2f3167082e7f8356f with this patch: https://github.com/espressif/esp-idf/issues/2337#issuecomment-418038595
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
I had a problem where we crash after a few seconds sending BLE data with modem sleep enabled.
It was the TIMERG1 watchdog which triggered a reset. Which seemed like a smiliar problem as issue #2337. 

We got a patch (phylib.a https://github.com/espressif/esp-idf/issues/2337#issuecomment-418038595) for #2337 and it also solved the BLE crash problem also.
HOWEVER with that patch I have a new issue: After about 30 minutes duplex data transfer with a NRF52832 there is a BLE link timeout, the NRF device get a connection timeout event from it's Bluetooth controller.
But the ESP32 does NOT get a HCI Disconnect event, i.e our application still thinks there still is a BLE Link. The radio is still working, and we can for example connect to the ESP32 with SPP. 

So, there seems to be two errors here, first there should NOT be a connection timeout and link loss and this link loss should be reported from the ESP32 Bluetooth controller over HCI to us, but it's not.

The missing HCI Disconnect can't be reproduced in any other way, for example pulling the power of the NRF52 device, or walking out of range with it. In all those cases we get the HCi Disconnect event.

If I turn off Bluetooth Modem Sleep, none of those problems appear.

### Expected Behavior
Possible to send duplex BLE data between ESP32 and NRF52832 device without connection timeout/crash and if we get the connection timeout, ESP32 BT controller should report a HCI Disconnect event.
### Actual Behavior
Without phylib.a patch the ESP32 should not crash. With the patch the connection should not timeout, and if it does, we should receive a HCI Disconnect event.
### Steps to repropduce
1. Connect BLE from NRF52832 to ESP32
2. Send data in both directions

### Code to reproduce this issue

## Debug Logs

## Other items if possible
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2360280/sdkconfig.txt)

jakkra, 2018-09-13:
I have now tried on master branch today with the official fix: 7d561159929afd52949d8b39d1ea12b1314a4710
Unfortunately I still have the issue that I lose the LE link and no HCI Disconnect is generated. I also tried to connect to the device over BT classic after the link loss and it does not respond. Could it be that the radio dies somehow? The ESP32 does not hang or crash, I can communicate with it over UART. It works for about 30-90 minutes before the above issue happens. I'm running duplex data pumps.

As I wrote before, this does not happend if I pin Bluetooth to core 1, however I lose some data if I do that. But I can keep the link and have duplex data transfer running for hours. 

I have also seen the above issue once running duplex data transfer with an Android phone

jakkra, 2018-09-18:
I have made two new observations. First it seems like the BT Controller stops answering. I have added a timer which sends a HCI read MAC every 10 seconds. After the link is lost this HCI command i.e the BT Controller does not respond to it. 

Second, this issue is easier to reproduce if ESP32 is peripheral.

Further I have been sending duplex data between ESP32 (without BT modem sleep) and NRF52832 for 3 days stright without seeing this problem. When I enable BT Modem Sleep I see this issue already after just 10 to 60 minutes duplex data transfer.

mywang-espressif, 2018-09-20:
Hi @jakkra . I am looking into this issue. I have some questions on background information, could you please answer me that:

1. Did you use BLE and WIFI together at the same time in your test ?

2. Did you enable Dynamic Frequency Scaling(DFS) , i.e. set CONFIG_DFS_INIT_AUTO=y in menuconfig?

jakkra, 2018-09-20:
@mywang-espressif For the first two questions.

1. Not using WiFi at all
2. DFS is NOT enabled.

For the second two:

I added the prints as you said. The part below is the moments before it hangs.
See log below.

```
E (152897) btdm_sleep_check_duration : 17
E (152925) btdm_sleep_check_duration : 51
E (152927) btdm_sleep_check_duration : 48
E (152929) btdm_sleep_check_duration : 45
E (152931) btdm_sleep_check_duration : 42
E (152933) btdm_sleep_check_duration : 39
E (152935) btdm_sleep_check_duration : 36
E (152937) btdm_sleep_check_duration : 33
E (152939) btdm_sleep_check_duration : 30
E (152982) btdm_sleep_check_duration : 41
E (152984) btdm_sleep_check_duration : 38
E (152985) btdm_sleep_check_duration : 35
E (152987) btdm_sleep_check_duration : 32
165720:cb_gap.c:01374: Bluetooth Controlled does not respond
165720:cb_gap.c:01376: btdm_get_power_state: 3
175720:cb_gap.c:01374: Bluetooth Controlled does not respond
175720:cb_gap.c:01376: btdm_get_power_state: 3
185720:cb_gap.c:01374: Bluetooth Controlled does not respond
185720:cb_gap.c:01376: btdm_get_power_state: 3
```

Thanks!

mywang-espressif, 2018-09-20:
Many thanks for your information. 
Attached is a bluetooth controller library which I have added some functions for debugging. It can be used with the latest IDF master branch(commit 3276a1316). For further debugging, please replace this file with its counterpart in esp-idf, and use the following functions for more information when the issue is reproduced:
```
uint32_t btdm_deepslcntl_get(void);  // read the deepslcntl register
uint32_t btdm_deepslstat_get(void); // read the deepslstat register
uint32_t btdm_intcntl_get(void); // read the intcntl register
uint32_t btdm_intstat_get(void); // read the intstat register
uint32_t btdm_int_count_get(void); // read the total counts of bluetooth interrupts
uint16_t btdm_get_prevent_sleep_flag(void); // read a flag that prevent modem sleep
```

Please print the return values of those functions in HEX format when controller hangs. Please call `btdm_int_count_get()` multiple times to let me check whether bluetooth interrupt is still happening.
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/2401307/libbtdm_app.zip)

jakkra, 2018-09-20:
@mywang-espressif Here's the output:
```
51129:cb_gap.c:01443: BT Controller alive
61492:main.c:00328: Heap 8BIT    free/min/largest: 157648/155076/113804
66127:cb_gap.c:01380: Bluetooth Controlled does not respond
66127:cb_gap.c:01382: btdm_get_power_state: 3
66127:cb_gap.c:01391: btdm_deepslcntl_get: 3
66127:cb_gap.c:01392: btdm_deepslstat_get: 5
66127:cb_gap.c:01393: btdm_intcntl_get: 1
66127:cb_gap.c:01394: btdm_intcntl_get: 0
66128:cb_gap.c:01395: btdm_int_count_get: 22499
66128:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5

76127:cb_gap.c:01380: Bluetooth Controlled does not respond
76127:cb_gap.c:01382: btdm_get_power_state: 3
76127:cb_gap.c:01391: btdm_deepslcntl_get: 3
76127:cb_gap.c:01392: btdm_deepslstat_get: 5
76127:cb_gap.c:01393: btdm_intcntl_get: 1
76127:cb_gap.c:01394: btdm_intcntl_get: 0
76128:cb_gap.c:01395: btdm_int_count_get: 22499
76128:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5
81492:main.c:00328: Heap 8BIT    free/min/largest: 157544/155076/113804

86127:cb_gap.c:01380: Bluetooth Controlled does not respond
86127:cb_gap.c:01382: btdm_get_power_state: 3
86127:cb_gap.c:01391: btdm_deepslcntl_get: 3
86127:cb_gap.c:01392: btdm_deepslstat_get: 5
86127:cb_gap.c:01393: btdm_intcntl_get: 1
86127:cb_gap.c:01394: btdm_intcntl_get: 0
86127:cb_gap.c:01395: btdm_int_count_get: 22499
86128:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5

96127:cb_gap.c:01380: Bluetooth Controlled does not respond
96127:cb_gap.c:01382: btdm_get_power_state: 3
96127:cb_gap.c:01391: btdm_deepslcntl_get: 3
96127:cb_gap.c:01392: btdm_deepslstat_get: 5
96127:cb_gap.c:01393: btdm_intcntl_get: 1
96127:cb_gap.c:01394: btdm_intcntl_get: 0
96128:cb_gap.c:01395: btdm_int_count_get: 22499
96128:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5
```

It does not seem to change.

mywang-espressif, 2018-09-21:
Thank you very much for the details. Now I can decide that one step during waking up from deep sleep goes wrong, but I still can't figure out why there is no long bluetooth interrupt.  Perhaps I need to talk to our digital engineer and check the logic later.

I have another library file in which I modified the wake up delay. I don't know whether it can help but you can take it as a try. Please also keep the previous logging if the issue still happens.

By the way, I have another two questions:
1. Which CPU are your bluetooth host tasks pinned to, do controller and host run on the same CPU?
2. Can you tell me the connection interval for BLE connection in this case?
[libbtdm_app.a.zip](https://github.com/espressif/esp-idf/files/2404533/libbtdm_app.a.zip)

jakkra, 2018-09-21:
Hi @mywang-espressif and thank you

`esp_vhci_host_send_packet` is called on core 0
The application and Bluetooth stack is running on core 1
BT Controller is running on 0

Connection interval is 50ms.
I will test with some other connection intervals and see if anything is different.

With the latest libbtdm.a you provided it seems to work a little longer before it hangs, but otherwise same outcome. I have attached the log below.

```
3331126:cb_gap.c:01443: BT Controller alive
3341525:main.c:00328: Heap 8BIT    free/min/largest: 157472/155100/113804
3346112:cb_gap.c:01380: Bluetooth Controlled does not respond
3346112:cb_gap.c:01382: btdm_get_power_state: 3
3346112:cb_gap.c:01391: btdm_deepslcntl_get: 19
3346112:cb_gap.c:01392: btdm_deepslstat_get: 33
3346112:cb_gap.c:01393: btdm_intcntl_get: 1
3346112:cb_gap.c:01394: btdm_intcntl_get: 0
3346113:cb_gap.c:01395: btdm_int_count_get: 1663539
3346113:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5

3356112:cb_gap.c:01380: Bluetooth Controlled does not respond
3356112:cb_gap.c:01382: btdm_get_power_state: 3
3356112:cb_gap.c:01391: btdm_deepslcntl_get: 19
3356112:cb_gap.c:01392: btdm_deepslstat_get: 33
3356112:cb_gap.c:01393: btdm_intcntl_get: 1
3356112:cb_gap.c:01394: btdm_intcntl_get: 0
3356113:cb_gap.c:01395: btdm_int_count_get: 1663539
3356113:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5
3361525:main.c:00328: Heap 8BIT    free/min/largest: 157368/155100/113804

3366112:cb_gap.c:01380: Bluetooth Controlled does not respond
3366112:cb_gap.c:01382: btdm_get_power_state: 3
3366112:cb_gap.c:01391: btdm_deepslcntl_get: 19
3366112:cb_gap.c:01392: btdm_deepslstat_get: 33
3366112:cb_gap.c:01393: btdm_intcntl_get: 1
3366112:cb_gap.c:01394: btdm_intcntl_get: 0
3366113:cb_gap.c:01395: btdm_int_count_get: 1663539
3366113:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5

3376112:cb_gap.c:01380: Bluetooth Controlled does not respond
3376112:cb_gap.c:01382: btdm_get_power_state: 3
3376112:cb_gap.c:01391: btdm_deepslcntl_get: 19
3376112:cb_gap.c:01392: btdm_deepslstat_get: 33
3376112:cb_gap.c:01393: btdm_intcntl_get: 1
3376112:cb_gap.c:01394: btdm_intcntl_get: 0
3376113:cb_gap.c:01395: btdm_int_count_get: 1663539
3376113:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5
3381525:main.c:00328: Heap 8BIT    free/min/largest: 157264/155100/113804

3386112:cb_gap.c:01380: Bluetooth Controlled does not respond
3386112:cb_gap.c:01382: btdm_get_power_state: 3
3386112:cb_gap.c:01391: btdm_deepslcntl_get: 19
3386112:cb_gap.c:01392: btdm_deepslstat_get: 33
3386112:cb_gap.c:01393: btdm_intcntl_get: 1
3386112:cb_gap.c:01394: btdm_intcntl_get: 0
3386113:cb_gap.c:01395: btdm_int_count_get: 1663539
3386113:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5

3396112:cb_gap.c:01380: Bluetooth Controlled does not respond
3396112:cb_gap.c:01382: btdm_get_power_state: 3
3396112:cb_gap.c:01391: btdm_deepslcntl_get: 19
3396112:cb_gap.c:01392: btdm_deepslstat_get: 33
3396112:cb_gap.c:01393: btdm_intcntl_get: 1
3396112:cb_gap.c:01394: btdm_intcntl_get: 0
3396113:cb_gap.c:01395: btdm_int_count_get: 1663539
3396113:cb_gap.c:01396: btdm_get_prevent_sleep_flag: 5
```

jakkra, 2018-09-24:
@mywang-espressif  It's harder to reproduce with 12.5ms connection interval.

jakkra, 2018-09-24:
@mywang-espressif I can reproduced this between two ESP32's also, still only with our application running on both ESP's though. Just slowing down the data transfer rate to about 5kbps makes it hang after only a few minutes.

mywang-espressif, 2018-09-25:
Hi @jakkra,  I would like to check with your the following pieces of information regarding the running task of each CPU:
1. Bluetooth controller is running on CPU 0
2. Application and Bluetooth stack is running on CPU1, I think the Bluetooth stack also needs to use API `esp_vhci_host_send_packet()` to send hci packets to controller, right?
3. `esp_vhci_host_send_packet()` is called on CPU 0 in the context of FreeRTOS timer task to send HCI_Read_BD_ADDR command every 10 seconds
4. `esp_bt_controller_disable()` is called in application (CPU1) after some time of no communication with bluetooth controller.

Are the four items listed here right?

mywang-espressif, 2018-09-25:
@jakkra, I have another library file as attached. I have add more protections when accessing some critical registers in controlling sleep. I think the previous implementations are dubious and risky in dual-core user cases, but I am not sure of the effectiveness of this issue. Please take a try and also record the loggings as well.

mywang-espressif, 2018-09-25:
Add the missing attachment for the last comment.

file:///home/mywang/libbtdm_app.a.zip

jakkra, 2018-09-25:
@mywang-espressif 

1. Correct
```
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_0=y
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_1=
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=0
CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI=y
```
2. Correct, but the stack does not directly call `esp_vhci_host_send_packet()`, all outgoing packets from the stack are queued and then sent in another task on core 0 to the controller, because calling `esp_vhci_host_send_packet()` on core 1 causes a crash. So we had to make this context/core switch before calling `esp_vhci_host_send_packet()`
```
xTaskCreatePinnedToCore(stack_task, "stack_task", 4 * 1024, NULL, 1, NULL, appCore);

stack_task(...) {
 while (1) {
        hci_msg_t msg;
        if (pdTRUE == xQueueReceive(hci_out_queue, (void*)&msg, 1000)) {
            esp_vhci_host_send_packet(msg.hci_packet, msg.len);
            free(msg.hci_packet);
        }
    }
}
```

Crash when calling `esp_vhci_host_send_packet()` on core 0:
```

0x4008ef64: vListInsert at C:/Users/jakob.krantz/Documents/nina-w15/component/cb_rtsl_esp32/esp-idf/components/freertos/list.c:188 (discriminator 1)
0x4008d393: vTaskPlaceOnEventList at C:/Users/jakob.krantz/Documents/nina-w15/component/cb_rtsl_esp32/esp-idf/components/freertos/tasks.c:3564
0x4008e328: xQueueGenericReceive at C:/Users/jakob.krantz/Documents/nina-w15/component/cb_rtsl_esp32/esp-idf/components/freertos/queue.c:2037
0x400843d9: semphr_take_wrapper at C:/Users/jakob.krantz/Documents/nina-w15/component/cb_rtsl_esp32/esp-idf/components/bt/bt.c:941
0x4012db89: API_vhci_host_send_packet at ??:?
0x401240fa: btdm_controller_mem_init at C:/Users/jakob.krantz/Documents/nina-w15/component/cb_rtsl_esp32/esp-idf/components/bt/bt.c:941
0x4010c4c1: transport_send_packet at C:\Users\jakob.krantz\Documents\nina-w15/component/cb_bt_bk/port/esp32/btstack_port_esp32.c:66
0x4010f60d: hci_send_acl_packet_fragments at C:\Users\jakob.krantz\Documents\nina-w15/component/cb_bt_bk/btstack/src/hci.c:4063
0x4010f6c1: hci_send_acl_packet_buffer at C:\Users\jakob.krantz\Documents\nina-w15/component/cb_bt_bk/btstack/src/hci.c:4063
0x401136e1: l2cap_get_outgoing_buffer at C:\Users\jakob.krantz\Documents\nina-w15/component/cb_bt_bk/btstack/src/l2cap.c:1185
```
3. HCI_Read_BD_ADDR is sent through the stack and not directly by calling `esp_bt_controller_disable()` therefore it's the same as in 2.

4. Correct

jakkra, 2018-09-25:
@mywang-espressif Your attachment still failed to upload.

mywang-espressif, 2018-09-25:
Add the missing attachment of the last comment
[libbtdm_app.a.zip](https://github.com/espressif/esp-idf/files/2414064/libbtdm_app.a.zip)

jakkra, 2018-09-25:
@mywang-espressif Unfortunately that fix did not help, it could be that it took a little longer before the issue occured, but I'm not sure. 

I forgot to remove the esp_restart after BT Controller hangs and therefore only have one log. I'm running it again where it keeps reporting the below values after hang, will report back with that shortly.

```
2216238:cb_gap.c:01382: Bluetooth Controlled does not respond
2216238:cb_gap.c:01383: btdm_deepslcntl_get: 3
2216238:cb_gap.c:01384: btdm_deepslstat_get: 33
2216238:cb_gap.c:01385: btdm_intcntl_get: 1
2216238:cb_gap.c:01386: btdm_intstat_get: 0
2216238:cb_gap.c:01387: btdm_int_count_get: 607233
2216239:cb_gap.c:01388: btdm_get_prevent_sleep_flag: 5
```

mywang-espressif, 2018-09-25:
Hi @jakkra , thank you very much for the support. I talked to our digital engineer and he suggested to read some more registers to debug.  Please help test once more.

1. Please use the attached file:  
[libbtdm_app.a.zip](https://github.com/espressif/esp-idf/files/2415216/libbtdm_app.a.zip)
2. Please print three more registers that can be accessed by the following functions/expression, when the issue occurs:
- uint32_t btdm_lpclk_div_get(void)
- uint32_t btdm_lpclk_src_get(void)
- DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG); // you need to include "soc/dport_reg.h"

jakkra, 2018-09-26:
@mywang-espressif Did you change anything in the last lib compared to the previous? Because I have been running all night without it hanging. I'm trying again with a slightly different setup.

mywang-espressif, 2018-09-26:
Oh really? Yes indeed I have modified a place which influences the time of giving back the semaphore used in blocking the context task in esp_vhci_host_send_packet().
I will provide a version for you soon without this modification for test, see if we've found the root cause.

mywang-espressif, 2018-09-26:
Hi @jakkra , here is the library file without my modification on the vhci semaphore. This should be used to reproduce the issue, please have a try
[libbtdm_app.a.zip](https://github.com/espressif/esp-idf/files/2418703/libbtdm_app.a.zip)

I think now we are approaching the root cause, and I will further study it.

jakkra, 2018-09-26:
@mywang-espressif Hi again, I have now been running the last lib for two hour without it hanging, which is weird. I also double checked that I have problems with https://github.com/espressif/esp-idf/issues/2385#issuecomment-424225988 and I do. 

Did you change anything else between those two versions?

mywang-espressif, 2018-09-26:
@jakkra , between those two versions I only added two functions to read internal registers, except the vhci semaphore:
···
uint32_t btdm_lpclk_src_get(void);
uint32_t btdm_lpclk_div_get(void);
···

jakkra, 2018-09-26:
@mywang-espressif I will re-test once again with https://github.com/espressif/esp-idf/issues/2385#issuecomment-424225988 where you added:

> add more protections when accessing some critical registers in controlling sleep

jakkra, 2018-09-26:
@Weijian-Espressif Never mind my two last comments. It just hang, only it took logner than usually.

We can probably conclude it's the vhci semaphore 👍 

```
7741560:main.c:00328: Heap 8BIT    free/min/largest: 157940/155300/113804
7756175:cb_gap.c:01387: Bluetooth Controlled does not respond
7756175:cb_gap.c:01388: btdm_deepslcntl_get: 19
7756175:cb_gap.c:01389: btdm_deepslstat_get: 33
7756175:cb_gap.c:01390: btdm_intcntl_get: 1
7756175:cb_gap.c:01391: btdm_intstat_get: 0
7756176:cb_gap.c:01392: btdm_int_count_get: 2674606
7756176:cb_gap.c:01393: btdm_get_prevent_sleep_flag: 5
7756176:btstack_port_esp32.c:00313: DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG): 4294936576
7761560:main.c:00328: Heap 8BIT    free/min/largest: 157888/155300/113804
7766175:cb_gap.c:01387: Bluetooth Controlled does not respond
7766175:cb_gap.c:01388: btdm_deepslcntl_get: 19
7766175:cb_gap.c:01389: btdm_deepslstat_get: 33
7766175:cb_gap.c:01390: btdm_intcntl_get: 1
7766175:cb_gap.c:01391: btdm_intstat_get: 0
7766175:cb_gap.c:01392: btdm_int_count_get: 2674606
7766176:cb_gap.c:01393: btdm_get_prevent_sleep_flag: 5
7766176:btstack_port_esp32.c:00313: DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG): 4294936576
7776175:cb_gap.c:01387: Bluetooth Controlled does not respond
7776175:cb_gap.c:01388: btdm_deepslcntl_get: 19
7776175:cb_gap.c:01389: btdm_deepslstat_get: 33
7776175:cb_gap.c:01390: btdm_intcntl_get: 1
7776175:cb_gap.c:01391: btdm_intstat_get: 0
7776175:cb_gap.c:01392: btdm_int_count_get: 2674606
7776176:cb_gap.c:01393: btdm_get_prevent_sleep_flag: 5
7776176:btstack_port_esp32.c:00313: DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG): 4294936576
```

mywang-espressif, 2018-09-26:
@jakkra, thanks for your confirmation. I will consider more on this issue and try to test it on my side. I will provide you another version to verify before I propose a merge request to ESP-IDF.

jakkra, 2018-09-26:
Here is the log from the last lib you sent me, without modification on the vhci semaphore
```
3946357:cb_gap.c:01387: Bluetooth Controlled does not respond
3946357:cb_gap.c:01388: btdm_deepslcntl_get: 3
3946357:cb_gap.c:01389: btdm_deepslstat_get: 33
3946357:cb_gap.c:01390: btdm_intcntl_get: 1
3946357:cb_gap.c:01391: btdm_intstat_get: 0
3946357:cb_gap.c:01392: btdm_int_count_get: 1073980
3946358:cb_gap.c:01393: btdm_get_prevent_sleep_flag: 5
3946358:cb_gap.c:01395: btdm_lpclk_div_get: 1279
3946358:cb_gap.c:01396: btdm_lpclk_src_get: 67108865
3946358:btstack_port_esp32.c:00313: DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG): 4294936576

3956357:cb_gap.c:01387: Bluetooth Controlled does not respond
3956357:cb_gap.c:01388: btdm_deepslcntl_get: 3
3956357:cb_gap.c:01389: btdm_deepslstat_get: 33
3956357:cb_gap.c:01390: btdm_intcntl_get: 1
3956357:cb_gap.c:01391: btdm_intstat_get: 0
3956357:cb_gap.c:01392: btdm_int_count_get: 1073980
3956358:cb_gap.c:01393: btdm_get_prevent_sleep_flag: 5
3956358:cb_gap.c:01395: btdm_lpclk_div_get: 1279
3956358:cb_gap.c:01396: btdm_lpclk_src_get: 67108865
3956358:btstack_port_esp32.c:00313: DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG): 4294936576

3961741:main.c:00328: Heap 8BIT    free/min/largest: 157632/155160/113804
3966357:cb_gap.c:01387: Bluetooth Controlled does not respond
3966357:cb_gap.c:01388: btdm_deepslcntl_get: 3
3966357:cb_gap.c:01389: btdm_deepslstat_get: 33
3966357:cb_gap.c:01390: btdm_intcntl_get: 1
3966357:cb_gap.c:01391: btdm_intstat_get: 0
3966357:cb_gap.c:01392: btdm_int_count_get: 1073980
3966358:cb_gap.c:01393: btdm_get_prevent_sleep_flag: 5
3966358:cb_gap.c:01395: btdm_lpclk_div_get: 1279
3966358:cb_gap.c:01396: btdm_lpclk_src_get: 67108865
3966358:btstack_port_esp32.c:00313: DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG): 4294936576

3976357:cb_gap.c:01387: Bluetooth Controlled does not respond
3976357:cb_gap.c:01388: btdm_deepslcntl_get: 3
3976357:cb_gap.c:01389: btdm_deepslstat_get: 33
3976357:cb_gap.c:01390: btdm_intcntl_get: 1
3976357:cb_gap.c:01391: btdm_intstat_get: 0
3976357:cb_gap.c:01392: btdm_int_count_get: 1073980
3976358:cb_gap.c:01393: btdm_get_prevent_sleep_flag: 5
3976358:cb_gap.c:01395: btdm_lpclk_div_get: 1279
3976358:cb_gap.c:01396: btdm_lpclk_src_get: 67108865
3976358:btstack_port_esp32.c:00313: DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG): 4294936576

3981741:main.c:00328: Heap 8BIT    free/min/largest: 157528/155160/113804
3986357:cb_gap.c:01387: Bluetooth Controlled does not respond
3986357:cb_gap.c:01388: btdm_deepslcntl_get: 3
3986357:cb_gap.c:01389: btdm_deepslstat_get: 33
3986357:cb_gap.c:01390: btdm_intcntl_get: 1
3986357:cb_gap.c:01391: btdm_intstat_get: 0
3986357:cb_gap.c:01392: btdm_int_count_get: 1073980
3986358:cb_gap.c:01393: btdm_get_prevent_sleep_flag: 5
3986358:cb_gap.c:01395: btdm_lpclk_div_get: 1279
3986358:cb_gap.c:01396: btdm_lpclk_src_get: 67108865
3986358:btstack_port_esp32.c:00313: DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG): 4294936576
```

jakkra, 2018-09-27:
Hi @mywang-espressif , do you have an estimate when you think it can be finished? This issue is currently blocking our release. How much work is it to integrate this for release/v3.1?

Thank you

mywang-espressif, 2018-09-28:
Hi @jakkra , I have raised merge requests on this issue. The patch is under testing and hopefully it can be merged in this week. Github repo can be synced a couple of days later. I will inform you the update then.

Attached are the two updated libraries,  one for the master branch and another for release/v3.1 branch. You can use them to verify the issue. Thank you.

jakkra, 2018-09-28:
@mywang-espressif Thanks alot! I'm testing and so far so good. 
Can you update me when your merge has passed internal testing and is merged internally? 
That would allow me to do some pre-release testing before the fix is synced with GitHub repo.

mywang-espressif, 2018-09-28:
@jakkra, sure

mywang-espressif, 2018-10-09:
@jakkra, the bugfix for this issue is already merged in ESP-IDF, targeted on branch: master as as well as release/v3.1 and you an access these commits now:

For master branch:
commit ID: 42a9f99bc8dd3780b061de2da6fe1f7b37eaf3cf

For branch release/v3.1:
commit ID: 0846615e989ff001882fdb98061d9d31277749b9

The commits are actually merged on September 30 and I feel sorry for the late information.

jakkra, 2018-10-09:
@mywang-espressif Thank you! I saw that the fix was merged before, so no problem.

closed 2018-10-09 by jakkra