espressif/esp-idf #2387: Wemos Lolin32 & iPhone app lost connection

SuzukiYushin, 2018-09-07:
Hi, I'm trying example code of  “BLE_notify” on “Wemos-Lolin32” and simple BLE app of iPhone by core Bluetooth.
    
    I can connect each other. But these are disconnected a few seconds later.
    
    I also tried communicate with Android phone by “Light Blue” app.  It’s stable.
    And I also use “Red Bear BLE NANO” with the simple BLE app of iPhone. It’ also stable.
    
    Only ESP32 and iPhone disconnect automatically.
    
    Does anyone know some solution for this problem?

chegewara, 2018-09-07:
Hi, there is one thing that can help you with this issue.
https://github.com/asterics/esp32_mouse_keyboard/issues/4#issuecomment-398695065

chegewara, 2018-09-10:
You dont have to use all security options, from what ive found with help other users this should be enough, at least for ble hid application:
`setAccessPermissions(ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED);`

Also there can be some connection interval requirements (i dont have iPhone brick so i cant test it).

wxjcug, 2018-09-11:
@chegewara Hello I've read most code of this [repo](https://github.com/asterics/esp32_mouse_keyboard) and most your discussion about connection with iOS, and checked every place I can call setAccessPermissions but it just doesn't work with my iPhone6. The connection was successful but nothing happened when I send key strokes. I guess the reason is [here](https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf)  chapter 9.2. I can't find any document about esp32 using SSP yet, I'll try the password method and tell you the result.

chegewara, 2018-09-11:
@wxjcug If you have read this issue https://github.com/asterics/esp32_mouse_keyboard/issues/4 then you can see that all users confirmed it is working now with this solution. Last post is confirmation that iPhone 6S is working too.

wxjcug, 2018-09-29:
@Weijian-Espressif Thank you for that, I've switched to the ble_hid_device_demo and it works very well with my iPhone.

chegewara, 2018-10-02:
I will try to borrow iPhone and will work on this library, but its not possible before next week.

SuzukiYushin, 2018-10-03:
@chegewara Thank you very much!

chegewara, 2018-10-07:
Hi @SuzukiYushin 
according to [this document](https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf) its important to setup properly advertising data (10.4-10.6). I know there is no issue with discovering and connecting esp32, but try to change BLEAdvertising.cpp with this code https://gist.github.com/chegewara/192d3a4b267592c828f2a0d25f9b8eb0

Unfortunately i dont have access to iPhone(only 30 minutes today and maybe some time tomorrow).
Also change your code according to this:
```
BLEDevice::init("MyESP");

// Create the BLE Server
pServer = BLEDevice::createServer();
BLEAdvertising* pAdvertising = pServer->getAdvertising(); /// <--- call this right after createServer()
pAdvertising->setMinInterval(0x100);   //try also 0x20
pAdvertising->setMaxInterval(0x100);   // try also 0x20
```

chegewara, 2018-10-08:
@SuzukiYushin Im sorry, but i could not make it work yet. Good news is that code based on my hid library which is in this repo works I tested it personally on iPhone):
https://github.com/asterics/esp32_mouse_keyboard

That means i can find issue and fix it, but dont know when, because i dont have access to iPhone anymore.

SuzukiYushin, 2018-10-09:
@chegewara Thank you very much.
Do you mean replace ESP32 libraries of Arduino to your "components/nkolban_BLE" folder?

chegewara, 2018-10-09:
@SuzukiYushin I just figured it out, this is not hid device code. What app on iPhone are you using to connect?

SuzukiYushin, 2018-10-09:
@chegewara Thank you for quick response.  I'm using own app that most simple code of core bluetooth. Also using LightBlue.

chegewara, 2018-10-09:
Is there any chance that you could try to build app with esp-idf instead of arduino? Arduino-esp32 is slightly behind esp-idf and this can be core bt stack issue.

SuzukiYushin, 2018-10-09:
@chegewara OK. I will try.

SuzukiYushin, 2018-10-09:
@chegewara I tried with LightBlue. It was disconnected as soon as I connected.

Weijian-Espressif, 2018-10-15:
@SuzukiYushin please test our gatt_security_demo in ESP-IDF, it works well with iPhones.

pctj101, 2018-10-16:
FYI I had difficulties keeping a BLE connection until I upgraded to esp-idf v3.1 at minimum.    That helped me save time by not trying to debug something that was already fixed.  

If arduino integration is lagging, one idea is to do the next most meaningful thing in your life and come back to Arduino BLE later.

SuzukiYushin, 2018-10-18:
@pctj101 I see. I think it's better way too.

SuzukiYushin, 2018-10-18:
@Weijian-Espressif Thanks. I will try.

Alvin1Zhang, 2018-10-25:
@SuzukiYushin Could you help share if any updates for this issue? Thanks.

chegewara, 2018-10-25:
@Alvin1Zhang I think this issue can be closed, since its problem with arduino and 3rd party library (Kolban's ble library). We can raise new issue in Kolban's repo or esp32-arduino.

closed 2018-10-25 by Alvin1Zhang