espressif/esp-idf #2388: UART fifo synchronization problem

qt1, 2018-09-07:
I have set up a simple serial to tcp bridge on Serial2, Serial is also used for monitoring.

After 30 minutes or so of operations the bytes arriving from Serial2 start to have a delay.
The last character of a message is kept by te ESP and is send along the next message.

My environment configuration is arduino under esp-idf

After a lot of experimentation I have narrowed may search and written the following test code:

In esp32-hal-uart.c

    unsigned uart_isr_counter=0;
    unsigned uart_rxfifo_cnt_wrong_counter=0;

    static void IRAM_ATTR _uart_isr(void *arg)
    {
        uint8_t i, c;
        BaseType_t xHigherPriorityTaskWoken;
        uart_t* uart;

        uart_isr_counter++;

        for(i=0;i<3;i++){
            uart = &_uart_bus_array[i];
            if(uart->intr_handle == NULL){
                continue;
            }
            uart->dev->int_clr.rxfifo_full = 1;
            uart->dev->int_clr.frm_err = 1;
            uart->dev->int_clr.rxfifo_tout = 1;
                
            while(uart->dev->status.rxfifo_cnt || (uart->dev->mem_rx_status.wr_addr != uart->dev->mem_rx_status.rd_addr)){
                if(uart->dev->status.rxfifo_cnt==0 &&  (uart->dev->mem_rx_status.wr_addr != uart->dev->mem_rx_status.rd_addr))
                    uart_rxfifo_cnt_wrong_counter++;
                c = uart->dev->fifo.rw_byte;
                if(uart->queue != NULL && !xQueueIsQueueFullFromISR(uart->queue)) {
                    xQueueSendFromISR(uart->queue, &c, &xHigherPriorityTaskWoken);
                }
            }
        }

        if (xHigherPriorityTaskWoken) {
            portYIELD_FROM_ISR();
        }
    }

I have added two counter: one that counts irs calls, and one that counts when the queue seems to get out of sync.
Here is what I get after running my application for a while:

    uart_isr_counter: 22443, uart_rxfifo_cnt_wrong_counter: 2

So it seems like the fifo gets out-of-sync with the  rxfifo_cnt

I would suggest to make the following permanent (I'll post a PR).

Can anyone that is familiar with the internals please confirm or comment?

Thanks!

negativekelvin, 2018-09-07:
Is there any pattern you see from the values of rxfifo_cnt, rd_addr, wr_addr?

https://github.com/espressif/esp-idf/issues/2204

qt1, 2018-09-07:
I did not go into additional investigation at this point, so I don't have information about the exact behavior of the FIFO. 
As far as I remember there is a known hardware bug related but I don't know the details.
I hope someone from ESP can check this in more details and maybe provide a more accurate workaround..

qt1, 2018-09-07:
This pull request is a simple fix (taken from flush),
https://github.com/espressif/arduino-esp32/pull/1849

It seems to fix the problem for me.
It should be reviewd by someone that understands the internals of the uart . Wrong condition may cause reading of random characters from the fifo..

xiongyumail, 2018-10-18:
Has the problem been solved?

negativekelvin, 2018-10-18:
@xiongyumail I think you can close this because it was fixed in arduino-esp32 and https://github.com/espressif/esp-idf/issues/2204 is still open here

Alvin1Zhang, 2019-05-20:
@qt1 Thanks for reporting the issue. Feel free to reopen if the issue still exists. Thanks.

closed 2019-05-20 by Alvin1Zhang