espressif/esp-idf #2389: ESP32 TCP Socket Questions

MessCodes, 2018-09-08:
The ESP32 is connected to the router as a Station, forms a local area network with the Windows host, develops a TCP Server based on ESP32, and uses a Windows application as a TCP Client. TCP Server sends 60KB data to the client through the send function. The following questions need to be answered:

1. The following error occurred during the sending process:
```
D (11889) SWAP: wdptd - select...
I (11899) SWAP: wdptd_dispatcher >(sockfd=55)
E (24709) wifi: esf_buf: t=3 l=24 max:8, alloc:8 no eb
W (24709) wifi: alloc eb len=24 type=3 fail, heap:40760

W (24709) wifi: m f null

I (32279) SWAP: wdptd_dispatcher send data
```
2. According to the manual (ESP-WROOM-32), the phyaphysical phy rate is 150Mbps. The data volume transmitted by 1s should be more than 1MB, but the actual sending process is slow, which may be the reason?

negativekelvin, 2018-09-09:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#esp32-wi-fi-throughput

liuzfesp, 2018-09-18:
HI @MessCodes could you help to provide following info:
1. The IDF commit ID
2. Does this issue happens on all your ESP32 boards?
3. Not update your IDF and enable the Full Calibration by disabling "Store phy calibration data in NVS" via menuconfig and then check whether this issue still existed:
     make menuconfig => Component config => PHY => Store phy calibration data in NVS
4. Update your IDF to latest v3.0 or v3.1 or master, check whether this issue still existed.

The log "esf_buf: t=3 l=24 max:8, alloc:8 no eb" indicates the WiFi short buffer (less than or equal 64 Bytes) is not enough. 

In IDF v3.0, before the commit 487c4b672515fed39fe311b636b7dfe31a547dc9, the WiFi short buffer is limited to 8 (according to your log, I guess the IDF release your are using should be the one before this commit), the limitation has been removed after this commit. In latest IDF master branch and v3.1, no limitation about the total number WiFi driver can allocate unless there not enough heap memory. 

This problem general happens when the channel condition is very very bad or the board has some problem, e.g. the RF is not good etc.

Alvin1Zhang, 2018-11-20:
@MessCodes Hi, will close this issue for now, and feel free to reopen this issue if there are any updates. Thanks.

closed 2018-11-20 by Alvin1Zhang