espressif/esp-idf #2391: rename with SD card?

snahmad, 2018-09-08:
Does rename folder or file works using C rename function.

Folder is not working for me,

igrr, 2018-09-09:
Can you please post the part of code where you are calling  `rename`? What does `rename` function return when it is not working for you? What is the value of `errno` in such situation?

snahmad, 2018-09-09:
Check this thread : https://esp32.com/viewtopic.php?f=13&t=7114

Try this example please for folder rename.
Add folder on sd card and try to rename. kindly do check
https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/main/sd_card_example_main.c

snahmad, 2018-09-11:
not try yet. I don't need any more rename in my application. If you have time do try out.

snahmad, 2018-09-11:
Do you know how can i add timestamp to my files on folder on Esp32.

igrr, 2018-09-12:
@snahmad 
```
rename("/sdcard/dir1", "/sdcard/dir2")
```
works, provided that /sdcard/dir1 exists and /sdcard/dir2 does not.

For the question related to timestamps, you have already posted it on esp32.com. Please don't post same thing on esp32.com and on github at the same time.

closed 2018-09-12 by igrr

snahmad, 2018-09-12:
ok thanks.