espressif/esp-idf #2392: lwip stack broken in master commit

chegewara, 2018-09-08:
```
In file included from D:/msys32/home/imper/ble_test/components/cpp_utils/WiFi.cpp:25:0:
D:/msys32/home/imper/esp/esp-idf/components/lwip/lwip/src/include/lwip/dns.h:112:1: error: 'err_t' does not name a type
 err_t            dns_gethostbyname(const char *hostname, ip_addr_t *addr,
 ^
D:/msys32/home/imper/esp/esp-idf/components/lwip/lwip/src/include/lwip/dns.h:114:1: error: 'err_t' does not name a type
 err_t            dns_gethostbyname_addrtype(const char *hostname, ip_addr_t *addr,
 ^
```

negativekelvin, 2018-09-09:
These lines went missing

https://github.com/espressif/esp-idf/blob/35785d08fdde4f336d41ca956473cd11de380329/components/lwip/include/lwip/lwip/dns.h#L39-L41

igrr, 2018-09-09:
FWIW this include seems to be needed even without ESP_DNS. Will add.

gregjesl, 2018-09-09:
`sntp/sntp.h` is not present in the new lwip submodule.  Is this by design? The Azure IoT Hub Client (which we are using in our project) uses `sntp.h`. 
```
In file included from C:/Users/[Removed]/Documents/GitHub/[Removed]/components/azure-iot-sdk-c/c-utility/pal/lwip/sntp_lwip.c:10:0:
C:/Users/[Removed]/Documents/GitHub/[Removed]/components/azure-iot-sdk-c/c-utility/pal/lwip/sntp_os.h:14:28: fatal error: apps/sntp/sntp.h: No such file or directory
compilation terminated.
make[1]: *** [/c/esp-idf/make/component_wrapper.mk:286: c-utility/pal/lwip/sntp_lwip.o] Error 1
make: *** [C:/esp-idf/make/project.mk:481: component-azure-iot-sdk-c-build] Error 2
```

```

negativekelvin, 2018-09-09:
https://github.com/espressif/esp-idf/blob/6f5820814fbd5e3f21dfbe6899259455f4f27621/examples/protocols/sntp/main/sntp_example_main.c#L24

gregjesl, 2018-09-09:
@negativekelvin Thank you.  I've created a pull request for Azure (https://github.com/Azure/azure-c-shared-utility/pull/184)

liuzfesp, 2018-09-18:
@zhangyanjiao please help

closed 2018-09-27 by igrr