espressif/esp-idf #2393: SPI Slave MISO mode incorrect

ChrisHughes, 2018-09-08:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v1
- Core (if using chip or module): ESP32
- IDF version 6f58208
- Development Env: Make
- Operating System: Ubuntu
- Power Supply: external 5V

## Problem Description

Using SPI Slave to override an SPI mode 0 data stream in an application. Setup works on Arduino Uno. When observing output from the SPI slave demo in a logic analyzer, I'm noticing that MISO (tx output from ESP32) is half a clock phase out. When looking at the ESP32 documentation and registers, I was able to fix the problem by configuring ctrl2.miso_delay_mode = 1;

As this is set to 0 in spi_slave.c at line 187, I figured I'd post a trouble ticket here in the hope that it gets fixed here.

### Expected Behavior

SPI MISO output should match every other known working microcontroller and not be out half a clock phase.

### Actual Behavior

SPI MISO is out half a clock phase.

### Steps to repropduce

Observe the SPI slave demo example on a logic analyzer.

igrr, 2018-09-09:
Linking https://github.com/espressif/esp-idf/issues/1346#issuecomment-356404627 (and the comment which follows)

Alvin1Zhang, 2018-10-25:
@ChrisHughes Could you help share if any updates for this issue? Thanks.

Hoek67, 2019-02-13:
zomg.... I think I have this ussue... been trying to get SD Card to initialize and won't.

Checked everything and used same code on Due with the same SD card etc. Running OLED on the ESP32 worked fine... which indicated the BUS was initialized and running. Checked and rechecked wiring (lol 1 wire). Will see if I can fidget the code and get it working.

ginkgm, 2019-02-13:
@Hoek67 it seems to be a master issue, not related to this topic.
please open a new issue and describe it in detail if you still have the problem, thanks.

Hoek67, 2019-02-14:
Yes... it's a master issue but trying to narrow it down. I will put up a request in the right spot if can eliminate a few more variables. Big problem I have is ... no scope. 

I'm also working on a highly modified version of the SPI code but verified the original library (which it's derived from) has the same problem. I know the write byte function works and have verified the speed setting is working as had to make sure the 240Mhz front end was not causing code to time out.

Some of the older Sd code had timeouts fixed to "X" loop iterations which easily got exceeded on faster MCU's.

Good news is... my modified code is getting almost just over 1000 fps updating a small oled instead of a few hundred.

Alvin1Zhang, 2019-02-28:
@Hoek67 I will close this issue for now, feel free to reopen if you still have the issue as described in the issue subject, and please open a new issue if necessary. Thanks.

closed 2019-02-28 by Alvin1Zhang