espressif/esp-idf #2396: Recent changes causing slow bootloader startup times?

negativekelvin, 2018-09-10:
Has something changed recently where the bootloader is logging a startup time of over 1sec compared to previous 0.25sec?

```
I (31) boot: ESP-IDF v3.2-dev-512-gb4b1b361d 2nd stage bootloader
I (31) boot: compile time 22:06:16
I (31) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x08a04 ( 35332) map
I (108) esp_image: segment 1: paddr=0x00018a2c vaddr=0x3ffb0000 size=0x0212c (  8492) load
I (112) esp_image: segment 2: paddr=0x0001ab60 vaddr=0x3ffb212c size=0x00000 (     0) load
I (116) esp_image: segment 3: paddr=0x0001ab68 vaddr=0x40080000 size=0x00400 (  1024) load
I (125) esp_image: segment 4: paddr=0x0001af70 vaddr=0x40080400 size=0x050a0 ( 20640) load
I (142) esp_image: segment 5: paddr=0x00020018 vaddr=0x400d0018 size=0x125f8 ( 75256) map
I (169) esp_image: segment 6: paddr=0x00032618 vaddr=0x400854a0 size=0x03ec4 ( 16068) load
I (176) esp_image: segment 7: paddr=0x000364e4 vaddr=0x400c0000 size=0x00000 (     0) load
I (176) esp_image: segment 8: paddr=0x000364ec vaddr=0x50000000 size=0x00000 (     0) load
I (190) boot: Loaded app from partition at offset 0x10000
I (191) boot: Disabling RNG early entropy source...
I (196) cpu_start: Pro cpu up.
I (200) cpu_start: Starting app cpu, entry point is 0x40080ea0
I (0) cpu_start: App cpu up.
I (210) heap_init: Initializing. RAM available for dynamic allocation:
I (217) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (223) heap_init: At 3FFB3178 len 0002CE88 (179 KiB): DRAM
I (229) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (236) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (242) heap_init: At 40089364 len 00016C9C (91 KiB): IRAM
I (248) cpu_start: Pro cpu start user code
I (266) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

```
I (179) boot: ESP-IDF v3.2-dev-790-g45e3e14b1-dirty 2nd stage bootloader
I (180) boot: compile time 20:51:07
I (186) boot: Enabling RNG early entropy source...
I (202) boot: SPI Speed      : 40MHz
I (216) boot: SPI Mode       : DIO
I (228) boot: SPI Flash Size : 4MB
I (241) boot: Partition Table:
I (252) boot: ## Label            Usage          Type ST Offset   Length
I (275) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (298) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (321) boot:  2 factory          factory app      00 00 00010000 00100000
I (345) boot: End of partition table
I (358) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x08a04 ( 35332) map
I (462) esp_image: segment 1: paddr=0x00018a2c vaddr=0x3ffb0000 size=0x0212c (  8492) load
I (485) esp_image: segment 2: paddr=0x0001ab60 vaddr=0x3ffb212c size=0x00000 (     0) load
I (486) esp_image: segment 3: paddr=0x0001ab68 vaddr=0x40080000 size=0x00400 (  1024) load
I (514) esp_image: segment 4: paddr=0x0001af70 vaddr=0x40080400 size=0x050a0 ( 20640) load
I (591) esp_image: segment 5: paddr=0x00020018 vaddr=0x400d0018 size=0x125f8 ( 75256) map
I (755) esp_image: segment 6: paddr=0x00032618 vaddr=0x400854a0 size=0x03ec4 ( 16068) load
I (798) esp_image: segment 7: paddr=0x000364e4 vaddr=0x400c0000 size=0x00000 (     0) load
I (800) esp_image: segment 8: paddr=0x000364ec vaddr=0x50000000 size=0x00000 (     0) load
I (854) boot: Loaded app from partition at offset 0x10000
I (855) boot: Disabling RNG early entropy source...
I (862) cpu_start: Pro cpu up.
I (873) cpu_start: Starting app cpu, entry point is 0x40080ea0
I (1) cpu_start: App cpu up.
I (906) heap_init: Initializing. RAM available for dynamic allocation:
I (926) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (945) heap_init: At 3FFB3178 len 0002CE88 (179 KiB): DRAM
I (964) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (983) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1003) heap_init: At 40089364 len 00016C9C (91 KiB): IRAM
I (1023) cpu_start: Pro cpu start user code
I (350) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

negativekelvin, 2018-09-10:
https://github.com/espressif/esp-idf/pull/2398

igrr, 2018-09-19:
Fixed in #2398.

closed 2018-09-19 by igrr