espressif/esp-idf #2397: How do I get new SSL certificate

H0bbes, 2018-09-10:
Hello,
I am working on the OTA with simple_ota and I was surprised that you need the certificate in the source code. The certificate I use to connect to my web server will expire in a year. So how do i get a new one while the esp is running?
Is there a specific function to acquire a new one or should a make a HTTP request to get a new one and then use it to launch an OTA?
Or is there something I'm missing?
Thank You

negativekelvin, 2018-09-11:
If you compile the cert into the firmware, the new cert will be in the new firmware.  You just have to make sure you update your devices before the cert expires.  You may consider using a trusted root ca cert with long expiration ~20yrs.

H0bbes, 2018-09-11:
thank you for your answer.
But then how does your freshly installed computer  can access a https website directly? Someone must give him a first certificate? So I suppose there is a way for the esp to get a new one without having to make an firmware update.

chegewara, 2018-09-11:
To connect over https as a client you dont have to buy or generate any certificate, you eventually need root certificate which is publicly accessible, but you have to know what is root certificate for host you want to connect to.
esp_http_client library also allows you to connect with https without certificate, but it is not advised and its not safe.

H0bbes, 2018-09-12:
Ok now I get it.
Thanks for the explanation.

closed 2018-09-12 by mahavirj