espressif/esp-idf #2399: Document WROVER-B PSRAM Support

OtherCrashOverride, 2018-09-11:
What esp-idf revision is required for 8MB PSRAM?
How does one use the extra 4MB of PSRAM?
What are the limitations (if any) of its use?

chegewara, 2018-09-11:
https://esp32.com/viewtopic.php?t=6924

OtherCrashOverride, 2018-09-11:
It appears at the time of this writing:
1) There is no support for 8MB PSRAM in any esp-idf.
2) There is no documentation or sample (official or otherwise) for using it.
3) Only 4MB can be seen by ESP32 at any given time.

In the interests of clarity, this post is not intended to be a demand.  Rather it serves as documentation that it is not currently possible to use this feature of WROVER-B.

deltabeard, 2019-07-06:
@OtherCrashOverride
The official [ESP32-WROVER documentation](https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf) says that it has 8MB PSRAM integrated, so it seems bizarre that ESP-IDF wouldn't support 8MB PSRAM.

> There is no support for 8MB PSRAM in any esp-idf.

`SPIRAM_TYPE` has the option `SPIRAM_TYPE_ESPPSRAM64` ("ESP-PSRAM64 or LY68L6400"), so esp-idf seems to support 8MB PSRAM now (I haven't tested it yet).

> Only 4MB can be seen by ESP32 at any given time.

The option `CONFIG_SPIRAM_BANKSWITCH_ENABLE` seems to allow the use of external RAM larger than 4MiB via bank switching using [esp_himem API](https://github.com/espressif/esp-idf/blob/master/examples/system/himem/README.md).

projectgus, 2021-01-22:
Hi @OtherCrashOverride,

Sorry noone ever replied to you officially about this. The post from @deltabeard is correct about how to use this (the default "auto-detect size" also works as well. ESP32 can only address up to 4MB at a time, so the himem API is still needed. Newer chips (ESP32-S2, forthcoming ESP32-S3) can address larger PSRAM chips.

closed 2021-01-22 by projectgus