espressif/esp-idf #2400: Tickless Idle hangs up tasks - vTaskStep - assertion failed

Xasin, 2018-09-11:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Core (if using chip or module): ESP-WROOM32
- IDF version: 7dfb4386c
- Development Env:		Eclipse
- Operating System:		Debian
- Power Supply:               USB

## Problem Description

During normal code execution with frequent idle periods, automatic light sleep enabled, and a LEDC PWM fade in use, the ESP sometimes fails to continue task execution, throwing the warning:
```
/opt/ESP/esp-idf/components/freertos/tasks.c:2472 (vTaskStepTick)- assert failed!
abort() was called at PC 0x4008c8c7 on core 1
0x4008c8c7: vTaskStepTick at /opt/ESP/esp-idf/components/freertos/tasks.c:3567
```
The ESP does *not* restart, but instead simply stops executing the affected task.

The problem can be traced back to the source file, to a function related to FreeRTOS waking up from Tickless Idle mode. I have seen other reports regarding this exact problem, and it was theorized that it occurs when the ISR responsible for waking the chip up from an Idle phase skips execution for a tick. On the next execution, the assertion then fails, and tasks are not woken up, however, the chip does not reset either, "soft-locking" it.
In my opinion, I would either reset the chip if such a fault occurs or add a config option to ignore the failed assertion and continue execution normally. In either case, letting the chip hang up is not acceptable.

### Expected Behavior
The ESP should not hang up. It should either continue execution of the affected tasks or restart on such a fault condition.

### Actual Behavior
The ESP fails to continue execution of the affected tasks but does not restart. Essentially, it "soft-locks" and needs to be manually restarted.
### Steps to reproduce

1. Create a loop that executes a task at regular intervals. In my case, a loop that regularly flashed a LED with LEDC PWM and hardware fade was present.
2. Let the ESP run for a few minutes to a few hours. This issue is fairly rare and as such needs a long time to appear. It has appeared very consistently for me though if given enough time. *I have only been able to reproduce it with a LEDC Fade running*. Maybe the LEDC fade interrupt causes this?

### Code to reproduce this issue

https://gist.github.com/Xasin/ea3906ab18fa75859b217e0bb5dc7ac3

## Debug Logs

```
/opt/ESP/esp-idf/components/freertos/tasks.c:2472 (vTaskStepTick)- assert failed!
abort() was called at PC 0x4008c8c7 on core 1
0x4008c8c7: vTaskStepTick at /opt/ESP/esp-idf/components/freertos/tasks.c:3567
```
No further output on the console present.

## Other items if possible
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2369710/sdkconfig.zip)

Xasin, 2018-09-16:
Update:
The problem *can* be caused with other systems running. I had a successful vTaskStepTick assertion failure with the RMT module or the Capacitive Touch module running.

Removing the configASSERT in the offending line *does* fix the problem, and so far, *does not* cause any further problems. Tasks keep running, and it seems like nothing is having trouble with a skipped tick or two.
For those seeking a *temporary* fix for this, I can recommend doing the same ^^

dereks, 2018-09-27:
Me too. If I enalbe FreeRTOS tickless and power management with `make menuconfig`, calling 
```
vTaskDelay(1000 / portTICK_PERIOD_MS);
```
```
abort() was called at PC 0x400835a9 on core 0                                                                                                                                                           
Backtrace: 0x40094b30:0x3ffbde20 0x40094d05:0x3ffbde40 0x400835a9:0x3ffbde60 0x400836c9:0x3ffbde90 0x0                                                                                                  
CPU halted.                                                                                           
```

dereks, 2018-09-27:
In my app, it looks like `CONFIG_PM_DFS_INIT_AUTO` (CPU scaling on bootup) with `CONFIG_FREERTOS_USE_TICKLESS_IDLE` (ticklesss) causes my `abort()` during early app initialization -- it does not need to run for a long time.

I see that this bug's attached `sdkconfig` also has both of those options turned on simultaneously. I recommend disabling CONFIG_PM_DFS_INIT_AUTO as a possible workaround. (Or disable the tickless, but that is where the light sleep mode comes from.)

I do not yet know whether manually calling `esp_pm_configure()` within the app code also causes the `abort()`. I will test.

dereks, 2018-09-27:
Adding this manuall call to `esp_pm_configure()` also caused an abort:
```
    // Configure dynamic frequency scaling:
    // maximum and minimum frequencies are set in sdkconfig,
    // automatic light sleep is enabled if tickless idle support is enabled.
    esp_pm_config_esp32_t pm_config = {
            .max_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ,
            .min_freq_mhz = 10,
#if CONFIG_FREERTOS_USE_TICKLESS_IDLE
            .light_sleep_enable = true
#endif
    };
        
	if ((rc = esp_pm_configure(&pm_config)) != ESP_OK) {
#if CONFIG_FREERTOS_USE_TICKLESS_IDLE
		ESP_LOGE(TAG, "ERROR esp_pm_configure() (FreeRTOS USE_TICKLESS: on)");
#else
		ESP_LOGE(TAG, "ERROR esp_pm_configure() (FreeRTOS USE_TICKLESS: off)");
#endif
		goto exit;
	}
```
causes
```
abort() was called at PC 0x40083585 on core 1                                                         
                                                                                                      
Backtrace: 0x40094b0c:0x3ffbe420 0x40094ce1:0x3ffbe440 0x40083585:0x3ffbe460 0x400836a5:0x3ffbe490 0x0
                                                                                                      
CPU halted. 
```

However, now it happens later in my app code; the first `vTaskDelay()` now succeeds, but a later one aborts.

This is code I was handed, so I will dig deeper.

dereks, 2018-09-27:
On the ESP32 WROOM: 

```
// bad: causes the abort()
.min_freq_mhz = 10,

// bad: causes the abort()
.min_freq_mhz = 40;

// works but still bad: this value does not cause the abort(), 
// but it's just the max clock value, so it prevents CPU scaling for power management
.min_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ,
```

This is still with Tickless enabled.

igrr, 2018-09-27:
What you are seeing is probably another regression, https://github.com/espressif/esp-idf/issues/2459. We have a fix for that one in pipeline, will be merged soon.

Xasin, 2018-09-27:
@dereks 
I have also noticed crashes due to UART-Printing with automatic light sleep enabled. This is due to the UART module being paused during light sleep, so messages accumulate and don't get sent right.

Also, I am pretty sure your errors are unrelated to mine, as my error is very specific to the assertion called during wake-up from light sleep, *not* an `abort()`. The failed assertion is always visible in the crash message.
Consider opening a separate issue for your problem.

dereks, 2018-09-29:
In case anyone lands here from a search: My issue (`abort()` on `vTaskDelay())`  was different, and was resolved by the patch posted in #2459.

igrr, 2018-10-12:
@Xasin Sorry for the slow turnaround. Could you please try the attached patch?

[0001-freertos-prevent-tick-count-from-incrementing-in-vAp.zip](https://github.com/espressif/esp-idf/files/2471832/0001-freertos-prevent-tick-count-from-incrementing-in-vAp.zip)

Xasin, 2018-10-12:
So the problem was that, when entering the light sleep, the other CPU might still increment the counter by some value before also entering sleep?
Makes sense. I'll try the patch now - due to the nature of the bug I'll have to run the system for a few hours before I can be sure it worked.

igrr, 2018-10-22:
@Xasin did you get a chance to try the patch? Is the issue still observed?

igrr, 2018-10-24:
Thanks for the comment! We will keep the issue open until the patch makes its way into master. There is an additional defect identified during review which might lead to a deadlock, so the patch needs to be modified somewhat.

Xasin, 2018-10-24:
Haven't encountered a deadlock problem yet, but it's still good to iron that one out before.
Thanks for the quick patch!

Anyhow, time to open a new issue for you guys to work on :smiling_imp:

igrr, 2018-11-23:
The final fix (which is different from the patch I've posted above) was merged in https://github.com/espressif/esp-idf/commit/35d1c7b173a583956a4eab2526458330f3f6a2e3.

closed 2018-11-23 by igrr