espressif/esp-idf #2401: release/v3.1: BLE GATT Server Disconnects After 30 Seconds

eagi223, 2018-09-11:
## Environment

- Development Kit:                 [none]
- Kit version:                           [n/a]
- Core (if using chip or module): [ESP32]
- IDF version: 8e20c1334 (release/v3.1 latest)
- Development Env:		[other: Visual Studio Code]
- Operating System:		[MacOS]
- Power Supply:                     [Battery]

## Problem Description
I have an ESP32 based device with a custom board that is configurable from a smart phone app with a Bluetooth LE connection. Using Light Blue (a BLE development/testing application) on an iPhone 8+ I am able to see the ESP32 advertising, connect to the ESP32 and interact with the different BLE characteristics. Today, I updated the IDF, and now I'm only able to stay connected for around 1 minute before the ESP32 disconnects with the reason being 0x22 (ESP_GATT_CONN_LMP_TIMEOUT).  With previous versions of the IDF, I've not had an issue with this. The ESP32 will stay connected to the client device (phone) until I manually disconnect using the app.  I've been unable to find out where this disconnect is coming from, but it seems like it is from the IDF, since I've just had this start happening after updating my ESP-IDF version today. It happens both with my device's custom firmware as well as the gatt_server example

### Expected Behavior
ESP32 should stay connected to Bluetooth Client (smart phone) via Bluetooth Low Energy until Bluetooth Client (smart phone) disconnects.

### Actual Behavior
ESP32 disconnects after around 1 minute with the reason 0x22(ESP_GATT_CONN_LMP_TIMEOUT). This happens whether I am actively reading/writing characteristics or not.

### Steps to repropduce

1. Flash gatt_server example project to ESP32 board
2. Connect to GATT_SERVER_EXAMPLE bluetooth peripheral with a smart phone running either Light Blue or nRF Connect BLE development application.
3. Wait for up to 2 minutes for disconnect

### Code to reproduce this issue

I flashed my board with the GATT Server example project to verify that it was not a bug in my firmware, and had the same results. This can be reproduced with gatt_server example.

## Debug Logs

This appears to be where the disconnect is first triggered. I think this because after connecting to the ESP32 with Light Blue on an iPhone, there are no debug logs until this moment which is right before the app shows that the ESP32 has disconnected.

`
D (45770) BT_HCI: Receive packet event_code=0x13

D (45770) BT_HCI: Receive packet event_code=0x5

D (45770) BT_L2CAP: l2c_link_check_send_pkts
D (45780) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (45780) BT_L2CAP: TotalWin=10,LinkUnack(0x0)=0,RRCheck=0,RRUnack=0

D (45790) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (45800) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (45800) BT_OSI: osi_alarm_cancel failed to stop timer, err 0x103
`

I enabled all the bluetooth debug logs and attached them below.
[ble_debug_log.txt](https://github.com/espressif/esp-idf/files/2372818/ble_debug_log.txt)

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
I'm having trouble finding the sdkconfig file for the example directory, but I'm using the default IDF sdkconfig except in Bluetooth>Bluedroid Enable I have:
-Disabled GATT client module
-Disabled BLE security module
-Set all BT DEBUG LOG LEVELs to DEBUG
-Set BT/BLE MAX ACL CONNECTIONS to 2

I did this to ensure that the bluetooth settings for the example project matched my actual firmware's bluetooth settings. This being said, I also tried the gatt server example with the full default settings and had the same disconnect issue.

eagi223, 2018-09-12:
I just tested this with an Android device and I am not noticing the forced disconnect, so this may be iOS only.

eagi223, 2018-09-12:
It turns out that it was a bug in my specific iPhone. Resetting the phone ended up fixing the issue.

closed 2018-09-12 by eagi223

eagi223, 2018-12-30:
Just restart it. So power off, then back on and it worked. This seems to be a common problem with iPhones, but I’m not sure if it’s an iOS bug or an IDF bug. We’ve got multiple test iPhone that are various models and many have exhibited this same behavior.

marcogmaia, 2019-07-12:
I have the same problem, but with an old android. With new android phones it stays connected.