espressif/esp-idf #2402: Issue with using select on loopback socket

jacook81, 2018-09-11:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version: v4
- Core: ESP32-Wrover
- IDF version: V3.1, 53509c7bf
- Development Env:		Make
- Operating System:		MacOS
- Power Supply:         USB

## Problem Description

Attempting to use select to block on reading a loopback socket

//Detailed problem description goes here.
When using select to block on a loopback socket, the event is never triggered. I can wake it up by writing to a different socket and then read the loopback socket, the data that was written is then able to be read back.

### Expected Behavior
Select should return when the socket gets written to

### Actual Behavior
Select doesn't return

### Steps to repropduce
Create a loopback TCP socket and write to it.

### Code to reproduce this issue

```cpp

static const char* TAG = "PipeTask";
static volatile int writePipeFileDescriptor;
static volatile int readPipeDescriptor;

enum
{
    loopbackPort = 1000,
    listenQ      = 1
};

static void pipeListenTask(void *pvParameters)
{
    struct sockaddr_in tcpServerAddr;
    int    s;
    int    r;
    int    cs;
[... 252 lines cut ...]
I (11529) PipeTask:  WP[2]: s

I (11530) PipeTask:  WP[3]: t

I (11533) PipeTask:  WP[4]: i

I (11537) PipeTask:  WP[5]: n

I (11541) PipeTask:  WP[6]: g

testPipe
I (14940) PipeTask:  WP[0]: T

I (14940) PipeTask:  WP[1]: e

I (14940) PipeTask:  WP[2]: s

I (14941) PipeTask:  WP[3]: t

I (14944) PipeTask:  WP[4]: i

I (14948) PipeTask:  WP[5]: n

I (14952) PipeTask:  WP[6]: g
```

liuzfesp, 2018-09-18:
@zhangyanjiao please help

freakyxue, 2019-11-13:
@jacook81 
We will debug the problem as soon as possible

closed 2020-11-02 by Alvin1Zhang