espressif/esp-idf #2406: About component esp_https_ota problems

zhaochengwei, 2018-09-13:
According to https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/esp_https_ota.html?highlight=https,If I don't fill esp_http_client_config_t->cert_pem,the client can also communicate the https server. But in esp_https_ota.c, if i don't fill cert_pem, it will return ESP_FAIL.

I coment the check ,and it solve the problems.
[image]

chegewara, 2018-09-13:
This does not solve problem, this just does not check certificate validity so your connection is not secure.

zhaochengwei, 2018-09-13:
> This does not solve problem, this just does not check certificate validity so your connection is not secure.

So, If  I add the ssl certificate that means publick key?, that check fail ?

jitin17, 2018-09-17:
@zhaochengwei When it comes to normal HTTPS connection, server certificate verification is optional, but since firmware upgrade is a critical operation, we need to follow best security practices. So for best security practices, we have mandated HTTPS with server certificate verification, and it would be ideal if you provide root CA's certificate as server certificate.

zhaochengwei, 2018-10-22:
> @zhaochengwei When it comes to normal HTTPS connection, server certificate verification is optional, but since firmware upgrade is a critical operation, we need to follow best security practices. So [...]

yeah,I think what you said will be better. But I meet the problems if I put my firmware on aliyun oss, It will be hard to get the ca certificate. So can you help me?

AadiMehta, 2018-10-22:
I am trying to determine how to use secure HTTPS in case of private networks. ESP32 is trying to contact to a list of servers inside the network where ip to these servers are distributed by router through DHCP and there is no local DNS running inside the network. Any advice to resolve this issue?

jitin17, 2018-10-25:
@zhaochengwei I found this root CA certificate for aliyun oss:
```
-----BEGIN CERTIFICATE-----
MIIDdTCCAl2gAwIBAgILBAAAAAABFUtaw5QwDQYJKoZIhvcNAQEFBQAwVzELMAkG
A1UEBhMCQkUxGTAXBgNVBAoTEEdsb2JhbFNpZ24gbnYtc2ExEDAOBgNVBAsTB1Jv
b3QgQ0ExGzAZBgNVBAMTEkdsb2JhbFNpZ24gUm9vdCBDQTAeFw05ODA5MDExMjAw
MDBaFw0yODAxMjgxMjAwMDBaMFcxCzAJBgNVBAYTAkJFMRkwFwYDVQQKExBHbG9i
YWxTaWduIG52LXNhMRAwDgYDVQQLEwdSb290IENBMRswGQYDVQQDExJHbG9iYWxT
aWduIFJvb3QgQ0EwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDaDuaZ
jc6j40+Kfvvxi4Mla+pIH/EqsLmVEQS98GPR4mdmzxzdzxtIK+6NiY6arymAZavp
xy0Sy6scTHAHoT0KMM0VjU/43dSMUBUc71DuxC73/OlS8pF94G3VNTCOXkNz8kHp
1Wrjsok6Vjk4bwY8iGlbKk3Fp1S4bInMm/k8yuX9ifUSPJJ4ltbcdG6TRGHRjcdG
snUOhugZitVtbNV4FpWi6cgKOOvyJBNPc1STE4U6G7weNLWLBYy5d4ux2x8gkasJ
U26Qzns3dLlwR5EiUWMWea6xrkEmCMgZK9FGqkjWZCrXgzT/LCrBbBlDSgeF59N8
9iFo7+ryUp9/k5DPAgMBAAGjQjBAMA4GA1UdDwEB/wQEAwIBBjAPBgNVHRMBAf8E
BTADAQH/MB0GA1UdDgQWBBRge2YaRQ2XyolQL30EzTSo//z9SzANBgkqhkiG9w0B
AQUFAAOCAQEA1nPnfE920I2/7LqivjTFKDK1fPxsnCwrvQmeU79rXqoRSLblCKOz
yj1hTdNGCbM+w6DjY1Ub8rrvrTnhQ7k4o+YviiY776BQVvnGCv04zcQLcFGUl5gE
38NflNUVyRRBnMRddWQVDf9VMOyGj/8N7yy5Y0b2qvzfvGn9LhJIZJrglfCm7ymP
AbEVtQwdpf5pLGkkeB6zpxxxYu7KyJesF12KwvhHhm4qxFYxldBniYUr+WymXUad
DKqC5JlR3XC321Y9YeRq4VzW9v493kHMB65jUr9TU/Qr6cf9tveCX4XSQRjbgbME
HMUfpIBvFSDJ3gyICh3WZlXi/EjJKSZp4A==
-----END CERTIFICATE-----
```
So, can you copy it to `examples/system/ota/server_certs/ca_cert.pem` file and check whether `simple_ota_example` works or not.

mahavirj, 2018-11-15:
Closing (do no see any issue with `esp_https_ota` component).

closed 2018-11-15 by mahavirj