espressif/esp-idf #2407: Undefined reference to functions of regex.h library

GOST-UA, 2018-09-14:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v2
- Core (if using chip or module): ESP-WROOM32
- IDF version: master 31c14a5ee4e6019f82cac3dc663f28d2cd0a15de 
- Development Env:		Make
- Operating System:		Ubuntu
- Power Supply:         USB

## Problem Description
 When building a project with C regex (#include <regex.h>) linker reports about undefined reference of `regcomp/regexec` functions

### Expected Behavior

Project builds successfully.

### Actual Behavior

Failed on linkage phase

### Code to reproduce this issue

```cpp
#include <regex.h>

void app_main()
{
    regex_t regex;
   
    regcomp(&regex, "^a[[:alnum:]]", 0);
}

```
## Debug Logs

```
/home/esp/foo/build/main/libmain.a(main.o):(.literal.app_main+0x14): undefined reference to `regcomp'
/home/esp/foo/build/main/libmain.a(main.o): In function `app_main':
/home/esp/foo/main/main.c:388: undefined reference to `regcomp'
collect2: error: ld returned 1 exit status
```

negativekelvin, 2018-09-15:
I believe the supplied newlib does not provide regex support but you can use std::regex from libstdc++

GOST-UA, 2018-09-24:
But if I use only C language there are no standard regex library?

HrMitrev, 2019-05-08:
Has anyone found any altrnative to POSIX's regex? I'm contemplating on using https://github.com/kokke/tiny-regex-c, but it lacks some features. Is there any way to add the source code to esp-idf in order to make /components/newlib/include/regex.h work?

closed 2019-11-14 by espressif-bot