espressif/esp-idf #2411: aws-iot example in esp-idf does not receive published messages from aws

rhr407, 2018-09-14:
The subscribe-publish example inside esp-idf/examples/protocols/aws_iot folder publishes messages on aws but when I try to publish messages on aws side, esp32 does not seem to respond as it does not call iot_subscribe_callback_handler in subscribe_publish_simple.c file inside esp-idf/examples/protocols/aws_iot/subscrie_piblish/main folder. What is the issue? Can anyone please help?

mahavirj, 2018-09-17:
@rhr407 

Could you please provide more information - 
- What is ESP-IDF version that you are using?
- Any changes done to `subscribe-publish` example?
- Console/bootup log that you are observing?

mahavirj, 2018-09-18:
@rhr407 
Okay do not see any issue on device side. Probably some setting on you AWS IoT, may you please check policy attached with your `thing`, should look like,
```
    {
      "Effect": "Allow",
      "Action": "iot:*",
      "Resource": "*"
    }
```

rhr407, 2018-09-18:
Yes, it's working fine now. Thanks a lot.

closed 2018-09-18 by rhr407

sud92, 2018-11-15:
i am facing this similar problem, plz help me

`
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6216
load:0x40078000,len:10168
load:0x40080400,len:6552
entry 0x40080764
I (28) boot: ESP-IDF v3.2-beta1-34-gfb7ba1baa-dirty 2nd stage bootloader
I (28) boot: compile time 01:07:00
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1b34c (111436) map
I (133) esp_image: segment 1: paddr=0x0002b374 vaddr=0x3ff80000 size=0x00000 (     0) load
I (133) esp_image: segment 2: paddr=0x0002b37c vaddr=0x3ff80000 size=0x00000 (     0) load
I (139) esp_image: segment 3: paddr=0x0002b384 vaddr=0x3ffb0000 size=0x0324c ( 12876) load
I (153) esp_image: segment 4: paddr=0x0002e5d8 vaddr=0x3ffb324c size=0x00000 (     0) load
I (157) esp_image: segment 5: paddr=0x0002e5e0 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/sudipto/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (167) esp_image: segment 6: paddr=0x0002e9e8 vaddr=0x40080400 size=0x01628 (  5672) load
I (178) esp_image: segment 7: paddr=0x00030018 vaddr=0x400d0018 size=0x80da4 (527780) map
0x400d0018: _stext at ??:?

I (369) esp_image: segment 8: paddr=0x000b0dc4 vaddr=0x40081a28 size=0x0faf0 ( 64240) load
0x40081a28: esp_intr_disable at C:/msys32/home/sudipto/esp/esp-idf/components/esp32/intr_alloc.c:781

I (396) esp_image: segment 9: paddr=0x000c08bc vaddr=0x400c0000 size=0x00000 (     0) load
I (396) esp_image: segment 10: paddr=0x000c08c4 vaddr=0x50000000 size=0x00000 (     0) load
I (403) esp_image: segment 11: paddr=0x000c08cc vaddr=0x50000000 size=0x00000 (     0) load
I (422) boot: Loaded app from partition at offset 0x10000
I (422) boot: Disabling RNG early entropy source...
I (423) cpu_start: Pro cpu up.
I (427) cpu_start: Starting app cpu, entry point is 0x40081018
0x40081018: call_start_cpu1 at C:/msys32/home/sudipto/esp/esp-idf/components/esp32/cpu_start.c:245

I (0) cpu_start: App cpu up.
I (437) heap_init: Initializing. RAM available for dynamic allocation:
I (444) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (450) heap_init: At 3FFB92C0 len 00026D40 (155 KiB): DRAM
I (457) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (463) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (469) heap_init: At 40091518 len 0000EAE8 (58 KiB): IRAM
I (476) cpu_start: Pro cpu start user code
I (158) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (240) wifi: wifi driver task: 3ffc0eec, prio:23, stack:3584, core=0
I (240) wifi: wifi firmware version: a85ca5e
I (240) wifi: config NVS flash: enabled
I (240) wifi: config nano formating: disabled
I (240) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (250) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (280) wifi: Init dynamic tx buffer num: 32
I (280) wifi: Init data frame dynamic rx buffer num: 32
I (280) wifi: Init management frame dynamic rx buffer num: 32
I (290) wifi: Init static rx buffer size: 1600
I (290) wifi: Init static rx buffer num: 10
I (300) wifi: Init dynamic rx buffer num: 32
I (300) subpub: Setting WiFi configuration SSID TP-LINK_KAUSTAV...
I (380) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (380) wifi: mode : sta (24:0a:c4:96:64:d4)
I (380) subpub: AWS IoT SDK Version 3.0.1-
I (500) wifi: n:9 2, o:1 0, ap:255 255, sta:9 2, prof:1
I (1480) wifi: state: init -> auth (b0)
I (1490) wifi: state: auth -> assoc (0)
I (1510) wifi: state: assoc -> run (10)
I (1520) wifi: connected with TP-LINK_KAUSTAV, channel 9
I (1520) wifi: pm start, type: 1

I (2730) event: sta ip: 192.168.0.105, mask: 255.255.255.0, gw: 192.168.0.1
I (2730) subpub: Connecting to AWS...
E (5070) aws_iot: failed! mbedtls_ssl_handshake returned -0x2700
E (5070) aws_iot:     Unable to verify the server's certificate.
E (5070) subpub: Error(-4) connecting to ##############-ats.iot.ap-northeast-1.amazonaws.com:8883
E (7710) aws_iot: failed! mbedtls_ssl_handshake returned -0x2700
E (7710) aws_iot:     Unable to verify the server's certificate.
E (7720) subpub: Error(-4) connecting to ##############-ats.iot.ap-northeast-1.amazonaws.com:8883
E (9720) aws_iot: failed! mbedtls_ssl_handshake returned -0x2700
E (9720) aws_iot:     Unable to verify the server's certificate.
`

chegewara, 2018-11-15:
https://www.google.com/search?ei=ytftW-q6BIGjwAKX3ZKgBg&q=mbedtls+0x2700&oq=mbedtls+0x2700&gs_l=psy-ab.3...11340.16191..16668...0.0..0.133.1184.0j10......0....1..gws-wiz.......0i19.FS2SazOIzqM

sandipmaurya, 2022-11-04:
I am facing same issue please help me 

I (392) cpu_start: App cpu up.
I (422) cpu_start: Pro cpu start user code
I (422) cpu_start: cpu freq: 160000000
I (422) cpu_start: Application information:
I (426) cpu_start: Project name:     subscribe_publish
I (432) cpu_start: App version:      1
I (436) cpu_start: Compile time:     Nov  4 2022 10:39:50
I (442) cpu_start: ELF file SHA256:  af2379836c193664...
I (448) cpu_start: ESP-IDF:          v4.4.2-dirty
I (454) heap_init: Initializing. RAM available for dynamic allocation:
I (461) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (467) heap_init: At 3FFB7730 len 000288D0 (162 KiB): DRAM
I (473) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (480) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (486) heap_init: At 40094518 len 0000BAE8 (46 KiB): IRAM
I (494) spi_flash: detected chip: generic
I (497) spi_flash: flash io: dio
I (502) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (595) wifi:wifi driver task: 3ffc054c, prio:23, stack:6656, core=0
I (595) system_api: Base MAC address is not set
I (595) system_api: read default base MAC address from EFUSE
I (615) wifi:wifi firmware version: eeaa27d
I (615) wifi:wifi certification version: v7.0
I (615) wifi:config NVS flash: enabled
I (615) wifi:config nano formating: disabled
I (615) wifi:Init data frame dynamic rx buffer num: 32
I (625) wifi:Init management frame dynamic rx buffer num: 32
I (625) wifi:Init management short buffer num: 32
I (635) wifi:Init dynamic tx buffer num: 32
I (635) wifi:Init static rx buffer size: 1600
I (645) wifi:Init static rx buffer num: 10
I (645) wifi:Init dynamic rx buffer num: 32
I (645) wifi_init: rx ba win: 6
I (655) wifi_init: tcpip mbox: 32
I (655) wifi_init: udp mbox: 6
I (665) wifi_init: tcp mbox: 6
I (665) wifi_init: tcp tx win: 5744
I (665) wifi_init: tcp rx win: 5744
I (675) wifi_init: tcp mss: 1440
I (675) wifi_init: WiFi IRAM OP enabled
I (685) wifi_init: WiFi RX IRAM OP enabled
I (685) subpub: Setting WiFi configuration SSID myssid...
I (695) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (795) wifi:mode : sta (fc:f5:c4:01:ab:a8)
I (805) wifi:enable tsf
I (805) subpub: AWS IoT SDK Version 3.0.1-
I (815) wifi:new:<12,0>, old:<1,0>, ap:<255,255>, sta:<12,0>, prof:1
I (1745) wifi:state: init -> auth (b0)
I (1755) wifi:state: auth -> assoc (0)
I (2755) wifi:state: assoc -> init (400)
I (2755) wifi:new:<12,0>, old:<12,0>, ap:<255,255>, sta:<12,0>, prof:1
I (2755) subpub: Disconnected. Connecting to the AP again...
I (5175) subpub: Disconnected. Connecting to the AP again...
I (5215) wifi:new:<12,0>, old:<12,0>, ap:<255,255>, sta:<12,0>, prof:1
I (5215) wifi:state: init -> auth (b0)
I (5245) wifi:state: auth -> assoc (0)
I (5265) wifi:state: assoc -> run (10)
I (5315) wifi:connected with myssid, aid = 1, channel 12, BW20, bssid = 52:13:90:23:00:52
I (5315) wifi:security: WPA2-PSK, phy: bgn, rssi: -71
I (5315) wifi:pm start, type: 1

I (5315) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (6075) subpub: Connected with IP Address:192.168.207.9
I (6075) esp_netif_handlers: sta ip: 192.168.207.9, mask: 255.255.255.0, gw: 192.168.207.103
I (6075) subpub: Connecting to AWS...
E (16515) aws_iot: failed! mbedtls_ssl_handshake returned -0x6800
E (16515) subpub: Error(-4) connecting to a23neo0aw0yhem-ats.iot.ap-northeast-1.amazonaws.com:8883
[... 40 lines cut ...]
I (27) boot: compile time 10:39:54
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=17f2ch ( 98092) map
I (145) esp_image: segment 1: paddr=00027f54 vaddr=3ffb0000 size=03948h ( 14664) load
I (151) esp_image: segment 2: paddr=0002b8a4 vaddr=40080000 size=04774h ( 18292) load
I (159) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=84530h (542000) map
I (356) esp_image: segment 4: paddr=000b4558 vaddr=40084774 size=0fda4h ( 64932) load
I (383) esp_image: segment 5: paddr=000c4304 vaddr=50000000 size=00010h (    16) load
I (393) boot: Loaded app from partition at offset 0x10000
I (393) boot: Disabling RNG early entropy source...
I (405) cpu_start: Pro cpu up.
I (405) cpu_start: Starting app cpu, entry point is 0x40081198
0x40081198: call_start_cpu1 at /home/sandip/esp/esp-idf/components/esp_system/port/cpu_start.c:160

sandipmaurya, 2022-11-04:
> @rhr407 Okay do not see any issue on device side. Probably some setting on you AWS IoT, may you please check policy attached with your `thing`, should have permissions for all `iot` actions (something like below),
> 
> ```
>     {
>       "Effect": "Allow",
>       "Action": "iot:*",
>       "Resource": "*"
>     }
> ```

It is very helpfull.
Thank you