espressif/esp-idf #2415: error: invalid conversion from 'const void*' to 'cons t u8*

chegewara, 2018-09-16:
I am trying to use wpa sha256 library. I need to use hmac_sha256 ffrom it, but i am getting this error and cant find header to include to fix it:
`error: invalid conversion from 'const void*' to 'const u8* {aka const unsigned char*}' [-fpermissive]`

From my point of view its easier to write hmac_sha procedure than use existing one.
`hmac_sha256(const u8 *key, size_t key_len, const u8 *data, size_t data_len, u8 *mac)`
VS
```
#include "hwcrypto\aes.h"
#include "hwcrypto\sha.h"
#include <string>

#define IPAD 0x36
#define OPAD 0x5C
#define HMAC_SHA2_256

#ifdef HMAC_SHA2_256
	#define BLOCK_SIZE 64
	#define HMAC SHA2_256
#elif defined HMAC_SHA2_512
	#define BLOCK_SIZE 128
	#define HMAC SHA2_512
#endif
void _xor(const char *key, char pad, char* outp)
{
	for(int i=0;i<BLOCK_SIZE;i++)
	{
		outp[i] = key[i] ^ pad;
	}
}
void app_main(void) {

	esp_log_level_set("*", ESP_LOG_INFO);
	const char key[BLOCK_SIZE] = "secretPassw0rd";
	const char inp[] = "text to test HMAC";
	char conc[256] = {0};
	char conc2[256] = {0};
	unsigned char outp[BLOCK_SIZE/2] = {0};
	char tmp_out[BLOCK_SIZE] = {0};

	_xor(key, IPAD, tmp_out);
	memcpy(conc, tmp_out, BLOCK_SIZE);
	memcpy(conc + BLOCK_SIZE, inp, strlen(inp));
	
	esp_sha(HMAC, (const unsigned char *)conc, strlen((const char *)conc), outp);

	_xor(key, OPAD, tmp_out);
	memcpy(conc2, tmp_out, BLOCK_SIZE);
	memcpy(conc2 + BLOCK_SIZE, (const char *)outp, BLOCK_SIZE/2);

	esp_sha(HMAC, (const unsigned char *)conc2, BLOCK_SIZE * 1.5, outp);
	ESP_LOG_BUFFER_HEX("HASH step 2", outp, BLOCK_SIZE/2);
}
```

projectgus, 2018-09-17:
Hi @chegewara ,

Thanks for bringing this up. The hwcrypto functions are intended to provide hardware cryptography support for mbedTLS, rather than be a full crypto library themselves. We need to make this clearer in the docs.

You can calculate SHA-256 HMAC using the mbedTLS generic "message digest" support. This will use hardware SHA acceleration provided it's enabled in menuconfig.

```
#include "mbedtls/md.h"

mbedtls_md_context_t ctx;
 
mbedtls_md_init(&ctx);
mbedtls_md_setup(&ctx, mbedtls_md_info_from_type(MBEDTLS_MD_SHA256), true);
mbedtls_md_hmac_starts(&ctx, key, key_len);
mbedtls_md_hmac_update(&ctx, message, message_len);
mbedtls_md_hmac_finish(&ctx, result);
mbedtls_md_free(&ctx);
```

There are some HMAC implementations in the wpa_supplicant code as well, but these aren't public APIs and the long term plan is to refactor them to call mbedTLS implementations as well.

chegewara, 2018-09-17:
Thanks @projectgus, very helpful info.

closed 2018-09-17 by chegewara

chegewara, 2018-09-17:
@projectgus I have one more question. Since this code works perfect, i would like to random generate key. Its no big deal with mbedTLS library, but the issue is your this code is no longer producing valid hmac. This is my code to generate random key (copy/paste from mbedTLS website):
```
        mbedtls_ctr_drbg_context ctr_drbg;
	mbedtls_entropy_context entropy;

	char pers[] = "aes generate key";
	int ret;
	// unsigned char key[BLOCK_SIZE] = "secretPassw0rd";
	unsigned char key[32] = {0};

	mbedtls_entropy_init( &entropy );
	mbedtls_ctr_drbg_init( &ctr_drbg );

	if( ( ret = mbedtls_ctr_drbg_seed( &ctr_drbg, mbedtls_entropy_func, &entropy,
		(unsigned char *) pers, strlen( pers ) ) ) != 0 )
	{
		// printf( " failed\n ! mbedtls_ctr_drbg_init returned -0x%04x\n", -ret );
		goto exit;
	}

	if( ( ret = mbedtls_ctr_drbg_random( &ctr_drbg, key, 32) ) != 0 )
	{
		// printf( " failed\n ! mbedtls_ctr_drbg_random returned -0x%04x\n", -ret );
		goto exit;
	}
```

projectgus, 2018-09-17:
Hi @chegewara ,

Do you mean, if you use "secretPassw0rd" as key then it works, but using the random key doesn't produce a correct result?

Can you post the full piece of code somewhere? gist.github.com is a good option for this.

Do either of these function calls fail? (I noticed the error logging is commented out.)

chegewara, 2018-09-17:
Yes, with "secretPassw0rd" it works. I think i found the issue. `mbedtls_md_hmac_starts` expects that key is unsigned char[], but random gen key seems to be signed char[]. This is random key i copied and i am trying to reuse:
`unsigned char key[BLOCK_SIZE] = {0xd8, 0x38, 0x60, 0x93, 0xba, 0x93, 0x9b, 0xf9, 0xd2, 0x6b, 0x39, 0xc6, 0x41, 0xd0, 0x2b, 0x22, 0x7b, 0xe2, 0x94, 0x8c, 0x7e, 0x79, 0x5c, 0xf2, 0xf5, 0xc8, 0xd3, 0x1b, 0x27, 0xaf, 0x3c, 0xf0, 0x36, 0x0b, 0xf5, 0x15, 0xed, 0xb3, 0xe2, 0x54, 0xaa, 0x01, 0x92, 0xb9, 0xca, 0x52, 0x89, 0x68, 0xe6, 0xe0, 0xc1, 0xc5, 0x39, 0x05, 0xe3, 0xd0, 0x3b, 0xf2, 0xf4, 0xda, 0xe0, 0xf6, 0xf1, 0x46};`

Here is full example code:
https://gist.github.com/chegewara/8644431968a2a066a41358bda998fc69

chegewara, 2018-09-17:
@projectgus my bad, i made mistake in assumption. code is working with random key. sorry

projectgus, 2018-09-17:
@chegewara great :)