espressif/esp-idf #2416: problem compiling to master v3.2-dev-1029-g518942ec6

buckybots, 2018-09-17:
(if I back up to git checkout v3.1-rc2, no problem with compilation)

when I compile (after make clean)
make all -j5

this is the error in the current master :
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
In file included from C:/linuxShared/esp/esp-idf/components/esp_http_client/include/esp_http_client.h:19:0,
                 from C:/linuxShared/esp/esp-idf/components/esp_http_client/esp_http_client.c:28:
C:/linuxShared/esp/esp-idf/components/nghttp/port/include/http_parser.h:149:3: error: 'flags' defined as wrong kind of tag
   { F_CHUNKED               = 1 << 0
   ^
CC build/curl/lib/curl_ntlm_wb.o
CC build/esp32/pm_locks.o
make[1]: *** [/c/linuxShared/esp/esp-idf/make/component_wrapper.mk:286: esp_http_client.o] Error 1
make: *** [C:\linuxShared\esp\esp-idf/make/project.mk:481: component-esp_http_client-build] Error 2
make: *** Waiting for unfinished jobs....
CC build/driver/sdspi_crc.o
CC build/curl/lib/slist.o

igrr, 2018-09-17:
Could you please attach the output of `make V=1`, and your sdkconfig file?

buckybots, 2018-09-17:
I reverted back to Master, cleaned and recompiled.........

$ make V=1
including /c/linuxShared/esp/esp-idf/components/app_update/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/bootloader/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/coap/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/esp32/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/nghttp/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/partition_table/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/ulp/Makefile.projbuild...
PROJECT_PATH= COMPONENT_DIRS= make -C /c/linuxShared/esp/esp-idf/components/bootloader/subproject V=1 BUILD_DIR_BASE=/c/linuxShared/win/deviceVID/build/bootloader TEST_COMPONENTS= TESTS_ALL= EXCLUDE_COMPONENTS= /c/linuxShared/win/deviceVID/build/bootloader/bootloader.bin
make[1]: Entering directory '/c/linuxShared/esp/esp-idf/components/bootloader/subproject'
including /c/linuxShared/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including /c/linuxShared/esp/esp-idf/components/bootloader/subproject/main/Makefile.projbuild...
make -C /c/linuxShared/win/deviceVID/build/bootloader/bootloader_support -f /c/linuxShared/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/linuxShared/esp/esp-idf/components/bootloader_support/component.mk COMPONENT_NAME=bootloader_support build
make[2]: Entering directory '/c/linuxShared/win/deviceVID/build/bootloader/bootloader_support'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory '/c/linuxShared/win/deviceVID/build/bootloader/bootloader_support'
Target 'component-bootloader_support-build' responsible for '/c/linuxShared/win/deviceVID/build/bootloader/bootloader_support/libbootloader_support.a'
make -C /c/linuxShared/win/deviceVID/build/bootloader/log -f /c/linuxShared/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/linuxShared/esp/esp-idf/components/log/component.mk COMPONENT_NAME=log build
make[2]: Entering directory '/c/linuxShared/win/deviceVID/build/bootloader/log'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory '/c/linuxShared/win/deviceVID/build/bootloader/log'
Target 'component-log-build' responsible for '/c/linuxShared/win/deviceVID/build/bootloader/log/liblog.a'
make -C /c/linuxShared/win/deviceVID/build/bootloader/spi_flash -f /c/linuxShared/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/linuxShared/esp/esp-idf/components/spi_flash/component.mk COMPONENT_NAME=spi_flash build
make[2]: Entering directory '/c/linuxShared/win/deviceVID/build/bootloader/spi_flash'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory '/c/linuxShared/win/deviceVID/build/bootloader/spi_flash'
Target 'component-spi_flash-build' responsible for '/c/linuxShared/win/deviceVID/build/bootloader/spi_flash/libspi_flash.a'
make -C /c/linuxShared/win/deviceVID/build/bootloader/micro-ecc -f /c/linuxShared/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/linuxShared/esp/esp-idf/components/micro-ecc/component.mk COMPONENT_NAME=micro-ecc build
make[2]: Entering directory '/c/linuxShared/win/deviceVID/build/bootloader/micro-ecc'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory '/c/linuxShared/win/deviceVID/build/bootloader/micro-ecc'
Target 'component-micro-ecc-build' responsible for '/c/linuxShared/win/deviceVID/build/bootloader/micro-ecc/libmicro-ecc.a'
make -C /c/linuxShared/win/deviceVID/build/bootloader/soc -f /c/linuxShared/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/linuxShared/esp/esp-idf/components/soc/component.mk COMPONENT_NAME=soc build
make[2]: Entering directory '/c/linuxShared/win/deviceVID/build/bootloader/soc'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory '/c/linuxShared/win/deviceVID/build/bootloader/soc'
Target 'component-soc-build' responsible for '/c/linuxShared/win/deviceVID/build/bootloader/soc/libsoc.a'
make -C /c/linuxShared/win/deviceVID/build/bootloader/main -f /c/linuxShared/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/linuxShared/esp/esp-idf/components/bootloader/subproject/main/component.mk COMPONENT_NAME=main build
make[2]: Entering directory '/c/linuxShared/win/deviceVID/build/bootloader/main'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory '/c/linuxShared/win/deviceVID/build/bootloader/main'
Target 'component-main-build' responsible for '/c/linuxShared/win/deviceVID/build/bootloader/main/libmain.a'
make[1]: Leaving directory '/c/linuxShared/esp/esp-idf/components/bootloader/subproject'
python /c/linuxShared/esp/esp-idf/tools/check_python_dependencies.py
Python requirements from C:/linuxShared/esp/esp-idf/requirements.txt are satisfied.
Building partitions from /c/linuxShared/esp/esp-idf/components/partition_table/partitions_two_ota.csv...
python /c/linuxShared/esp/esp-idf/components/partition_table/gen_esp32part.py -q  --flash-size "4MB" --offset 0x8000  /c/linuxShared/esp/esp-idf/components/partition_table/partitions_two_ota.csv /c/linuxShared/win/deviceVID/build/partitions_two_ota.bin
[... cut ...]
make[1]: Entering directory '/c/linuxShared/win/deviceVID/build/aws_iot'
[... 664 lines cut ...]
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# TCP/IP Adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_LWIP=y

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

igrr, 2018-09-25:
@buckybots The issue happens because there are two `transport.h` files now: one in libssh2 component, the other one in `tcp_transport` component of ESP-IDF.

This is an oversight on our part, the name of the public header file should not be that generic. This will be fixed soon.

Also you might be able to fix this on your side by removing libssh2/src from the list of public include directories, and making it a private include directory instead (via `COMPONENT_PRIV_INCLUDEDIRS`).

projectgus, 2019-03-15:
Sorry for the lack of update. This was fixed in 40c6cbb3abf1d8531a3897ee88f640cf14d141bf.

closed 2019-03-15 by projectgus