espressif/esp-idf #2419: Mesh using NVS, crashes IRAM ISR ...

edjeed, 2018-09-17:
When a layer changes in the MESH network, the new layer is written to NVS by the mesh stack.
All interrupts that run in IRAM are not accessible when writing to NVS, so errors occur....

Is there a way to disable the use of NVS in the MESH stack?

It would be nice if it uses the same storage as the WiFi:
esp_wifi_set_storage(WIFI_STORAGE_RAM);

Edjeed

igrr, 2018-09-19:
@edjeed Could you please post or attach the crash dump, decoded using idf_monitor (`make monitor`)?
This crash dump will likely point to the function you are calling from an ISR, which is not in IRAM. The rule for ISRs placed into IRAM is that they must not call functions which aren't in IRAM. If you place an ISR into IRAM, then all the code it calls must be in IRAM.

edjeed, 2018-09-21:
@igrr I changed a lot of code, if the error occurs again I will post the dump... Thanks for the fast reply.
@dongdong004 Thanks!!!

Alvin1Zhang, 2019-03-15:
Feel free to reopen, thanks. @edjeed

closed 2019-03-15 by Alvin1Zhang