espressif/esp-idf #2420: CMake missing common flash targets

paulreimer, 2018-09-17:
## Problem Description
When running from a build dir, using CMake/ninja directly (no idf.py), there are some targets that are missing (and maybe others -- these are the ones I use frequently):
- [x] flash (this target *is* present, so the others should be possible I think)
- [ ] erase_flash
- [ ] monitor

### Expected Behavior
```
mkdir build
cd build
cmake .. -G Ninja
ninja erase_flash
ninja flash
ninja monitor
```

paulreimer, 2018-09-17:
FYI I currently use the following patch to `esp-idf`, in order to get `flash` command working. I'm pretty sure this is also a bug. (I also need to `export ESPPORT=/dev/ttyUSB0` and `export ESPBAUD=921600` before running `ninja`)
```
diff --git a/components/esptool_py/project_include.cmake b/components/esptool_py/project_include.cmake
index 31ce13d6..d635aff0 100644
--- a/components/esptool_py/project_include.cmake
+++ b/components/esptool_py/project_include.cmake
@@ -43,7 +43,8 @@ add_custom_target(app ALL DEPENDS "${PROJECT_NAME}.bin")
 #
 function(esptool_py_custom_target target_name flasher_filename dependencies)
     add_custom_target(${target_name} DEPENDS ${dependencies}
-        COMMAND ${ESPTOOLPY} -p ${CONFIG_ESPTOOLPY_PORT} -b ${CONFIG_ESPTOOLPY_BAUD}
+        COMMAND ${ESPTOOLPY} -p ${ESPPORT} -b ${ESPBAUD}
         write_flash @flash_${flasher_filename}_args
         WORKING_DIRECTORY ${CMAKE_BINARY_DIR}
         )
```

renzbagaporo, 2019-08-21:
Hi @paulreimer, a commit implementing `monitor` and `erase_flash` is on its way. Sorry for the delay.

closed 2019-10-13 by igrr