espressif/esp-idf #2421: BasicOTA not working

Dawge, 2018-09-17:
Hi,
I have the following issue with BasicOTA on my ESP-WROOM 32.
I just modified the field "ssid" and "password" with my wifi logs.

When i run it, i get this output on Serial monitor

> ets Jun  8 2016 00:22:57
>
>rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
>configsip: 0, SPIWP:0xee
>clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
>mode:DIO, clock div:1
>load:0x3fff0018,len:4
>load:0x3fff001c,len:956
>load:0x40078000,len:0
>load:0x40078000,len:13076
>entry 0x40078a58
>Booting
>E (21) wifi: misc_nvs_load 43 ret=4364
>E (22) wifi: wifi_init 1408 ret=4364
>Ready
>IP address: 10.0.0.6

I added nvs_flash_init() in app_main as mentioned in #1478, but still not fixing the problem.

However, a new Network port is available on Arduino IDE, but when i try to upload a new sketch i get the following message (without asking any password)

>Sending invitation to 10.0.0.6 ..........
>18:44:39 [ERROR]: No response from the ESP
>18:44:39 [ERROR]: No response from the ESP

Thanks for your time and your help.

tzxespressio, 2018-09-19:
please try `make erase` ，then try `make flash` again

Alvin1Zhang, 2019-03-13:
@Dawge Hi, feel free to reopen the issue, thanks.

closed 2019-03-13 by Alvin1Zhang