espressif/esp-idf #2425: LwIP TCP ack live-lock/crash on double zero-window

nordlien, 2018-09-18:
This was experience on loopback on esp32:

Two TCP sockets (A & B) connect and sends data, filling up the window. No data is read from either.

The sockets will enter a loop where:

1) A -> B ack with window-size 0
2) B -> A ack with window-size 0

As it's processed on the loopback device, the packages are processed in a single callback, thus locking up the tcpip thread. While it will not live-lock on non-loopback devices, the ack-ack race would probably still happen.

Suggested fixes:

It's clear this ack->ack loop should not happen. I'm not very familiar with these details of the TCP stack, so I'm not sure what the correct fix is. Please advise and I can create an appropriate fix.

1) https://github.com/espressif/esp-lwip/blob/e6bb43349ed859eba4c1e3cae15b932627c45c4b/src/core/tcp_in.c#L1708: patch to first check if seqno = rcv_next and rcv_wnd = 0, and in that case skip the responding ack.
2) https://github.com/espressif/esp-lwip/blob/e6bb43349ed859eba4c1e3cae15b932627c45c4b/src/core/tcp_out.c#L1025 add additional check for wnd > 0. This would mean no acks are sent while window-size is zero, but I'm not sure that is intended behavior (data can still be received and processed locally, even though receiver is stuck).

I've verified both fixes addresses the issue, and will correctly trigger the zero window probe with a backoff.

liuzfesp, 2018-09-25:
@nordlien  please help

zhangyanjiaoesp, 2018-11-19:
@nordlien I'm sorry for not replying to you in time. I can't reproduce your probelm, can you provide your steps to reproduce this issue?

Alvin1Zhang, 2019-04-19:
@andersjohnsen Hi, would you help share if any updates for the issue? Thanks.

andersjohnsen, 2019-05-02:
I applied the following fix to our fork:

https://github.com/toitware/esp-lwip/commit/5c448b22810c41c10198c63b89989c20fdff8b8f

It's stopping us from issuing an immediate ack, if the prober is already installed.

freakyxue, 2019-11-13:
hi @andersjohnsen Thanks for the contribution code, we will review

freakyxue, 2019-12-17:
@andersjohnsen how can i reproduce the problem,because I don't understand your fix

liuzfesp, 2020-03-09:
Hi @andersjohnsen, I think it may have problem.

On one hand, if the receive window of peer is 0, the persistent timer will be started, the zero-window-probe frame will be sent out when the persistent timer expires. The sender of zero-window-probe expects to get the ACK from the receiver.
On the other hand, if my receiving window is 0, upon receiving the zero-window-probe frame, we should reply the ACK immediately to notify my latest receiving window to peer.
Moreover, if both side has zero window, both side will send zero-window-probe ACK when persistent expires. 

As for sending the zero-probe-window probe, LWIP constructs the zero-widow-probe packet with data len is 1, to be specified, it contains a TCP sequence that already sent out. So, on the receiver side, it will found that the zero-probe-window as a duplicated data and responses the TCP ACK immediately (see https://github.com/espressif/esp-lwip/blob/e6bb43349ed859eba4c1e3cae15b932627c45c4b/src/core/tcp_in.c#L1374).

I think the scenario describe in this issue should be very special, it should NOT happen in real application. Moreover, if it do happens, I think it should last a very short time, the zero-probe-window will stop once the data is received by application. 

So I would like to close this issue, feel free to reopen it if you think this issue really matters.

Alvin1Zhang, 2020-03-10:
@andersjohnsen Thanks for reporting. And @liuzfesp thanks for the explanation. Feel free to reopen the issue. Thanks.

andersjohnsen, 2020-03-10:
> Hi @andersjohnsen,
> 
> On one hand, if the receive window of peer is 0, the persistent timer will be started, the zero-window-probe frame will be sent out when the persistent timer expires. The sender of zero-window-probe expects to get the ACK from the receiver.
> On the other hand, if my receiving window is 0, upon receiving the zero-window-probe frame, we should reply the ACK immediately to notify my latest receiving window to peer.
> Moreover, if both side has zero window, both side will send zero-window-probe ACK when persistent expires.
> 
> As for sending the zero-probe-window probe, LWIP constructs the zero-widow-probe packet with data len is 1, to be specified, it contains a TCP sequence that already sent out. So, on the receiver side, it will found that the zero-probe-window as a duplicated data and responses the TCP ACK immediately (see https://github.com/espressif/esp-lwip/blob/e6bb43349ed859eba4c1e3cae15b932627c45c4b/src/core/tcp_in.c#L1374).
> 
> I think the scenario describe in this issue should be very special, it should NOT happen in real application. Moreover, if it do happens, I think it should last a very short time, the zero-probe-window will stop once the data is received by application.
> 
> So I would like to close this issue, feel free to reopen it if you think this issue really matters.

Hi @liuzfesp,

Thanks for looking into the issue. I'll reply in opposite order.

We have experienced this when doing loopback communication over TCP. It sadly doesn't happen as you describe; once this state was reached, it would deadlock the entire system, stuck busy-looping while trying to send out the "ACK 0 window size". We have thus been forced to patch LWIP to not brick our devices.

That aside, to the best of my knowledge, I agree with what you have written down here - it's indeed how I expect this logic to behave. What's the problem here is that there is a bug, such that it get's stuck in 

1) enqueue a 0-window-size ack
2) try send said 0-window-size ack
3) that will enqueue a 0-window-size ack
4) etc...

thus a busy-loop state is reached, on top of the event loop. My fix is an attempt to avoid that (don't allow sending out a 0-window-size message more than once, every `zero-probe-window` (at least that was my goal)).

Now, I'm no expert of either LWIP or TCP in general. I can only confirm that this was a blocking issue for us. Unfortunately, our system has an entire VM and process model built on top of esp-idf, so reproducing it in code that can live outside our repository is very cumbersome to do. Should it though be necessary for this bug to be identified, I will definitely attempt :)

Thanks,

Anders

liuzfesp, 2020-03-10:
Hi @andersjohnsen is it convenient for you to provide a demo code for me to replicate this issue?

freakyxue, 2020-03-16:
hi @andersjohnsen 
Can you provide the following reproduction code and method.

Alvin1Zhang, 2020-06-08:
@andersjohnsen Thanks for the updates, would you please help share any more updates for the issue? Thanks.

Alvin1Zhang, 2020-06-30:
@andersjohnsen Thanks for the updates, would you please help share any more updates for the issue or how we could reproduce the issue? Thanks.

Alvin1Zhang, 2020-08-05:
@andersjohnsen hi, would you please help share any more updates for the issue or how we could reproduce the issue? Thanks.

closed 2021-02-26 by Alvin1Zhang