espressif/esp-idf #2426: esp-idf update breaks arduino projects that use arduino as component

qt1, 2018-09-18:
Arduino projects that are under esp-idf are broken after last  `cd esp-idf && git pull`

There are various problems that arise from compilation. e.g.:
https://github.com/espressif/esp-idf/issues/2392
https://github.com/espressif/arduino-esp32/issues/1853

There may be other, more sublate problems that may be affect runtime.

This is a repeating problem.

Looking in the git log it seem like the the C code is maintained in separate streams of updates.

Suggestions:
1. Test esp-idf for compatibility with arduino-as-component before publishing to the esp-idf master
2. Add revision tags to esp-idf git so it will be possible to rollback to known working position
3. Make should decisively select on of the versions and include only one source in compilation
4. Maintain a single source for idf library for both idf and arduino (e.g. make changes only in idf and auto-copy to arduino)

Please advise

qt1, 2018-09-18:
Hello @me-no-dev ,
The esp team is doing a great job and I like you product and your efforts and that everything is open.

Unfortunately every now and then there are issues that can easily be avoided simply by preforming compilation test against arduino-esp before publishing to esp-idf master.

Personally I have already reverted to an older idf and everything works.

I would suggest that there would be a permanent branch or tag that will be known to compile.
Ideally it would be master.. but if not than maybe 'arduino_safe' branch on idf that will always work well with current arduino component.

igrr, 2018-09-19:
@qt1 Our general approach is that downstream projects should use fixed version of ESP-IDF which they were tested against. For example, [ADF](https://github.com/espressif/esp-adf), MDF(https://github.com/espressif/esp-mdf), and [esp32-at](https://github.com/espressif/esp32-at) all contain ESP-IDF as a submodule. Submodule points to this IDF version which is known to be compatible.
Arduino-esp32 project doesn't specify ESP-IDF version using a submodule at the moment, so you have to check which is the supported IDF version by looking at git history: https://github.com/espressif/arduino-esp32/commits/master/tools/sdk.
Sometimes it might be okay to use a more recent version of IDF than the one which Arduino was last tested against, but as you have found, at times this doesn't work.
Including arduino-esp32 into our CI flow to get early alerts of such issues is a good suggestion, although it will likely not be used to prevent ESP-IDF from being synced to Github.

closed 2018-09-19 by igrr

qt1, 2018-09-21:
@igrr, @me-no-dev

Unfortunately ESP-IDF and arduino-component are tied together but are on separate and unrelated git trees.
This makes it difficult to update correctly.
This is not a trivial thing, I encountered this several times already and I see others have too..
In some cases this can not be avoided just by not updating, for example when installing a fresh environment.

A possible (theoretical?) solution would be to to have both esp-idf and components/arduino be git sub-modules of the project so a project will automatically pull consistent revisions of both.
(this will take about 800MB, but will be consistent)

A more minimalistic option would be to add a very short script in arduino-esp32 that will pull the correct revision of esp-idf .
E.g something like `cd $ESP_IDF && git pull origin/CORRECT_REVISION`

Keeping this script up-to-date will require occasional maintenance from the arduino-esp team but I think it will prevent problems an keep the environment more consistent so it will probably be worth it..

chegewara, 2018-09-21:
Its not so simple as you said. This script can be created but `CORRECT_REVISION` will always point to most recent release version, and this is simple. But if you expect that it will point to most recent working master commit that is working then it requires from arduino team to test every single commit against arduino-esp32 last commit. If you have 30 commits (sometimes its more in one master update) then its a lot work.

This is not that i dont understand your point, in esp-idf master you can have some updates/patches that are not present in most recent release version, but you can always try with master commit and if there is some conflict then you can downgrade to release version. 

I see one more option. If its possible we can have scripts with short how to that will allows us to compile esp-idf arduino-esp32. This way if we need we could TRY to compile master esp-idf and to have control over arduino-esp32 libraries. I know its only for experienced users, but there is few that could have use it.

qt1, 2018-09-21:
I would suggest to automate this in a very simple way:  every time you perform a git commit to arduino (pre-commit hook) the IDF revision will also be recorded.

It is practical, virtually no maintenance, and will insure consistency.

me-no-dev, 2018-09-22:
@qt1 it's the way it's done every time https://github.com/espressif/arduino-esp32/search?q=idf-update&type=Commits

qt1, 2018-09-22:
@me-no-dev Maybe I am missing something. I am looking at the link, where can I find the IDF revision ?

qt1, 2018-09-22:
Here is a pre-commit hook that I added to my project so I can keep track of idf that was used if I need to get back to a revision in future.
Something similar can be added, maybe use python for better instead of bash for better windows/mac compatibility.

    #!/bin/bash
   
    pushd $IDF_PATH >/dev/null 
    idf_revision=$(git rev-parse --verify HEAD)
    popd >/dev/null
    echo $idf_revision > .idf_revision
    git add .idf_revision

The difference between this and [https://github.com/espressif/arduino-esp32/search?q=idf-update&type=Commits](https://github.com/espressif/arduino-esp32/search?q=idf-update&type=Commits) is that here the revision is updated every commit - no  need to know exactly what's happening inside IDF..