espressif/esp-idf #2430: Random WiFi performance

andrewvoznytsa, 2018-09-18:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version: v3
- IDF version: c5265b1
- Development Env: Make
- Operating System: Windows
- Power Supply: USB
- WiFi AP: MikroTik-RBD52G-5HacD2HnD
- WiFi AP config: AP, 2.4+5Ghz, WPA2, PSK (QuickSet 'Home AP Dual' according to MikroTik)
- WiFi Signal Level: -54..-52dB (according to MikroTik reports), stable

## Problem Description

HTTPS download bandwidth is not consistent - if you log on log.txt (see attached zip) then you notice that download process (blocking recv() call at the end) does not return any data for 'idle' amount of time (48 seconds, 35 seconds, and so on - piece of log is below).

During other runs I've seen much higher download bitrates and smaller/fewer idle periods. Most likely you'll need to run test app a few times.

Problem exists (visible) if **downloading over https and from iradeo.com**

I don't see (or didn't notice) such issues from other sites.

This bug was also tested on home made board based on ESP-WROOM32 on other continent - the same results. See attached screenshot.

wget https://link on PC (over wires and the same ISP link or the same WiFi AP which ESP32 uses) always downloads this file in 3 seconds.

I think this issue does not depend on WiFi AP and code. Most likely it is related to mbedtls and/or some iradeo.com specific TLS settings or behavior. Or WiFi phy.

I'm open to run some extra test to help you resolve that issue.

## Steps to reproduce

1. configure defaults - make defconfig
2. configure WiFi - make menuconfig and go to 'Example Configuration'
3. compile https_get example
4. run https_get example
5. watch log
6. repeat 4, 5 a few times to collect more logs

## Code to reproduce this issue

[https_get.zip](https://github.com/espressif/esp-idf/files/2392494/https_get.zip)

## Debug Logs

Full log can be found in [https_get.zip](https://github.com/espressif/esp-idf/files/2392494/https_get.zip)

Here is small snippet to show issue.

I (5097) https_get: time = 1, size = 512, avg bitrate = 4096
I (5098) https_get: heap = {243968, 0}
I (5734) https_get: time = 2, size = 98816, avg bitrate = 395264
I (5735) https_get: heap = {234372, 0}
E (55705) https_get: idle = 48
I (55705) https_get: time = 51, size = 262656, avg bitrate = 41200
I (55705) https_get: heap = {248924, 0}
I (55709) https_get: time = 52, size = 263168, avg bitrate = 40487
I (55716) https_get: heap = {248924, 0}
I (56938) https_get: time = 53, size = 311808, avg bitrate = 47065
I (56939) https_get: heap = {243752, 0}
I (57872) https_get: time = 54, size = 442880, avg bitrate = 65611
I (57873) https_get: heap = {200200, 0}
E (93716) https_get: idle = 35
I (93717) https_get: time = 90, size = 492032, avg bitrate = 43736
I (93717) https_get: heap = {252424, 0}

[image]

liuzfesp, 2018-10-23:
Hi @andrewvoznytsa , sorry for late response, could you help to try this IDF commit: d57132b2129838a0d7193fbebc363391ba3e1b6d, we have done a lot of WiFi optimizations to improve the throughput/stability.

andrewvoznytsa, 2018-10-23:
Hi,

thanks for update - below is typical log from my side. I'd note that
download speed was greatly improved. But there are still idle periods. On
first sight (did 4 test runs) idle periods are much shorter (usually ~1
second) but I noted a few times 10, 30, 36 and 60 seconds idles. For me it
means that issue is not resolved (60 second delay on resource constrained
streaming audio player is not acceptable - I'm not able to compensate it).

Important note - I was testing current master (
129d32772e5e5eafe88be5b9eb34687e84a6f8b8).

If you think that it was not right and I had to try
d57132b2129838a0d7193fbebc363391ba3e1b6d as you asked - let me know and
I'll do that.

Log is here:

I (11519) https_get: heap = {223304, 0}
I (12536) https_get: time = 9, size = 1245696, avg bitrate = 1107285
I (12536) https_get: heap = {225028, 0}
I (13517) https_get: time = 10, size = 1556992, avg bitrate = 1245593
I (13517) https_get: heap = {223304, 0}
I (14550) https_get: time = 11, size = 1884672, avg bitrate = 1370670
I (14550) https_get: heap = {223304, 0}
I (15516) https_get: time = 12, size = 2195968, avg bitrate = 1463978
I (15517) https_get: heap = {226756, 0}
E (21939) https_get: idle = 5
I (21939) https_get: time = 18, size = 2408960, avg bitrate = 1070648
I (21940) https_get: heap = {240544, 0}
E (24282) https_get: idle = 1
I (24282) https_get: time = 20, size = 2458112, avg bitrate = 983244
I (24282) https_get: heap = {235492, 0}
I (24556) https_get: time = 21, size = 2507264, avg bitrate = 955148
I (24557) https_get: heap = {238120, 0}
I (25594) https_get: time = 22, size = 2523648, avg bitrate = 917690
I (25595) https_get: heap = {240572, 0}
E (28195) https_get: idle = 1
I (28196) https_get: time = 24, size = 2589184, avg bitrate = 863061
I (28196) https_get: heap = {238852, 0}
E (29595) https_get: idle = 1
I (29595) https_get: time = 26, size = 2605568, avg bitrate = 801713
I (29595) https_get: heap = {238892, 0}
E (32493) https_get: idle = 1
I (32494) https_get: time = 28, size = 2654720, avg bitrate = 758491
I (32494) https_get: heap = {231928, 0}
I (32535) https_get: time = 29, size = 2671104, avg bitrate = 736856
I (32536) https_get: heap = {235376, 0}
I (33555) https_get: time = 30, size = 2982400, avg bitrate = 795306
I (33555) https_get: heap = {235376, 0}
I (34527) https_get: time = 31, size = 3293696, avg bitrate = 849986
I (34528) https_get: heap = {237100, 0}
I (35509) https_get: time = 32, size = 3588608, avg bitrate = 897152

liuzfesp, 2019-03-22:
Hi @andrewvoznytsa we have fix some issues related WiFi throughput, could you help to check whether your issue still exists in latest release (v3.0 or 3.1 or 3.2 or 3.3 or master)?

andrewvoznytsa, 2019-03-22:
Hi,

I'm using 0f927791bec3c18b2c2c96d13fd45e807ddc1b57 last two weeks - works MUCH better than any other builds which I've used before (I'm updating every 3-4 months, on average).

BT and WiFi (balanced) coexist without any noticeable issues, HTTPS download test, which I mentioned above, performs much better (~1-2Mbps download rate).

One thing to note - WiFi AP is 2 meters away from ESP32 module (previous WiFi configs/locations are not easy replicate).

If WiFi AP is at is usual location (10m away, ~50cm of wall bricks between them in total) then performance is HORRIBLE. Other WiFi clients work at these conditions perfectly.

liuzfesp, 2019-03-22:
@andrewvoznytsa for cross-war and far-away WiFi performance, we will continuously improve it.

liuzfesp, 2020-03-13:
Hi @andrewvoznytsa, we have more optimization in latest IDF release, maybe you can check it on latest master branch.

andrewvoznytsa, 2020-03-15:
Hi @liuzfesp, perhaps at some point as part of my hobby - that project where we were using ESP32 is over.

closed 2020-07-14 by Alvin1Zhang