espressif/esp-idf #2431: Bug report on http client used in native_ota_example: socket not closed

DavideTomasella, 2018-09-18:
I noticed a bug in the esp_http_client.c file. 
I trying to add ota update to my project following the example native_ota_example and creating an HTTP connection (instead of https). When the server (python server hosts on raspberry pi 3) is running and the file is present there are no problems and it works correctly.
This is the modified function:
[Ota.txt](https://github.com/espressif/esp-idf/files/2393560/Ota.txt)

I started trying different situation (for example problems with the server or file not present) and I noticed esp update fails with an error:
`E (877318) HTTP_CLIENT: Connection failed`
`E (877318) ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT`
This is the correct behaviour because the board cannot establish the connection.

At this point I tried to reastart the server (it works correctly) but the board still fail the update with this error:
`E (877308) TRANS_TCP: Error create socket`
`E (877318) HTTP_CLIENT: Connection failed`
`E (877318) ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT`
Reboot is necessary to make it connecting with the server. 

I searching into the library the problem and I noticed that fails at line 884 of esp_http_client.c because cannot create the socket at line 69 of transport_tcp.c. I though that the previous failed connection wasn't correctly closed by the library and the connection port was still busy (I studied electronics so I know these problems). The problem is present also in the old version of the library.

After that I started looking for something strange during closing process and I noticed this at line 978 of esp_http_client.c
`if (client->state >= HTTP_STATE_INIT) {`
This is the condition to close the transport level of the communication and make the port free to the next communication. The state change from HTTP_STATE_INIT to HTTP_STATE_CONNECT during connection operations at line 889 of esp_http_client.c so at line 884, when the connection will fail if the the server isn't on, the state is still HTTP_STATE_INIT but the port is busy because the socket has already be created to open the connection to the server.

I hope I have been enough clear and you can resolve the problem in short time.
Thanks
Davide

jitin17, 2018-09-19:
@DavideTomasella Hey, the file that you have shared is based on the older version of OTA example , which didn't have `esp_http_client` APIs. So it will be more insightful if you can share the file where you have used `esp_http_client` APIs based on the latest `native_ota_example`. Having said that, my speculation is that `esp_http_client_cleanup` API is not being correctly used, or may be not being called at all, but it would be great if you can share modified function with esp_http_client APIs.

jitin17, 2018-09-19:
If you find same issue in the old version of OTA example, which doesn't use `esp_http_client` APIs then I think it's not a `esp_http_client` bug, isn't it?  And since you are using a modified version of the idf example, we would like to understand few more things. 
First, are you using sockets for any other service in your application? The default value of maximum number of sockets in LWIP is 10, which can be changed from `make menuconfig -> components -> LWIP`.
Second, after first failure, are you immediately invoking OTA function again? 
Do you mind sharing your sdkconfig?

DavideTomasella, 2018-09-19:
This is my config.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2397815/sdkconfig.txt)
I have also implemented an udp server but I can deseable it and anything change. The first time the socket is opened, but if it isn't able to find the server running the second time it doesn't work.
I call ota update every 10 seconds.

I'll share you all my [project](https://github.com/DavideTomasella/AQUARIUM-ESP32-ESPIDF-VSCODE-2).

jitin17, 2018-09-21:
@DavideTomasella I modified the `native_ota_example`; I took `esp_http_client_init()` and `esp_http_client_open()` inside a loop and commented out `task_fatal_error()`, and emulated your case of restarting the server and it worked. 
[native_ota_example.txt](https://github.com/espressif/esp-idf/files/2405224/native_ota_example.txt)
So, can you try these things:
1) Update your `esp-idf` and re-run your app, and then try the scenario that you mentioned above.
2) If (1) fails then run `native_ota_example` as a standalone app, with the changes that I have made and then try the same scenario. 

If any of these succeed then the problem is not with `esp_http_client`. 
Meanwhile, your issue might be similar to #2414, but too early to say.

Alvin1Zhang, 2018-10-24:
@DavideTomasella Could you help share if any updates for this issue? Thanks.

kosukekurimoto, 2019-02-04:
Is my problem similar to this?
https://github.com/espressif/esp-idf/issues/2991

I published the entire source code including this problem.
https://github.com/kosukekurimoto/esp32-homekit-natureremo

shubhamkulkarni97, 2019-12-31:
@DavideTomasella Is your issue resolved?

Alvin1Zhang, 2020-03-05:
@DavideTomasella Thanks for reporting. Will close the ticket due to short of feedback. Feel free to reopen with more details. Thanks.

closed 2020-03-05 by Alvin1Zhang

Mauricio-Alencar, 2022-02-09:
I'm having exactly the same problem using esp_http_client, after a few hours of use, the client no longer returns a connection to any socket from the lib, however, I have NTP and mqtt running, and it works fine... The http lib can only close connection with socket after software reset (esp_restart) or hardware reset (enable), please reopen the ticket to give some kind of page....