espressif/esp-idf #2432: SPI master driver lockup

michprev, 2018-09-18:
## Environment

- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 35785d08f

## Problem Description

Not sure this is a bug but it is at least worth discussion as it seems very unintiutive to me. When using SPI master driver on both cores one can easily get into a deadlock.
Consider following example:

- call `spi_bus_initialize` and `spi_bus_add_device` on CORE 0

- keep using `spi_device_transmit` on CORE 1 without a problem

- enter a spinlock on CORE 0 (`vPortCPUAcquireMutex`) which gets unlocked after a call to `spi_device_get_trans_result` on CORE 1

- call `spi_device_get_trans_result` on CORE 1

- CORE 1 gets stuck (never returns from `spi_device_get_trans_result`) on this line https://github.com/espressif/esp-idf/blob/b5365c19f79884546af657dcee77a9d6525bfa6d/components/driver/spi_master.c#L852
more precisely here
https://github.com/espressif/esp-idf/blob/b5365c19f79884546af657dcee77a9d6525bfa6d/components/freertos/queue.c#L1591

### Expected Behavior
CORE 1 should return from `spi_device_get_trans_result` after provided amount of ticks.

Even though it is not a good idea to keep starving CPU from FreeRTOS idle task this still seems like an issue to me.

Let me know if you need more information.

ginkgm, 2018-09-19:
Hi @michprev , 
Thanks for your detailed report.

Some truth:

1. The ISR will only be called on the core where it is registered, in this case, it's core 0.
2. The spinlock will disable the interrupt until it is released.

So essentially you caused a dead lock, by register the ISR on core 0 and disable it from running by using the spinlock.

A basic point is, never wait for anything, delay or even do anything that takes long time when the spinlock is taken. The safe way is to give up the spinlock once you found the event you're waiting has not come, and then try again some time later. You can use QueueReceive/SemaphoreTake with timeout=0 to do this.

BTW, the tick is increased in the core 0, that's why when core 0 is hanged up, the timeout will not happen in core 1.

michprev, 2018-09-19:
Hello @ginkgm,
Thank you for your answer.

> The ISR will only be called on the core where it is registered, in this case, it's core 0.

I thought so.

This is not certainly a bug but it might be worth adding a note into docs that all transactions will be performed by core that called `spi_bus_initialize`.

ginkgm, 2018-09-19:
@michprev ok, we will add a note to the spi_bus_initialize function

closed 2018-11-20 by igrr