espressif/esp-idf #2433: Can we gracefully disable softap while keeping sta connected with AP, without reconnect again?

outermanle, 2018-09-19:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32D]
- IDF version (eea49d250):
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description

We generally use sta+softap mode as configure purpose, we configure station connect to AP with internet,
after station connect to AP, we want to disable AP mode and restore to station mode only.

Now, we need to call wifi stop api then set station mode only and reconnect again, this will lead to station disconnect with AP.

Can we gracefully disable softap while keeping station connected with AP, without reconnect again? 
But  I can't find specific api to disable softap mode only in sta+softap mode.

I think many scene will need this api. Thanks a lot.

### Expected Behavior
In sta+soft mode, we can gracefully disable softap while keeping station connected with AP, without reconnect.

### Actual Behavior
Need to stop wifi then set station mode only and reconnect, this will lead to disconnect with AP.

### Steps to repropduce
100%

liuzfesp, 2018-09-25:
@outermanle, you can call esp_wifi_set_mode(WIFI_MODE_STA) to sta + soft mode to sta only mode, no need to call esp_wifi_stop() to stop WiFi.

outermanle, 2018-09-30:
I will try, thanks a lot!

outermanle, 2018-10-15:
test passed，Thanks！

closed 2018-10-19 by outermanle

ankitdaf, 2019-03-22:
Does this also work for all cases ? That is, can we switch from any mode (AP, STA, AP+STA) to any other mode (AP,STA,AP_STA) without calling esp_wifi_stop() followed by esp_wifi_start() ?

liuzfesp, 2019-03-22:
Sure, @ankitdaf

jhoney2016, 2021-07-30:
Is it possible to change from AP_STA to AP mode without stopping the soft AP